betaflight/betaflight #6066: DSHOT Buzzer & Turtle Mode Interfering with Each Other

EggsBenedict, 2018-06-08:
**Issue:** If, for example, DSHOT buzzer and Pre-arm are on the same switch, DSHOT commands to reverse motors are extremely unreliable (I assume that any situation where the DSHOT buzzer was active during arming would have the same issue).  There appears to have some relationship with timing (moment of arm or length of time you're on the pre-arm/buzzer switch).  The innocuous situation is that turtle mode just never reverses the props.  The scary situation is that turtle mode DOES get enabled, but when you switch back the props aren't reversed and in the case of 3.2.4 the quad goes nuts and tazzes out (I assume anti-taz will take over in later firmwares).  Confirmed across 3 different builds (one of which is on the latest *butterflight* firmware which has anti-taz disabled by default ::face palm::)

So, it would appear the DSHOT commands are stepping on each other.  I guess this is all edge case stuff, but it really could be a nasty one for some people.  I don't know the order of operations here: Perhaps there's a way to stop dshot beacon commands long enough prior to attempting to switch motor directions to avoid conflict?

- Targets: CLRacingF4, Betaflight F3, (and Helio ...) 
- Versions: BF 3.3.2, BF 3.2.4, (and Butterflight 3.5.1 Odin)
- ESCs: Wraith V2 (blheli_32), no-name blheli_s, banggood special blheli_s 4in1 (airbot)

I've tested this on three different builds and configurations (butterflight config omitted).

```# diff

# version
# Betaflight / CLRACINGF4 (CLR4) 3.3.2 May  4 2018 / 09:23:35 (64a292f37) MSP API: 1.38

# name

# resources

# mixer

# servo

# servo mix

# feature
feature TELEMETRY
feature AIRMODE
feature ANTI_GRAVITY
feature DYNAMIC_FILTER

# beeper
beeper -GYRO_CALIBRATED
beeper -RX_LOST
beeper -RX_LOST_LANDING
beeper -DISARMING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -BAT_CRIT_LOW
beeper -BAT_LOW
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -BLACKBOX_ERASE

# map
map TAER1234

# serial
serial 2 64 115200 57600 0 115200
serial 3 2048 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1350 1650 0
aux 1 1 3 1300 1700 0
aux 2 13 2 1750 2100 0
aux 3 35 1 1725 2100 0
aux 4 36 2 1725 2100 0
aux 5 39 3 1700 2100 0

# adjrange

# rxrange

# vtx

# rxfail

# master
set gyro_lowpass_hz = 120
set gyro_notch1_hz = 0
set gyro_notch2_hz = 0
set gyro_stage2_lowpass_hz = 300
set rssi_channel = 12
set rc_interp_ch = RPYT
set serialrx_provider = CRSF
set motor_pwm_protocol = DSHOT1200
set vbat_warning_cell_voltage = 36
set ibata_scale = 270
set beeper_dshot_beacon_tone = 1
set yaw_motors_reversed = ON
set small_angle = 90
set pid_process_denom = 1
set osd_warnings = 33
set osd_rssi_alarm = 80
set osd_cap_alarm = 1000
set osd_vbat_pos = 460
set osd_rssi_pos = 2509
set osd_tim_2_pos = 2552
set osd_throttle_pos = 2528
set osd_current_pos = 2084
set osd_mah_drawn_pos = 2098
set osd_avg_cell_voltage_pos = 2092
set osd_stat_max_spd = OFF
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF
set vtx_band = 1
set vtx_power = 4
set vtx_freq = 5865
set vcd_video_system = 1

# profile
profile 0

set dterm_lowpass_type = PT1
set dterm_notch_hz = 0
set anti_gravity_gain = 3000
set setpoint_relax_ratio = 0
set dterm_setpoint_weight = 100
set p_pitch = 50
set d_pitch = 27
set p_roll = 46
set i_roll = 45
set d_roll = 25
set p_yaw = 65
set p_level = 35

# rateprofile
rateprofile 3```

***Config 2***
```
# diff
# Betaflight / BETAFLIGHTF3 (BFF3) 3.2.4 Jan 10 2018 / 22:39:42 (90d32fd56) MSP API: 1.36

feature -TELEMETRY
feature AIRMODE
feature ANTI_GRAVITY
feature DYNAMIC_FILTER
beeper -GYRO_CALIBRATED
beeper -RX_LOST
beeper -RX_LOST_LANDING
beeper -DISARMING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -BAT_CRIT_LOW
beeper -BAT_LOW
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -BLACKBOX_ERASE
map TAER1234
serial 2 2048 115200 57600 0 115200
aux 0 0 0 1300 1700
aux 1 1 3 1300 1700
aux 2 2 3 1775 2100
aux 3 13 2 1750 2100
aux 4 35 1 1725 2100
aux 5 36 2 1725 2100
set gyro_notch1_hz = 0
set gyro_notch2_hz = 0
set acc_trim_pitch = 8
set acc_trim_roll = 6
set rssi_channel = 16
set dshot_idle_value = 459
set motor_pwm_protocol = DSHOT600
set vbat_warning_cell_voltage = 36
set ibata_scale = 217
set beeper_dshot_beacon_tone = 1
set yaw_motors_reversed = ON
set small_angle = 180
set pid_process_denom = 2
set osd_cap_alarm = 900
set osd_vbat_pos = 329
set osd_rssi_pos = 2445
set osd_tim_1_pos = 54
set osd_tim_2_pos = 2519
set osd_flymode_pos = 333
set osd_throttle_pos = 2497
set osd_vtx_channel_pos = 377
set osd_crosshairs = 200
set osd_ah_sbar = 200
set osd_ah_pos = 200
set osd_current_pos = 2084
set osd_mah_drawn_pos = 2098
set osd_craft_name_pos = 362
set osd_gps_speed_pos = 218
set osd_gps_lon_pos = 82
set osd_gps_lat_pos = 65
set osd_gps_sats_pos = 51
set osd_home_dir_pos = 302
set osd_home_dist_pos = 303
set osd_compass_bar_pos = 266
set osd_altitude_pos = 247
set osd_pid_roll_pos = 423
set osd_pid_pitch_pos = 455
set osd_pid_yaw_pos = 487
set osd_debug_pos = 1
set osd_power_pos = 321
set osd_pidrate_profile_pos = 345
set osd_warnings_pos = 329
set osd_avg_cell_voltage_pos = 2092
set osd_pit_ang_pos = 257
set osd_rol_ang_pos = 289
set osd_battery_usage_pos = 392
set osd_disarmed_pos = 138
set osd_nheading_pos = 311
set osd_nvario_pos = 279
set osd_esc_tmp_pos = 82
set osd_esc_rpm_pos = 83
set vcd_video_system = 1
profile 0

set dterm_lowpass_type = PT1
set dterm_notch_hz = 0
set anti_gravity_gain = 5000
set setpoint_relax_ratio = 40
set dterm_setpoint_weight = 109
rateprofile 0

set rc_expo = 15
set rc_expo_yaw = 15
set roll_srate = 75
set pitch_srate = 75
set yaw_srate = 75
```

Saxin, 2018-06-08:
May be related to #6053

EggsBenedict, 2018-06-08:
Thanks! Yeah, I did see that issue - I tried to figure out if the DSHOT buzzer was active there but couldn't quite tell (hence opening a separate issue).  I'm almost positive that the DSHOT beacon being active while trying to switch to turtle mode is to blame.  This is very reproducible on my end across multiple builds.  If we think these are related I can close this, etc... etc.. just lemmie know.

etracer65, 2018-06-08:
@EggsBenedict Can you test the current 3.4 build? I believe this is fixed.

https://ci.betaflight.tech/job/Betaflight/lastBuild/artifact/obj/

EggsBenedict, 2018-06-08:
No problem. Tested the CLRacingF4 Target (betaflight_3.4.0_CLRACINGF4-911-ef12eebe5.hex).
_Unfortunately, the issue remains_.  This is the build with wraith v2 bl_heli_32. Could've been a fluke - but on one instance none of the motors changed direction. On another - some did.

EggsBenedict, 2018-06-08:
Follow-up: I tested a few more times on the build above.  It's definitely an improvement.  I haven't seen it successfully switch to turtle mode on all the motors yet (frequently one or two motors will switch and the rest will stay the same) however on all attempts thus far - it successfully switches back to normal operation.

EggsBenedict, 2018-06-08:
hm...the configurator doesn't allow you to turn of the dshot beacon and neither does CLI.  Is this intentional?
```
# set beeper_dshot_beacon_tone = 0
###ERROR### Invalid value
Allowed range: 1 - 5
```

etracer65, 2018-06-08:
Yes.  There's a new `beacon` command in the cli and it will be in the next configurator.  Lets you configure the beacon like you do the beeper.

mikeller, 2018-06-09:
Yes, this is a known issue, and unfortunately there is no good fix for it (and it's also dependent on the ESC firmware that is used.

What happens is that with some (or all) of the ESC firmware, after a beacon command is received, the ESC produces the beacon sound for a given period of time (which is dependent on the ESC firmware and beacon tone selected), and does not listen for Dshot frames while it's doing that. This means that, if the beacon is on, and the 'reverse motors' signal happens to be sent within this period of time, the 'reverse motors' signal will be missed.

I don't see a way to fix / work around this problem in software, the only thing is to ask users to **not** configure Dshot beacon / crash flip to be active at the same time.

EggsBenedict, 2018-06-09:
thanks for the responses! I thought that this might be the case - I don't know if this is such a remote edge case that you shouldn't worry about it, but I wonder if there's any way to work a warning into the configurator or something for those that activate the dshot beacon? For those that rely on turtle mode this can disable its benefits in such a way that's not easy to see.  And for those that disable runaway take off prevention (this is what antitaz became, right?) can turn into a nasty/unexpected taz event.

mikeller, 2018-06-09:
@EggsBenedict: Yeah, I think a warning in configurator is probably the best we can do. Can you open a feature request for this in https://github.com/betaflight/betaflight-configurator/issues please?

etracer65, 2018-06-09:
@mikeller A couple of random thoughs - maybe nothing.

Two possible suggestions - both assume we have an idea how long it takes for the ESC's to complete a beep and be available again.

1. Maybe set up an arming disabled reason for cases where the pilot tries to arm "too soon" after a dshot beacon command was sent.  Would encourage people not to have their modes set up so their arming passes through the same switch as RX_SET beacon.  Also solves the case where there's another switch with it active and arming is attempted.  Of maybe limit it to only when entering turtle mode.

2. Add a similar delay condition to `tryArm()` in fc_core.c so that arming delays the ESC beep "grace period" - Basically just not arm until the time since last beacon beep is long enough.

EggsBenedict, 2018-06-09:
@mikeller no problem. as requested - issue created.
@etracer65 - those ideas sounds great. thanks for brainstorming. I'm going to leave this open for this, but please feel free to close the issue if you guys determine that this isn't going anywhere (or I can - just let me know).

mikeller, 2018-06-09:
@etracer65: This is definitely an option. The problem is that some of the beeps can be long - on KISS one of the beeps is about 2 seconds, and we don't have a way to know how long the particular beep is. But I guess if we add a generous amount of delay it will work.

etracer65, 2018-06-09:
@EggsBenedict I have a fix that should help.  #6079 

If a dshot beacon command has been sent within 2 seconds of arming, the arming will delay the time remaining in the 2 second window to help prevent the motor direction command from failing.

Likewise the beacon won't be sent until 2 seconds after disarming to prevent interfering with the motor direction reset commands.

Before these fixes I was able to replicate the motors inconsistently being reversed when using turtle mode and arming while the beacon was active.  After the changes I could not cause the problem.

Let me know what target you need and I'll give you a build to try.

EggsBenedict, 2018-06-09:
COOL! That sounds pretty ideal - CLRACINGF4 would be the easiest.  (Second option would be, BETAFLIGHTF3).  I probably can't test tonight but can try to test tomorrow.

etracer65, 2018-06-09:
@EggsBenedict Here you go:

[betaflight_3.4.0_CLRACINGF4.hex.zip](https://github.com/betaflight/betaflight/files/2087422/betaflight_3.4.0_CLRACINGF4.hex.zip)
[betaflight_3.4.0_BETAFLIGHTF3.hex.zip](https://github.com/betaflight/betaflight/files/2087423/betaflight_3.4.0_BETAFLIGHTF3.hex.zip)

PR #6079 (2018-06-10): Add guard time between dshot beacon and arming/disarming
Fixes #6066 

Tries to prevent DSHOT beacon commands from interfering with commands to set the motor direction.

Adds a 2 second delay after disarming before DSHOT beacon commands will be sent. This attempts to prevent the beacon commands from interfering with the motor direction reset that happens after using crash flip mode and disarming.

During arming if a DSHOT beacon command has been sent within 2 seconds the arming will be delayed until the 2 seconds have passed. This attempts to prevent interference with the motor direction comma [... cut]
closed 2018-06-10 by mikeller