betaflight/betaflight #6086: Add noticeable RC roll/pitch deadbands in anti-turtle mode

controlflow, 2018-06-10:
Hi,

I have a pretty loose Taranis sticks (intentionally) and non-zero, but small RC deadbands (like 5 or so). Sometimes I accidentally have anti-turtle mode enabled after crash while I'm starting normal flying in LOS to check if quad works after crash at all — arming results in two props spinning and this confuses me a lot. It seems like two of the ESCs are fried, until I use googles or if I try to move roll/pitch a bit and realize it's turtle-mode is enabled, that's just my roll stick is slightly out of deadband to the right...

It seems for me that the fact that some props are spinning (even at idle speed) or not is too important to be expressed via roll/pitch stick being inside RC deadband (that is usually) or not.

When quad is flipped over somewhere far away I may not realize that after I've armed in turtle mode two props are already actively trying to spin, before I try to use roll/pitch stick to try rotating.

I propose to temporary change roll/pitch deadbands in anti-turtle mode to something like `max(100, normal_deadband_for_flying)` to avoid accidental props spins.

p.s. Can DSHOT beacon can also be enabled in anti-turtle mode while props are not spinning?

bkeinert, 2018-06-10:
Which BF version are you referring to? 3.4 RC1 has 15% stick deadband on crash flip. That's 75, with 500 being the maximum stick deflection in one direction. There's some additional deadband on the motor outputs of 20.

Currently, I wouldn't recommend to use DSHOT beacon on crash flip mode, but I think that's something being worked on.

controlflow, 2018-06-10:
I was using 3.3.x. That’s cool to hear someone implemented this in 3.4, will check it soon!

bkeinert, 2018-06-10:
Be aware that crash-flip is now also possible on yaw (using 2 motors, nice to "look around" when stuck on a flat surface or to wiggle out of a tree) and over the diagonal (using 1 motor, helps in desperate situations).

closed 2018-07-17 by stale[bot]