betaflight/betaflight #6088: Arming twice doesn't work unless the turtle says you can

micaheli, 2018-06-11:
Summary:
FC boots seemingly OK (ESC tones are out of sync - which is different than my 3.2.1 setup I came from).  Aircraft arms correctly, but if disarmed, will not re-arm.   OSD says "ARMED" when additional attempts are made. The only arming status lines (when connected to BF) are MSP and CLI.   If turtle mode is activated and then deactivated, you can then arm the quad, but you must do this every time you wish to arm.

Configuration:
FC: Omnibus F7 V2
ESC: DYS F30A 4n1
VTX: IRC Tramp
RX: FrSky M-XSR
TX: FrSky X9D+

Things I've Tried:
* Tried this exact setup on the following versions:
Betaflight 3.2.5
Betaflight 3.3.3
Betaflight 3.2.0
Butterflight 3.5.0

All of those worked great.  Might be worth noting that BF 3.2.5 and 3.3.0 had the ESC bootup tones consistently in sync.  All of the rest were out of sync.

Steps to reproduce:
Flash FC with 3.4.0 RC1
Set gyro_to_use = SECOND (MPU 6000)
Calibrate the accelerometer
Configure ports for IRC Tramp, S.Port telemetry and Serial RX
Configuration tab: Reverse motors, DSHOT 600, Small Angle 180, turn off baro and mag,  run 8k/8k, turn on telemetry, air mode, set model name.
PID Tuning tab: Set filter settings to D-Term Notch ON
Receiver tab: TAER1234 and RSSI Channel
Modes Arm/Arm on Aux1 and Aux3 high
Modes Beeper on Aux 3 Mid
FlipOverAfterCrash on Aux 3 High
Set up OSD
CLI: 
    set gyro_to_use = SECOND
    beacon rx_set rx_lost

micaheli, 2018-06-11:
Worth noting:  One time, it DID arm a second time, but ONLY on motor 1.

micaheli, 2018-06-11:
(and just tried it again and all but motor 3 came on for the second arm)

micaheli, 2018-06-11:
Whatever.. it is what it is.  Die github now that it's owned by MS.

Robogenisis, 2018-06-11:
I'm having the same issue with 3.4RC1 on my Hobbywing F4 (OmnibusF4SD), arming only works the first time unless you flip the beacon (or turtle?) switch first or power cycle. Sometimes one of the motors will spin on second arm. This only happens with turtle mode enabled, disable it and the issue goes away.

https://github.com/betaflight/betaflight/issues/6053

EGParadox, 2018-06-11:
Same issue:
`# Betaflight / MATEKF405 (MKF4) 3.4.0 Jun 10 2018 / 02:36:39 (a5ba01666) MSP API: 1.39
board_name MATEKF405
manufacturer_id 
# resources
resource MOTOR 1 C07
resource MOTOR 2 C09
resource MOTOR 3 C06
resource MOTOR 4 C08
resource MOTOR 5 NONE
resource MOTOR 6 NONE
resource SERIAL_TX 11 A15
resource CAMERA_CONTROL 1 A08
# feature
feature LED_STRIP
feature AIRMODE
# serial
serial 0 8192 115200 57600 0 115200
serial 30 32 115200 57600 0 115200
# aux
aux 0 0 0 1700 2100 0
aux 1 1 2 900 1300 0
aux 2 13 1 1700 2100 0
aux 3 35 2 1700 2100 0
aux 4 36 1 1700 2100 0
aux 5 39 0 900 1700 0
# master
set mag_hardware = NONE
set baro_hardware = NONE
set min_check = 1025
set max_check = 1975
set rssi_channel = 8
set blackbox_p_ratio = 128
set motor_pwm_protocol = DSHOT600
set align_board_yaw = 180
set bat_capacity = 1300
set small_angle = 180
set pid_process_denom = 1
set osd_rssi_pos = 2512
set osd_vtx_channel_pos = 2103
set osd_mah_drawn_pos = 2519
set osd_pidrate_profile_pos = 2082
set osd_warnings_pos = 2313
set osd_avg_cell_voltage_pos = 2505
set osd_disarmed_pos = 2282
set vtx_band = 5
set vtx_channel = 8
set vtx_freq = 5917
set camera_control_key_delay = 150`

Saxin, 2018-06-11:
Maybe related to #6053

etracer65, 2018-06-11:
@EGParadox I tested with your config and couldn't replicate the problem.  Based on your mode settings I presume you have your pre-arm (aux 2) on a momentary switch which you hold and then arm.  You also have the beeper on the same momentary switch.

If you are using DSHOT beacon (motor beeping) then it's normal that there will now be a 2 second delay when arming because the motors are beeping (described in the release notes).  So try holding the pre-arm switch longer after you hit the arming switch (aux 1).

etracer65, 2018-06-11:
@micaheli Unfortunately I wasn't able to replicate with your config either.  Arms every time.

Saxin, 2018-06-11:
@etracer65  are you trying on same targets, asking because with #6053 at least for me it was target dependent ?

Saxin, 2018-06-11:
I see so at least for me on build 90x (don't remember exact) build # - it was target specific, i.e. on Omnibus Fireworks - worked perfectly but OmnibusF4SD made problems with this issue, details are in #6053

Saxin, 2018-06-11:
Was able to reproduce on **OmnibusF4** target, HGLRC xjb428 stack.
Flashed clean 3.4RC1, minimal setup (RX and modes).

Attached two diffs - one with the issue and another without - clearly affected by "flip after crash" mode being set (not active) in modes tab.
[3.4RC1 Diff No Arming Issue.txt](https://github.com/betaflight/betaflight/files/2091745/3.4RC1.Diff.No.Arming.Issue.txt)
[3.4RC1 Diff Arming issue.txt](https://github.com/betaflight/betaflight/files/2091746/3.4RC1.Diff.Arming.issue.txt)

etracer65, 2018-06-11:
If anybody is available to test, let me know what target you need.  I have a version that won't (probably) fix the problem, may help track down where to look.

Saxin, 2018-06-12:
@etracer, omnibus f4 please.

EGParadox, 2018-06-12:
@etracer65 i dont use DSHOT beacon, same thing on second quad with AirbotF4, today at evening i will test with removed turtle mode on "mode" tab.

chris2396, 2018-06-12:
Same issue using SPEDIX 25 escs on blheli_s fw. Matek f405ctr board.

etracer65, 2018-06-12:
@Saxin Please try this version.

[betaflight_3.4.0_OMNIBUSF4.hex.zip](https://github.com/betaflight/betaflight/files/2093908/betaflight_3.4.0_OMNIBUSF4.hex.zip)

In particular I want to see if:
1. Multiple armings work okay.
2. Confirm that turtle mode still works (also check that motors are reversed while in turtle mode)
3. Confirm motor direction after exiting turtle mode and disarmed (use the motors tab in the configurator to spin motors one by one.
4. Confirm that you can rearm after using turtle mode (I suspect this might fail).

Probably easiest to do all tests while connected to the configurator in the motors tab with the safety switch enabled.  That way between armings you can use the sliders to confirm motor direction.

Bradw-17, 2018-06-12:
I'm having this identical issue on Matek F405-CTR/ Hakrc 30A 4-in-1.

Robogenisis, 2018-06-12:
@etracer65 I just tested that new version and everything seemed to work correctly. I'm able to arm repeatedly, use my turtle/beacon switches as expected, and my motors spin the way they're supposed to.

etracer65, 2018-06-12:
@Robogenisis Thanks for the feedback.  Some extra things to test for you and @Saxin:

Use turtle mode and disarm while the motors are spinning.  For example hold some right stick to spin those motors and while they're spinning disarm (then release the roll stick).  Then test to see if you can arm normally.

Also try it again and this time check the motor rotations in the motors tab after doing the disarm while spinning in turtle mode.

Since I can't replicate the problem this debugging is very helpful to narrow down where the problem is.

Robogenisis, 2018-06-12:
@etracer65 Disarming turtle mode with motors spinning causes those motors to not spin on rearm or in the motors tab. They spin again once you reengage turtle mode and disarm normally.

etracer65, 2018-06-12:
@Robogenisis Thanks, that's what I was expecting and confirms where the problem is (at least the area).  Basically it appears there's a problem with the DSHOT command queueing when the motors are running.

Previously on disarming the firmware would always reset the motor direction if turtle mode was configured.  This occurred right when the motors were being stopped from spinning.  The change in this debug version is to only reset the direction if disarming from turtle mode.  In normal mode disarming the motor direction command is no longer being sent.  In turtle mode the first tests you did worked because you were disarming with the motors not spinning.  Then you confirmed that the problem is related to commands being queued while motors are running by disarming from turtle mode while they were spinning.

Thanks again.

mikeller, 2018-06-13:
@Saxin, @Robogenisis, @Bradw-17: Can you please test with the attached firmware (from #6108):

[betaflight_3.4.0_MATEKF405.zip](https://github.com/betaflight/betaflight/files/2098336/betaflight_3.4.0_MATEKF405.zip)
[betaflight_3.4.0_OMNIBUSF4.zip](https://github.com/betaflight/betaflight/files/2098337/betaflight_3.4.0_OMNIBUSF4.zip)

ServusServorumDei, 2018-06-13:
I have the HGLRC Zeus (3.3.2?) with 16.65, and NOX(3.3.0) also found this
on BetaFPV brushed 3.3.1. I have a DX9 with AUX switch assigned to prearm
(pos -1) and turtle mode (pos 0) arm switch on disarmed. The copters will
only rearm unless I switch to turtle mode then back to prearm.  I observed
isolated incidents in which the craft will rearm immediately without
switching prearm.

Bradw-17, 2018-06-13:
Didn't work unfortunately. Using 3.4 RC1, motor 1 was the only to arm. On the given hex file, no motors armed.

Robogenisis, 2018-06-13:
@mikeller #6108 seems to have the same issues as the current release. The version @etracer65 provided works unless you disarm from turtle mode with motors spinning.

ServusServorumDei, 2018-06-13:
All my birds does not have to have spinning motors. Disarm>turtle
mode>prearm>arm

micaheli, 2018-06-14:
Any chance I can get a test firmware for my F7 board? :) . Omnibus F7 V2

Saxin, 2018-06-14:
@mikeller Just tested the FW you posted, target OmnibusF4,
Same diff's used which were post above by me.

Everything is the same, was able to reproduce via. same steps.
(As before disabling flip after crash - makes the quad able to re-arm)

Robogenisis, 2018-06-14:
I wanted to add that I was also able to get the motors to freeze up by spamming my arm switch repeatedly (at least with etracer's version, I haven't tried with any others). Switching to turtle or beacon mode and back allows them to spin again.

frozenskys, 2018-06-15:
Getting this on an Omnibis F3 (OMNI Target) Waith a racerstar 4-in-1 ESC
Let me know if there's anything you want me to test - just getting a build environment setup now...
```

# diff
# Betaflight / OMNIBUS (OMNI) 3.4.0 Jun 10 2018 / 02:41:11 (a5ba01666) MSP API: 1.39

name Frozenskys

feature -RX_PPM
feature RX_SERIAL
feature LED_STRIP
feature AIRMODE
beeper -RX_LOST
beeper -SYSTEM_INIT
beeper -ON_USB
serial 2 64 115200 57600 0 115200
led 0 5,12::CB:2
led 1 5,13::A:0
led 2 5,14::A:10
led 3 5,15::L:0
led 4 10,15::L:0
led 5 10,14::A:0
led 6 10,13::A:13
led 7 10,12::CB:2
aux 0 0 0 1600 2100 0
aux 1 2 2 1300 1700 0
aux 2 13 2 1700 2100 0
aux 3 35 1 1700 2100 0
aux 4 36 3 1700 2100 0
rxfail 3 h
set acc_hardware = NONE
set acc_trim_pitch = -8
set acc_trim_roll = 22
set baro_hardware = NONE
set min_check = 1010
set rssi_channel = 9
set serialrx_provider = IBUS
set airmode_start_throttle_percent = 35
set rx_min_usec = 996
set blackbox_device = NONE
set motor_pwm_protocol = DSHOT300
set failsafe_delay = 5
set align_board_yaw = 270
set bat_capacity = 1300
set vbat_max_cell_voltage = 42
set pid_process_denom = 2
set osd_warn_esc_fail = OFF
set osd_cap_alarm = 1300
set osd_tim1 = 1025
set osd_tim2 = 2560
set osd_vbat_pos = 2103
set osd_rssi_pos = 2113
set osd_tim_1_pos = 2081
set osd_tim_2_pos = 33
set osd_flymode_pos = 300
set osd_throttle_pos = 225
set osd_vtx_channel_pos = 377
set osd_crosshairs_pos = 200
set osd_ah_sbar_pos = 200
set osd_ah_pos = 200
set osd_current_pos = 86
set osd_mah_drawn_pos = 88
set osd_craft_name_pos = 362
set osd_gps_speed_pos = 218
set osd_gps_lon_pos = 82
set osd_gps_lat_pos = 65
set osd_gps_sats_pos = 51
set osd_home_dir_pos = 302
set osd_home_dist_pos = 303
set osd_compass_bar_pos = 266
set osd_altitude_pos = 33
set osd_pid_roll_pos = 423
set osd_pid_pitch_pos = 455
set osd_pid_yaw_pos = 487
set osd_debug_pos = 1
set osd_power_pos = 321
set osd_pidrate_profile_pos = 345
set osd_warnings_pos = 2313
set osd_avg_cell_voltage_pos = 480
set osd_pit_ang_pos = 257
set osd_rol_ang_pos = 289
set osd_battery_usage_pos = 460
set osd_disarmed_pos = 2282
set osd_nheading_pos = 311
set osd_nvario_pos = 279
set osd_esc_tmp_pos = 82
set osd_esc_rpm_pos = 83
set osd_stat_tim_1 = ON
set osd_stat_tim_2 = OFF
set osd_stat_max_spd = OFF
set osd_stat_endbatt = ON
set osd_stat_max_curr = OFF
set osd_stat_used_mah = OFF
set osd_stat_max_alt = ON
set osd_stat_bbox = OFF
set pwr_on_arm_grace = 0
set vcd_video_system = PAL
profile 0

rateprofile 0

set roll_rc_rate = 113
set pitch_rc_rate = 113
set yaw_rc_rate = 135
set roll_expo = 10
set pitch_expo = 10
set yaw_expo = 10
set roll_srate = 75
set pitch_srate = 75

# 
```

flint723, 2018-06-16:
Not sure if related but this happened (or something like it).  I went into osd menu and when I exited, the osd froze.  So I power cycled it, and then it armed and flew fine.  Afterwards though, it didn't have the first session logged.  Wondering if BB erase is getting triggered somehow?  I do not have BB erase mode set up.

However in the next flight I went into osd, changed a setting, hit save/exit and it saved and armed fine, no freezing of osd or losing BBL.

This was 3.4 929, turtle mode is not enabled at all.  So maybe it isn't related to this issue.

ctzsnooze, 2018-06-16:
I've been playing with builds of 3.4 for a long time and only noticed recently that sometimes I couldn't re-arm.  MSP and CLI status flags only in CLI, but on disconnection from computer still would not arm.  FF_PIKOF4 target, DShot 300.  Disarming while crashing seems to make it more likely.  Power cycling the Taranis or the quadcopter would allow re-arming.  Can't figure why power cycling the Taranis would allow re-arming, but it did.  Didn't try enabling turtle mode.  Will do more testing.

mikeller, 2018-06-16:
@micaheli, @Saxin, @Robogenisis, @Bradw-17, @frozenskys: Can you please test with the attached firmware (from #6130):

[betaflight_3.4.0_OMNIBUSF7.hex.zip](https://github.com/betaflight/betaflight/files/2108129/betaflight_3.4.0_OMNIBUSF7.hex.zip)
[betaflight_3.4.0_OMNIBUSF4.hex.zip](https://github.com/betaflight/betaflight/files/2108130/betaflight_3.4.0_OMNIBUSF4.hex.zip)
[betaflight_3.4.0_OMNIBUS.hex.zip](https://github.com/betaflight/betaflight/files/2108131/betaflight_3.4.0_OMNIBUS.hex.zip)
[betaflight_3.4.0_MATEKF405.hex.zip](https://github.com/betaflight/betaflight/files/2108132/betaflight_3.4.0_MATEKF405.hex.zip)

Bradw-17, 2018-06-16:
Same result. Only motor 1 re-arms.

mikeller, 2018-06-16:
@Bradw-17: What ESCs / ESC firmware version are you using?

Saxin, 2018-06-16:
@mikeller - no re-arm with #6130, same behavior.
Target Omnibus F4, ESC HGLRC XJB428 Blheli_s 16.67 (from Chrome app)

mikeller, 2018-06-16:
@micaheli, @Saxin, @Robogenisis, @Bradw-17, @frozenskys: Another set of hex files to test (from #6130):

[betaflight_3.4.0_OMNIBUSF7.hex.zip](https://github.com/betaflight/betaflight/files/2108251/betaflight_3.4.0_OMNIBUSF7.hex.zip)
[betaflight_3.4.0_OMNIBUS.hex.zip](https://github.com/betaflight/betaflight/files/2108252/betaflight_3.4.0_OMNIBUS.hex.zip)
[betaflight_3.4.0_OMNIBUSF4.hex.zip](https://github.com/betaflight/betaflight/files/2108253/betaflight_3.4.0_OMNIBUSF4.hex.zip)
[betaflight_3.4.0_MATEKF405.hex.zip](https://github.com/betaflight/betaflight/files/2108254/betaflight_3.4.0_MATEKF405.hex.zip)

etracer65, 2018-06-16:
Everybody who is having this issue please report their ESC firmware type (BLHeli_32 or BLHeli_S) and version.

We already have @saxin using BLHeli_S with version 16.67

chris2396, 2018-06-16:
I'm using spedix25 blheli_s escs.

etracer65, 2018-06-16:
@chris2396 Please also supply the BLHeli_S firmware version you are using.

Bradw-17, 2018-06-16:
G-H-30, 16.66, 16.67_dshotcmd
HAKRC 30A 4-in-1

etracer65, 2018-06-16:
Ok, some semi-good news.  I was finally able to replicate the problem and it's related to BLHeli_S firmware version 16.67.  If you upgrade to the official 16.7 release version there is no problem.

We still don't have an answer as to why BLHeli 16.67 is misbehaving as the DSHOT protocol seems to be correct.

etracer65, 2018-06-16:
@chris2396 No it won't  BLHeli_S 16.7 has official support for DSHOT commands and works fine.

Bradw-17, 2018-06-16:
This may seem trivial, but do the different startup tones still work? It has reverted to OEM tone after flashing.

ctzsnooze, 2018-06-16:
OK I can confirm 100% repeatability on RC1 code with FF_PIKOF4 DShot 300 BLHeli 16.67.
It will not re-arm if disarmed while motors are still spinning.  Even if throttle is zero, if motors are spinning still, cannot re-arm afterwards.  Fixes:
- power cycling the Tx, 
- entering turtle mode, attempting to arm (motors don't spin), and exiting turtle mode
- power cycling the quad

ctzsnooze, 2018-06-17:
OK and I can confirm flashing 16.7 to the ESC = no problem :-)

Saxin, 2018-06-17:
@mikeller @etracer65 - Just tried build 938 (a7fae6b8f). Target Omnibus F4SD, Blheli_S 4in1 ESC, Typhoon 35A (Airbot)

ESC FW 16.67 (chrome app)  Issue resolved.

1. Rearming works as expected.
2. Rearming after engaging "turtle" mode works.
3. Turtle mode works, rearming in turtle mode works.
4. Turtle mode - while holding stick to the side (2 motors spinning in turtle mode) disarm, rearming works, rearming into turtle mode also works from this situation.

frozenskys, 2018-06-17:
Can confirm Blheli_S 16.7 has fixed this issue with the Omnibus F3, Racerstar 4-in-1 Blheli_S ESC, and DSHOT300 combo.

mikeller, 2018-06-17:
@micaheli: If this is fixed now, can you please close this?

brycedjohnson, 2018-06-20:
@mikeller I'll pull my down for 16.67.  That was a prototype version and 16.7 has surpassed that.  Definitely more robust for dshot commands then 16.67.  @DieHertz Are you OK with removing them from the blheli_s configurator?

mikeller, 2018-06-20:
@brycedjohnson: Fair enough - the issue is worked around now in Betaflight (by adding a delay between sending a command and sending a motor update), but the problematic sequence might still be sent by other flight controller firmware.

brycedjohnson, 2018-06-20:
@mikeller Removing it from the blheli_s configurator would be the best.  Most people only had it installed because it had dshot_cmds in the name.  Anyone that wants good startup music should which to blheli_32.

mikeller, 2018-06-20:
@brycedjohnson: Ok. I assumed that your Dshot command changes would be merged at some point into the 'open' fork of BLHeli_S, in order to have a fully fledged alternative to the 'official' distribution. If this is not the case, and this is stale, then removing it from the BLHeli configurator is the best option.

If you want good music you should buy a music player, and not an ESC. :stuck_out_tongue_closed_eyes:

micaheli, 2018-06-20:
Sorry, I've been busy. I'll test this tonight and close this.

chris2396, 2018-06-21:
Flashing 16.7 has now resolved my issue. Thanks a lot

micaheli, 2018-06-24:
Confirmed this is resolved.  Closing ticket.

closed 2018-06-24 by micaheli