betaflight/betaflight #6089: GPS Rescue in Betaflight 3.4.0 RC1 climbs past gps_rescue_initial_alt limit

shannonbaker, 2018-06-11:
Whilst flying I enabled GPS Rescue mode and the quadcopter continued to ascend past the gps_rescue_initial_alt limit.

gps_rescue_initial_alt is set to 20 meters.
Metek F405 CTR FC running BF 3.4.0 RC1.
Baro has open cell foam covering it.
I have a UBLOX M8N mini gps without compass.
I both powered and armed facing north.
Noted that I had GPS coordinates and altitude of zero and a heading of zero.

I was in air mode a few hundred meters away at a rough altitude of 15 meters.
The quadcopter turned to face home which looked correct in FPV.

The issue is that the quadcopter began to climb in altitude to a height of approx 140 meters at which point I disabled GPS Rescue mode and was returned to air mode and operating normally.

My expectation was the quadcopter would ascend or descend to the value from gps_rescue_initial_alt and return home until it reached gps_rescue_descent_dist and then begin to descend.

AirBreak69, 2018-06-11:
Thanks for the detailed description. I experienced the same during the last weeks. It gets notably on higher altitudes above MSL. In my case the quad virtually would "rise to the moon" (550m+), if I hadn't disabled the failsafe switch which I used to test the rescue feature.
I made some modified custom builds and now I'am quite sure:
@s0up : gps_rescue.c may catch the absolute altitude as the maxAltitude (without offset included). Although idleTasks() now is inactive unless arming state is there, the offset can be determined too late.
AltitudeOffsetSet from position.c should be used in gps_rescue.c to inhibit memorization of maxAltitude until the offset is surely evaluated.
I hope to be able to publish some code fragments or an own PR during the next days.

s0up, 2018-06-11:
I changed it to only run idleTasks() when armed, but I think because of the frequency that it runs, it can still get a non-offset altitude when the quad is armed but the offset is not applied yet.  I did consider in gps rescue doing 

`    if (!ARMING_FLAG(ARMED) || calculateThrottleStatus() == THROTTLE_LOW) {
        return;
    }`

But i'm not sure if that is the proper way to fix it.

mmakam2, 2018-06-13:
i had this issue also with DALRCF405 Target and a BN-880 gps module. Set the inital alt setting on the lowest possible, quad still climbed very high.

Quickred, 2018-06-14:
I had the same problem and to test the feature I set gps_rescue_initial_alt = 50. If GPS rescue is activated now it reaches an altitude of approx. 60 to 90 meter. It would be great, if the gps_rescue_initial_alt value will be equal to the alt value in the OSD

s0up, 2018-06-14:
Rescue mode should try to go to your rescue initial alt OR the highest alt
hit while armed, whichever is greater.  It should stay pretty close to the
intended altitude assuming the PIDs for rescue mode are setup okay.  There
is an issue on RC1 though that could cause it to climb too high if you're
above sea level!

shannonbaker, 2018-06-15:
Because we fly below sea level?.... :D

s0up, 2018-06-15:
On RC1 there is a chance that it will log the current altitude prior to
applying the offset of your alt above sea level.  So assuming you're say
300 meters above sea level, rescue will think you flew up to 300 meters,
when you arm it will apply the offset and read zero, so now it will be
+300m relative to where you took off from :(

shannonbaker, 2018-06-25:
Would this be fixed with #6161 fix in RC3?

s0up, 2018-06-25:
Should be 100% resolved in RC3

mikeller, 2018-07-07:
@shannonbaker: Can you please confirm that this is working now, and close?

closed 2018-07-07 by shannonbaker

Uroczyroamn, 2018-07-20:
Hi,

I'm using final version of BF 3.4 and there is still problem with unstable altitude in GPS rescue mode.
My altitude sensor reading is ok (I see it on my FPV). I have gps_rescue_initial_alt is set to 30m.
Yesterday my quad in GPS rescue climbed to 300m. What is wrong with this?
Sometimes is working ok (30m), sometimes not. Could You advice something?