betaflight/betaflight #6091: Little desync in blackbox log time

McGiverGim, 2018-06-11:
I don't know if this is possible or not. Here is the problem:

- I have a log video of about 6:30 minutes. This video has been verified to be pretty accurate (we have recorded a chronometer and the chronometer and the video time are more or less exact).
- We have made a flight with the same camera and recorded a blackbox log.
- The blackbox log has being synced with the start of the log and the movements of the video match the movements of the log very precise.
- The video and the log are being desynced when the video/log advances. About 500ms each 4 min 30 seg.

Is possible that in the firmware, the internal clock used to write to the blackbox not be accurate? Some rounding or similar...

DieHertz, 2018-06-11:
Time on FC is as precise as it's oscillator, times between two independent devices need not match

mikeller, 2018-06-12:
In addition to clock skew, I would suspect that if frames are dropped because we can't keep up with writing to the storage medium this might also cause the frames after the drop to fall out of synch. Have you tried if this reproduces with a (much) lower logging rate?

McGiverGim, 2018-06-12:
The Blackbox log file has not dropped frames. The Blackbox Viewer looks for them, it knows the relation between I and P frames (P denom) and verifies that there are not dropped frames. If there are they appear as a cut in the graph lines.

But if a dropped frame occurs, there is not a desync, the next I frame (one each 32ms) contains the timestamp in the first field, so it can resync without problem.

mikeller, 2018-06-12:
@McGiverGim: Thanks for clarifying that.

McGiverGim, 2018-06-13:
Yes. I think that is better that I close the issue. If the problem is a clock skew I suspect we can't do nothing with it.

closed 2018-06-13 by McGiverGim