betaflight/betaflight #6094: Smart Audio Not Working with 3.3.X

HussoFPV, 2018-06-11:
Hey all,

I'm raising a concern with BF 3.3.X smart audio functionality.

Since flashing my FC with the latest firmware, 3.3.3, my smart audio has stopped working.

It wont let me access the menu in the OSD. It just says "StartX & back"

Flashed to 3.3.2, 3.3.1 and even the pre-release 3.3.4, still no luck.

Went back to 3.2.5 and the smart audio starts working as normal.

Hardware;

DYSF4 FC 
TBS unify pro HV - SA wired to TX6 on DYS F4

The issue seems to lie within the latest versions of BF as it works perfectly fine in 3.2.5. Has anyone else had this issue and is there a possible fix for it?

Thank you in advance, hopefully someone can help me out :)

jflyper, 2018-06-11:
@HussoFPV What is your vtx brand and model?

h0st1le, 2018-06-16:
I have this same issue. Using a TBS Unify Pro v1 (5v model) possibly using smart audio v1

jflyper, 2018-06-16:
@h0st1le Was it working with previous version of firmwares?

h0st1le, 2018-06-16:
Yes, I have 2 of the exact same builds, when I upgrade either to 3.3.x, smart audio no longer works. Setting both back to 3.2.5 both work again. If I plug earphones into my goggles the pulsing sound smart audio makes is not there on 3.3.x and above.

jflyper, 2018-06-17:
Can't reproduce.

- Tested using Omnibus F4 AIO (gen 1), almost compatible with DYS F4 Pro.
- Tested with both v3.3.3 and v3.4.0-RC1 for following configurations.
- Unify 5G8 Pro (SA Version 2) and HV Pro (SA Version 2) on UART6 (TX6).
- Unify 5G8 Pro (SA Version 1) on SOFTSERIAL1 over PWM6(PA8).

HussoFPV, 2018-06-17:
Im happy to stick with 3.2.5 as it  flies sweet with the right tune. Hopefully later releases will resolve this issue

h0st1le, 2018-06-17:
Setup is:

* OmnibusF4SD
* Unify Pro 5G8 V1 on UART3 (TX3)

FPVDjp, 2018-06-18:
I am having the same issue with 2 x Asgard AIO F4/F7 32 boards, 3.3.x or 3.4 exhibits this behaviour, 3.2.5 it works fine again.  Also UART 2 which smart audio is connected to on this board disappears on 3.4.

HussoFPV, 2018-06-19:
There's definitely an issue here if were not the only ones. Has anyone found a solution or workaround possibly with the CLI?

FPVDjp, 2018-06-19:
Just solved mine, it appears that the pad I had been using has now been mapped to LED, I have moved to TX6 and now working again

PIFPV, 2018-06-19:
I also have the same issue ever since flashing to 3.3.3 my smart audio comes and goes, now its totally disappeared. been working fine for 3-4 months on 3.2 since flashing couple weeks ago.

FPVDjp, 2018-06-23:
OK, I just had this happen on another board as well, Omnibus F4 Pro.  Turning the VTX to pit mode caused all smart audio communication to cease when it had been working fine after the 3.3 upgrade.  Nothing I did would get it working again.  I moved from TX6 to TX3 and hey presto all working again, it is as if the VTX needs a hard reset signal from another UART to kick it back in to life.  try moving to a different pin and see if yours start to work again also.  3 different FC boards, moved the pin on all 3 and now all working again.

hokasan, 2018-07-08:
I have this problem on 3.4.0 RC5. Im using a Matek F405 CTR with a eachine vtx (this one https://www.banggood.com/Eachine-Wizard-TS215-FPV-Racing-RC-Drone-Spare-Part-5_8G-72CH-25mw-200mw-500mw-800mw-VTX-p-1295235.html?rmmds=search&cur_warehouse=CN)

It worked on 3.2.2.

closed 2018-08-14 by stale[bot]

ju-mpe-r, 2018-11-25:
Have same issue, after upgrading on version 3.5.2 no SmartAudio at OSD, just "-------"