betaflight/betaflight #6095: DShot beeper not working in 3.4 RC 1

ollzer, 2018-06-11:
DShot beeper not working in 3.4 RC 1, worked up to 3.3.3 or 3.3.2.

Tested with:
- KAKUTEF4V2 + Wraith32 v2
- TATTOOF4S

Way to reproduce:
- Pick a board
- set beeper_dshot_beacon_tone something 1-5
- Flip switch
- Nothing 

`
# diff

# version
# Betaflight / KAKUTEF4V2 (KTV2) 3.4.0 Jun 10 2018 / 02:32:57 (a5ba01666) MSP API: 1.39

board_name KAKUTEF4V2
manufacturer_id 

# name
name 

# resources

# mixer

# servo

# servo mix

# feature
feature AIRMODE
feature ESC_SENSOR

# beeper
beeper -GYRO_CALIBRATED
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ARMED
beeper -ON_USB

# beacon

# map

# serial
serial 3 2048 115200 57600 0 115200
serial 4 1024 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 3 1700 2100 0
aux 1 13 4 1700 2100 0
aux 2 35 0 1700 2100 0

# adjrange
adjrange 0 0 2 900 2100 12 2 0 0

# rxrange

# vtx

# rxfail
rxfail 3 h

# master
set gyro_lowpass_hz = 120
set acc_hardware = NONE
set mag_hardware = NONE
set baro_hardware = NONE
set min_check = 1005
set max_check = 1995
set rssi_channel = 16
set rc_interp = MANUAL
set rc_interp_ch = RPYT
set rc_interp_int = 14
set rc_smoothing_type = FILTER
set blackbox_device = SPIFLASH
set dshot_idle_value = 430
set motor_pwm_protocol = DSHOT1200
set bat_capacity = 1300
set current_meter = ADC
set ibata_scale = 275
set ibatv_scale = 275
set beeper_dshot_beacon_tone = 3
set small_angle = 180
set gps_provider = UBLOX
set gps_auto_baud = ON
set deadband = 3
set yaw_deadband = 5
set pid_process_denom = 1
set runaway_takeoff_prevention = OFF
set osd_warn_esc_fail = OFF
set osd_warn_core_temp = OFF
set osd_rssi_alarm = 38
set osd_cap_alarm = 1350
set osd_vbat_pos = 2433
set osd_rssi_pos = 2439
set osd_tim_1_pos = 54
set osd_tim_2_pos = 2455
set osd_flymode_pos = 66
set osd_throttle_pos = 344
set osd_vtx_channel_pos = 2424
set osd_crosshairs_pos = 233
set osd_ah_pos = 200
set osd_current_pos = 288
set osd_mah_drawn_pos = 2370
set osd_craft_name_pos = 2443
set osd_gps_lon_pos = 82
set osd_gps_lat_pos = 65
set osd_gps_sats_pos = 51
set osd_home_dir_pos = 302
set osd_home_dist_pos = 303
set osd_compass_bar_pos = 266
set osd_altitude_pos = 247
set osd_pid_roll_pos = 423
set osd_pid_pitch_pos = 455
set osd_pid_yaw_pos = 487
set osd_debug_pos = 1
set osd_power_pos = 321
set osd_pidrate_profile_pos = 345
set osd_warnings_pos = 2345
set osd_avg_cell_voltage_pos = 2401
set osd_pit_ang_pos = 257
set osd_rol_ang_pos = 289
set osd_battery_usage_pos = 392
set osd_disarmed_pos = 2315
set osd_nheading_pos = 311
set osd_nvario_pos = 279
set osd_esc_tmp_pos = 82
set osd_esc_rpm_pos = 83
set osd_stat_tim_1 = ON
set osd_stat_max_dist = ON
set osd_stat_min_batt = OFF
set osd_stat_battery = ON
set osd_stat_used_mah = OFF
set osd_stat_max_alt = ON
set osd_stat_bbox = OFF
set vtx_band = 5
set vtx_channel = 8
set vtx_freq = 5917
set vcd_video_system = NTSC

# profile
profile 0

set dterm_lowpass2_hz = 250
set iterm_relax = RPY
set d_pitch = 30
set p_roll = 42
set d_yaw = 15

# rateprofile
rateprofile 0

set thr_mid = 40
set roll_rc_rate = 73
set pitch_rc_rate = 73
set yaw_rc_rate = 73
set roll_srate = 63
set pitch_srate = 50
set yaw_srate = 63

# 
`

Saxin, 2018-06-11:
use command `beacon` in CLI,
read here - https://github.com/betaflight/betaflight/pull/5891
close if resolved

ollzer, 2018-06-11:
Ok, thanks. 
Quite weird to need separately add the beacon and  beeper_dshot_beacon_tone.

closed 2018-06-11 by ollzer

Saxin, 2018-06-11:
Tone - chooses tone style, 
Command used to enable / disable under what condition.

richard-scott, 2018-06-12:
This Configurator issue has been resolved for the next release.