betaflight/betaflight #6100: Rssi bug on 3.4 betaflight

djtdf, 2018-06-12:
After flashing my bf f4 fc with the new betaflight 3.4 firmware the rssi stop to 42%!!!!!!

jflyper, 2018-06-12:
@djtdf 
1. Re-flash with full-chip erase or do cli `defaults`.
2. Review your configuration for rssi; source, scale, etc.
3. If the problem persists, please attach cli `diff` result AS INSTRUCTED IN THE PREFORMATTED TEMPLATE FOR A NEW ISSUE.

closed 2018-06-12 by jflyper

tavdog, 2018-06-14:
I updated to 3.4RC1 and the default rssi scale is set to 41.  Is it supposed be set to 41 ?

mikeller, 2018-06-14:
@tavdog: No, it's not, the default is 100. Can you see what it comes out as after you reset to defaults?

djtdf, 2018-06-14:
# version
# Betaflight / BETAFLIGHTF4 (BFF4) 3.4.0 Jun 10 2018 / 02:20:37 (a5ba01666) MSP API: 1.39

board_name BETAFLIGHTF4
manufacturer_id 

# name

# resources
resource SERIAL_TX 1 NONE
resource SERIAL_TX 3 NONE
resource SERIAL_TX 11 A09
resource SERIAL_TX 12 B10

# mixer

# servo

# servo mix

# feature
feature SOFTSERIAL
feature RSSI_ADC

# beeper

# beacon

# map

# serial
serial 1 0 115200 57600 0 115200
serial 30 2048 115200 57600 0 115200
serial 31 32 115200 57600 0 115200

# led

# color

# mode_color

# aux

# adjrange

# rxrange

# vtx

# rxfail

# master
set mag_hardware = NONE
set baro_hardware = NONE
set motor_pwm_protocol = DSHOT600
set current_meter = ADC
set battery_meter = ADC
set pid_process_denom = 1
set osd_rssi_pos = 2093

# profile
profile 0

# rateprofile
rateprofile 0

djtdf, 2018-06-14:
# get rssi
rssi_channel = 0
Allowed range: 0 - 18

rssi_src_frame_errors = OFF
Allowed values: OFF, ON

rssi_scale = 100
Allowed range: 1 - 255

rssi_offset = 0
Allowed range: -100 - 100

rssi_invert = OFF
Allowed values: OFF, ON

osd_rssi_alarm = 20
Allowed range: 0 - 100

osd_rssi_pos = 2093
Allowed range: 0 - 3071
Default value: 234

osd_stat_min_rssi = ON
Allowed values: OFF, ON

djtdf, 2018-06-14:
and aswell dont leave me save the rssi channel, i set it to my channel i save it, and when i check is not saved!!!

djtdf, 2018-06-14:
when i put rssi_scale = 255 it show me rssi=106, from here i need to calibrate it or? its wrong?

djtdf, 2018-06-14:
no is not working.....

tavdog, 2018-06-14:
Resetting Setting's button does set the rssi_scale to 100.  I guess setting defaults should be part of the firmware upgrade process.

djtdf, 2018-06-14:
yea
but still dont work, i wil downgrade to 3.3 to check again

djtdf, 2018-06-14:
but aswell when i put rssi scale to 240 show me the rssi=100 but i check the range and the same is hold to this procent (100)

tavdog, 2018-06-14:
You've got your rssi_adc feature turned on. This will overrisde rssi over an RX channel.  try disabling rssi_adc and then set your rssi_channel and then check it out.

djtdf, 2018-06-14:
I cant set the rssi channel , dont want to save the channel what i set!!!

djtdf, 2018-06-14:
thanks mate! that was, sorry for my mistake, case close!thank you again!