betaflight/betaflight #6103: 3.4 RC1 - RC filtering with Crossfire

zytra, 2018-06-12:
Hi guys
I just loaded 3.4RC1 to check out the RC filtering and realized maybe the default settings aren't good for crossfire running at 150 Hz.
With the default settings (for SBUS I suppose) I still have major steps which have an impact on the PID sum.

thanks

etracer65, 2018-06-13:
@zytra Also please share the log.  I suspect what happened is the your Crossfire receiver switched from 150Hz to 50Hz in flight.  The 50hz mode is 20ms so that aligns with what you saw.  Crossfire will do this when the link becomes poor.  Currently RC Smoothing doesn't handle this change but will be enhanced soon.  But the `rc_smoothing_info` command and log will help confirm this is what happened.

zytra, 2018-06-13:
Thank you, I will run the command when I get home.

Note however that I just switched to CRSF recently and I am planning close attention to the mode, power output and LQ, and on this flight (close proximity) my LQ didn't go lower than 95 and doubt that it switched from 150 to 50 Hz. Interestingly enough the steps are here right after arming.

I read in a PR that there is consideration for real time adjustment of the filter based on the measured frame length; I could switch back to interpolation until then, but since I just switched to CRSF I figured I'd give it a shot. If I do I'll have to tune the interpolation interval which of course will not be tuned properly if the system does switch into 50 Hz mode.

thanks again

zytra, 2018-06-13:
I just ran that command and I think it's outputting what it should...

`# RC Smoothing Type: FILTER
# Detected RX frame rate: 6.684ms
# Auto input cutoff: 67hz
# Active input cutoff: 67hz (auto)
# Auto derivative cutoff: 67hz
# Active derivative cutoff: 67hz (auto)
`

etracer65, 2018-06-13:
@zytra Yes, that looks correct.  Can you please provide the log?  It might show more detail on what was going on with the RX frame rate.  For example depending on your `rc_interp_ch` settings we might be able to see the input steps in yaw if it's not smoothed and calculate the frame rate from that.  You could also try capturing another log with:

`set debug_mode = RC_SMOOTHING`

To capture more detail.  This till log much more detail including the raw input (roll channel by default).  But providing the original log is a good start.

zytra, 2018-06-14:
Thank you, I will either upload the log tonight or fly with RC_SMOOTHING debug enabled.
Thanks

ITSFPV, 2018-06-20:
i have the exact same issue with Crsfx dropping to 50hz from 150hz and in turn causing spiking problems for RC interpolation in filter mode.  It was recommended for me to set the following in the interim. 
rc_smoothing_type = interpolation
rc_interp = manual
rc_interp_int = 22
[image]

etracer65, 2018-06-23:
RC Smoothing has been updated to support Crossfire's ability to change RX rate while in flight.  See #6194 

The changes will be available soon in 3.4 RC3, but if you want to try it sooner you can download the latest build from here:

https://ci.betaflight.tech/job/Betaflight/lastStableBuild/

closed 2018-06-23 by etracer65