betaflight/betaflight #6105: RevoltOSD Target, arming code 12, never flown, gyro won't calibrate

mmakam2, 2018-06-13:
Brand new FC, brand new build, everything else seems to be good, except arming code 12 won't go away. here is my diff:

# diff

# version
# Betaflight / REVOLTOSD (ROSD) 3.4.0 Jun  9 2018 / 03:04:35 (32fc49482) MSP API: 1.39

board_name REVOLTOSD
manufacturer_id 

# name
name 210rV2

# resources
resource MOTOR 2 A02
resource MOTOR 3 B01
resource MOTOR 4 A03

# mixer
mixer QUADX

# servo

# servo mix

# feature
feature TELEMETRY
feature OSD
feature AIRMODE
feature DYNAMIC_FILTER

# beeper

# beacon

# map
map TAER1234

# serial
serial 2 2048 115200 57600 0 115200
serial 5 1024 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1800 2100 0
aux 1 13 0 1400 1600 0
aux 2 26 1 1425 1600 0
aux 3 31 1 1800 2100 0

# adjrange

# rxrange

# vtx

# rxfail

# master
set gyro_calib_noise_limit = 90
set gyro_use_32khz = ON
set acc_hardware = NONE
set rssi_channel = 8
set rc_interp = MANUAL
set rc_interp_int = 12
set serialrx_provider = CRSF
set blackbox_device = SPIFLASH
set dshot_idle_value = 300
set motor_pwm_protocol = DSHOT1200
set current_meter = ADC
set battery_meter = ADC
set deadband = 4
set yaw_deadband = 4
set pid_process_denom = 2
set tlm_halfduplex = OFF
set osd_rssi_pos = 2145
set osd_tim_2_pos = 2433
set osd_vtx_channel_pos = 2401
set osd_crosshairs_pos = 2253
set osd_avg_cell_voltage_pos = 2177
set vtx_band = 1
set vtx_channel = 5
set vtx_power = 4
set vtx_low_power_disarm = ON
set vtx_freq = 5785

# profile
profile 0

set vbat_pid_gain = ON
set setpoint_relax_ratio = 5
set dterm_setpoint_weight = 100

# rateprofile
rateprofile 0

mmakam2, 2018-06-13:
tried 3.3.3, same issue..

mmakam2, 2018-06-13:
flightone, I was able to arm and fly, so this is a betaflight issue not a gyro/hardware issue.

mmakam2, 2018-06-13:
update, I am able to arm and fly if i turn off 32khz mode, so this is an issue related to 32khz mode

mikeller, 2018-06-13:
32kHz mode is only 'experimental' according to the manufacturer of the MPU. There is no guarantee that it will work with every MPU.

closed 2018-06-13 by mikeller

mmakam2, 2018-06-13:
why would the manufacturer advertise 32khz capable then?

mmakam2, 2018-06-13:
there is zero noise on the gyro, this is a software compatibility issue or bug not hardware.

mikeller, 2018-06-14:
@mmakam2: There is nothing that can be done to change the fact that the MPU on the board that you have got does not support 32k.

mmakam2, 2018-06-14:
Manufacturer claims it does..

diggidydaz, 2018-12-09:
> Manufacturer claims it does..

I just wanted to let you know that I have nearly the same exact setup and board.  My quad arms only with 32Khz off just as you described.  I too think that it may a BF issue or the 32k advertised by the RevoltBoard may only be applicable to their proprietary software.   In the meantime, my quad still flies great with BF 3.5 with 32K off.