betaflight/betaflight #6107: ESC DSHOT beacon handling change request

etracer65, 2018-06-13:
Even though this is not a Betaflight issue, raising it here as a centralized place to track.

@ronlix @sskaug @mikeller 

We have a problem with the way DSHOT beacon commands are being processed.  In particular they appear to be blocking in that the ESC will not accept and silently drops any other commands while it's playing the "beep" through the motor.  We see this while trying to set other parameters like motor reversed/normal when entering turtle mode if the user has the DSHOT beacon active.  The end result can be very bad in that the motor directions end up incorrect resulting in a "flip-out" on arming.

To work around this we've had to implement a "guard time" and delay resetting the motors direction (and subsequently delaying arming) if the pilot had the beacon active.  Because we have no idea how long the ESC will be ignoring commands we have to just have an arbitrary delay of about 2 seconds.  Obviously this is not ideal.

I would like to propose a change to the DSHOT protocol handling in the ESC such that commands are still processed while the beacon tone is sounding.

Also I need to acknowledge that I can't confirm this behavior on KISS as I don't have any appropriate ESC's to test - but including for completeness.

ronlix, 2018-06-13:
Thanks,

This Problem should be fixed with the next kissesc Beta. as i already changed the beeps to be non blocking.

Regards

Felix.

mikeller, 2018-06-13:
As a suggestion, now that we are looking into addressing this, and in order to address future issues of this kind:

I think we should define a maximum time span (either from the start or from the end of a Dshot frame being received) than an ESC is allowed to take to process the command or update, until it has to be ready to receive the next command / update.

This is so far missing in the Dshot specification, and once we've got it defined, will enable the flight controller to implement tight limits on how fast consecutive commands and updates are sent.

Thoughts?

etracer65, 2018-06-13:
Also related... @ronlix @sskaug  Can you define the maximum amount of time the current firmwares will block during beacon commands?  Currently we just chose an arbitrary 2 second guard period without having any actual information so that may be overdoing it quite a bit.  2 seconds doesn't seem like much but it's a big perceived delay to the pilot while they're trying to arm (and possibly having to hold a momentaty pre-arm switch).

flyduino, 2018-06-13:
on the kiss esc's, the beeps was the only blocking thing that can happen when the motor dont runs. 
so with the next beta FW's it should read all signals that come in. only for save and exit command a delay of maybe 1second to give it time to reeboot should be good. but with the new beeps it eaven reads the input signals during startup beep or voice.

sskaug, 2018-06-13:
Well, for BLHeli_S and BLHeli_32, interrupts are off and inputs signals are not read during beeps. Altering this will mean pretty significant changes to the codes.

But 2 seconds is most certainly more than required delay after a command. I can take a look at what delays would be required.

etracer65, 2018-06-13:
@ronlix Just to be clear, are you saying 1 second should be safe to send the next command after a beacon on the current firmware?

@sskaug I think we need to know what we can and can't do from a command sending perspective.  So knowing the restrictions for different types of commands would be very helpful.  We'll probably end up with just a generic short delay for most commands and a longer one for beacons or saves.

sskaug, 2018-06-13:
Ok, that makes sense. I'll look into what the delays are and report back.

ronlix, 2018-06-13:
@etracer65 no, just after the dshot save and reset command. else there are no delays needed for the kiss escs.

regards

felix

mikeller, 2018-06-14:
@ronlix: So except for the save and reset command, it is save to send a second, different command 1 us after the first one? Should we define a safe minimum interval (even if it's just to make sure the Dshot frame is detected properly on the line)?

ronlix, 2018-06-14:
@mikeller, Well the dshot frame takes more then 1us ;) there need to be at least one bit time between the the dshot frames to be able to read them. But you can fire dshot commands in the same speeds as you do with throttle signals. .. tested with up to ~40khz.

sskaug, 2018-06-14:
I have updated the BLHeli_S and BLHeli_32 spec docs with required wait times:
https://github.com/bitdump/BLHeli/blob/master/BLHeli_S%20SiLabs/Dshotprog%20spec%20BLHeli_S.txt
https://github.com/bitdump/BLHeli/blob/master/BLHeli_32%20ARM/Dshotprog%20spec.txt

mikeller, 2018-06-14:
@ronlix, @sskaug: Thanks. We'll look into adding these wait times.

mikeller, 2018-07-14:
Implemented in #6079.

closed 2018-07-14 by mikeller