betaflight/betaflight #6109: BF 3.4.0 - HOTT Telemetry not working

Nexus1212, 2018-06-13:
Hello,

the Hott Telemetrie does not work with BF 3.4.0 and 3.3.0.

I'm using a OMNIBUSF4SD Target (Omnibus F4 V3 Pro).

With BF 3.2.5 is working!

I read about the problem in another issue: 
https://github.com/betaflight/betaflight/issues/3145

Is the fix in the current version or is that a new bug?

Thank you!

Scavanger, 2018-06-14:
Similar Problem here. 
Dont't work for me since 3.3.2, But in 3.3.0 HoTT Telemetrie is working.

Nexus1212, 2018-06-14:
You are right! I checked the version again. The problem is since 3.3.2.

I looked on the output of the telemetrie port. There goes nothing out since the version 3.3.2

mikeller, 2018-06-16:
@Nexus1212, @Scavanger: Can you please test to see if HoTT telemetry works in 3.4.0-RC1, and provide feedback here?

Scavanger, 2018-06-18:
Sorry, no HOTT Telemetry on 3.4.0-RC2 too.

Nexus1212, 2018-06-18:
Hello, I tested 3.4.0-RC1 and  3.4.0-RC2 today. On both versions hott telemetry does not work.  

Also I logged with a arduino uno the UART1. There is nothing. The receiver sends the requests 80 80,  80 8E and so on but the flightcontroller does not answer. Bevor version 3.3.2 the flight controller answer normal 80 8E (EAM Modul) .

My settings are:
UART1 Hott telemetry
UART3 receiver SUMD
UART6 IRC Tramp

mikeller, 2018-06-18:
This is odd. I had a look at the changes between 3.3.1 and 3.3.2, and there is nothing even related to telemetry.

I don't have any HoTT gear, so I can't test. Can one of you guys do a bisect to find out what change exactly caused the breakage?

Nexus1212, 2018-06-18:
I want to help but my it skills are bad. What is to do for a bisect?

I have seen "ziegeOne" is working on a hott telemetry text mode. He is a expert for hott. Maybe he can take a look on the code?

mikeller, 2018-06-18:
Yes, @ziegeOne might be able to help here.

Scavanger, 2018-06-19:
I think the bug is not in the HoTT Telemetrie code, i did some debugging and  serialRxBytesWaiting(hottPort) (hott.c, hottCheckSerialData(..) ) always returns 0 since 3.3.2, and the buffer is empty. 
Are changes in the UART since 3.3.2, so it is incompatible with HoTT?

Scavanger, 2018-06-19:
I've found it!
@Ziege-One reported HoTT telemetry works for him on the same Target on UART6.
@Nexus1212 and I used UART1. The only thing that was changed between 3.3.1 and 3.3.2 was this: https://github.com/betaflight/betaflight/pull/5675/ This fix break HoTT telemetry on UART1, I revert all changes and HoTT Telemetry now works now in 3.4 RC2!

mikeller, 2018-06-19:
@Scavanger: This is very odd. The change in #5675 only affected how the bidirectional setting `tlm_halfduplex` works for UART1. This opens a couple of questions:
- what pin of UART1 is your HoTT telemetry wire connected to (RX / TX)?
- what is your setting for `tlm_halfduplex`?

Scavanger, 2018-06-19:
@mikeller 

- I use PA10 (RX) directly and PA9 (TX) connected via Diode.
- `tlm_halfduplex ` is `ON ` (Standard value)

mikeller, 2018-06-19:
@Scavanger: This is a weird setup that you are running: `tlm_halfduplex` instructs the MCU to expect TX and RX data flow to go through the TX pin, which means that your setup won't work on an UART that properly supports it (because of the diode). If you connected your telemetry wire directly to the TX pin (and not at all to the RX pin), and left `tlm_halfduplex` on, or alternatively kept your wiring, and set `tlm_halfduplex` off, it should work in 3.3.2 and newer.

Scavanger, 2018-06-19:
@mikeller:
This is the standard setup. I only configure (in Confurator) UART1 to HoTT telemetry and, according to the documentation, connect Rx/TX via Diode., 
I will try it. Perhaps this should changed in the documentaion.

mikeller, 2018-06-19:
@Scavanger: It looks like the documentation was last updated before the option to use  a single pin for bidirectional communication was introduced in the firmware. Where is the documentation?

Scavanger, 2018-06-19:
@mikeller:
Here: https://github.com/betaflight/betaflight/blob/master/docs/Telemetry.md an in the header of hott.c

mikeller, 2018-06-19:
@Scavanger: Thanks. Can you please confirm if connecting directly to TX (only) and setting `tlm_halfduplex = on` is working? If so, the documentation can be updated accordingly. (Or open a pull request for the update yourself. :wink:)

Scavanger, 2018-06-19:
Unfortunately no, but with `tlm_halfduplex = off` and diode it works. 
I'm in contact with @Ziege-One, for him it works when connecting directly to TX.

mikeller, 2018-06-19:
@Scavanger: Just to make sure: What version of the firmware are you testing with? There was a bug prior to 3.3.2 that affected how `tlm_halfduplex` works on UART1.

Scavanger, 2018-06-19:
@mikeller:
Ok, version <= 3.3.1 HoTT is working "from scratch" with diode, just set port to "HoTT Telemetry" and enable `feature telemetry`.
**For me**, on version >3.3.1 (incl. 3.4 RC2) HoTT is working ONLY with diode and tlm_halfduplex = off.
@Ziege-One, for him HoTT Telemetry (on the same target, OMNIBUSF4SD !) runs at least on BF 3.3.2 with telemetry connected only to RX.

Ziege-One, 2018-06-19:
@mikeller: 3.3.3 and 3.4.0 RC2 works with OMNIBUSF4SD on UART6 with tlm_halfduplex = ON connected to TX6
[image]

Nexus1212, 2018-06-19:
If I'm configure "tlm_halfduplex = off" HOTT telemetry works on 3.4.0 RC2! I'm using also a diode.

Nexus1212, 2018-06-19:
AND: 
If I connect the telemetry port of the receiver without a diode to TX1 and "tlm_halfduplex = on" it wirks also 👍 I don't need longer a diode!!! Good work :)

Tested on 3.4.0 RC2 with GR-12l (with new firmware)

Nexus1212, 2018-06-19:
@Scavanger Did you test with a connection to TX1 or RX1?

mikeller, 2018-06-20:
Ok, this is odd. But in general, bidirectional serial ports can be sensitive to the right pull up / pull down resistors being used (and that could even vary with different RX hardware). Adding the note about single wire being supported to the documentation was a good move, and I suspect when looking at the above reports that we won't be able to come up with a guide on how to make all the different combinations work.

@Nexus1212: The change that made HoTT stop working on UART1 the way it did before 3.3.2 was needed to make UART1 behave the same as other UARTs, so it will not be reverted. But I think we've shown ways to make it work by reconfiguring. Can this be closed?

Nexus1212, 2018-06-20:
Yes, I think the issue can be closed. But I think also it is very important to change the dokumentation. The diode is not longer needed to communicate with hott telemetry. If anyone use the diode, it will not work because half dublex i the default now.

Could you answer my last question please:
Now is the RX1 free. Is it possible to use the RX1 to receive sumd from the receiver and TX1 to communitcate to telemetry of the receiver? In this case can I use UART3 for another connection!

mikeller, 2018-06-20:
@Nexus1212: No, that is not possible, at least not in this simple way, since RX1 is still used to receive the HoTT data - it's just piped through the TX1 pin.
There is a way around this that works on some boards: It is possible to use a hardware UART in bidirectional configuration by connecting TX and RX (or either TX or RX for unidirectional protocols) to the TX pin, and then using the free RX pin for a SOFTSERIAL. Typically, you'd connect SUMD to the hardware UART / TX pin (enable `serialrx_bidir` for this), and then assign a SOFTSERIAL to the freed up RX pin, and connect telemetry to this (since there is a slightly larger chance of the SOFTSERIAL dropping out, and losing telemetry is a lot less serious than losing RX).

Nexus1212, 2018-06-21:
Thank you very much, I want to try it out :)

closed 2018-06-24 by Nexus1212