betaflight/betaflight #6111: 3.4 RC1 Dys F4PRO Resource command broken

xermalk, 2018-06-13:
DYSF4PRO (DYS4) 3.4.0 Jun 10 2018 / 02:25:09 (a5ba01666)

using resource to remap motor outputs isnt working.

I can remap Motor 3 and 4 just fine. But 

resource MOTOR 1 B00
resource MOTOR 2 B01

and 

resource MOTOR 1 B01
resource MOTOR 2 B00

Does not change what motor spins using the configurator motor tab. 
Remapping in 3.3 and Butterflight works just fine.

mikeller, 2018-06-14:
```
resource MOTOR 1 B01
resource MOTOR 2 B00
save
```

```
# resource list
Currently active IO resource assignments:
(reboot to update)
--------------------
[...]
B00: MOTOR 2
B01: MOTOR 1
```

Works just fine for me. What's your `resource list` say after modifying the resources and saving / rebooting?

eitama, 2018-07-07:
@mikeller 

Just an FYI, just installed BF 3.4 (RC5) on DYS F4 Pro (V1)

And remapped Motor 3 to LED_STRIP without any issues.

mikeller, 2018-07-07:
@xermalk: Can you please confirm that this is fixed, and close this?

xermalk, 2018-07-15:
Its closed, after a reflash it started saving resource changes.

closed 2018-07-15 by xermalk