betaflight/betaflight #6112: rc_advance

Plasmatree, 2018-06-13:
Hello everybody,

this is not an issue, but more a proposal of an idea, so I’m not sure if this is the right spot. I can’t get into modifying and testing of the source code myself, since there are not enough hours in a day, but maybe someone else would like to implement and try it.
The whole discussion about RC-interpolation and especially UAV-techs video on delay got me thinking:
If we add something like a derivative term to the RC-signal, we can compensate for lag induced by the interpolation/low sampling rate. It’s based on the assumption that if we start changing the RC-command, we will probably further do so for some cycles. 
* It will induce slight overshoot at flips that has to be limited -> A limiting max-value is required (`maxadv`)
* single parameter (`alpha`) scales strength of the effect (0=off, 0.5=decent, 1=no delay, 2=faster than your fingers)
* jittery RC-input (-level) will be amplified 

I did a quick test in python on a log I found somewhere (that had raw rc_data). Should be readable as pseudo code/easy to transfer to C… Of course in a more efficient/less ugly way, but I hope you get the point. 

The `code` is messed up in the `code` thing so here it is: 
[rc_advance.txt](https://github.com/betaflight/betaflight/files/2099261/rc_advance.txt)
and here as image for reading:
[image]
Here are some test results, done on the test log (alpha=0.5):
[image]
[image]
Overshoot (limited by `maxadv`)
[image]
Alpha=1 ('zero' delay but also some overshoot):
[image]

With alpha=0.5 it reduces the delay by ~50% (~5ms in this case).
With alpha=1 it completely eliminates delay in some spots, but causes stronger artefacts/roughness

**It's similar to setpoint_weight/smart_feedforward, but is placed prior to rc_smoothing to counteract the delay.**

What do you think? Useful?

Flo

halfcab1981, 2018-06-22:
I am no programmer but this looks smart to me. If only because the smoothed advance trace seems to track the raw rc signal very nicely, which is what we want isn't it?

DaveFPV, 2018-07-12:
No devs reaction on this idea? This should be a PR to get more visibility.

closed 2018-08-19 by stale[bot]