betaflight/betaflight #6115: RevoltOSD Target, 32khz (8khz/8khz) mode ON, gyro won't calibrate, 32khz mode OFF, gyro calibrates fine

mmakam2, 2018-06-13:
There is zero noise on the gyro, this is a software compatibility issue or bug not hardware. with 32khz mode on, setting 8khz/8khz, gyro won't calibrate, with 8khz/8khz mode with 32khz mode OFF, gyro calibrates with no issue. This is an issue specific to the 32khz mode.

If anyone has any fix for this, I would love to test it out..

mmakam2, 2018-06-13:
[diff.txt](https://github.com/betaflight/betaflight/files/2100341/diff.txt)

etracer65, 2018-06-14:
When the gyro runs in 32KHz mode it results in significantly more noise which can exceed the calibration limit.  It's normal to need to increase this limit when trying to use 32KHz.

Try increasing `gyro_calib_noise_limit` gradually in the CLI until you get consistent calibration.

mmakam2, 2018-06-14:
I did this, went to 200 with no success

mmakam2, 2018-06-14:
I also tried caps on xt60 and 5v leads, no luck

etracer65, 2018-06-14:
Presuming that you're not moving the quad during the calibration, well then I'm sorry to say it sounds like you have a hardware problem.  I just tested with a 32KHz gyro on BF3.4 and no matter the gyro/pid rate it calibrates fine.  Doesn't matter if I use 8K, 16K or 32K.

mmakam2, 2018-06-14:
Which FC?