betaflight/betaflight #6118: Date from GPS is 1 day behind.

atomiclama, 2018-06-14:
The date from GPS is 1 day behind. The output from status command is from today 14/06/2018.
If I use GPS pass through and hook up u-centre the date is correct.

UTC time is correct.

This was correct in the past can't place when it changed.

`Entering CLI Mode, type 'exit' to return, or 'help'

# status
System Uptime: 204 seconds
Current Time: 2018-06-13T06:20:35.574+00:00
Voltage: 0 * 0.1V (0S battery - NOT PRESENT)
CPU Clock=168MHz, Vref=3.28V, Core temp=43degC, GYRO=MPU6000, ACC=MPU6000, BARO=BMP280
SD card: Manufacturer 0x2, 7639040kB, 09/2012, v1.1, 'SA08G'
Filesystem: Ready
Stack size: 2048, Stack address: 0x1000fff0
I2C Errors: 0, config size: 2543, max available config: 16384
CPU:8%, cycle time: 126, GYRO rate: 7936, RX rate: 50, System rate: 10
Arming disable flags: RXLOSS CLI MSP
`

[image]

```
Entering CLI Mode, type 'exit' to return, or 'help'

# diff

# version
# Betaflight / OMNIBUSF4SD (OBSD) 3.4.0 Jun 10 2018 / 02:42:06 (a5ba01666) MSP API: 1.39

board_name OMNIBUSF4SD
manufacturer_id

# name
name tikka

# resources

# mixer
mixer CUSTOM
mmix 0  1.000  0.667 -0.859  1.000
mmix 1  1.000 -1.000  0.065  1.000
mmix 2  1.000 -0.474  0.793 -1.000
mmix 3  1.000  0.474  0.793  1.000
mmix 4  1.000  1.000  0.065 -1.000
mmix 5  1.000 -0.667 -0.859 -1.000

# servo

# servo mix

# feature
feature -RX_PARALLEL_PWM
feature RX_SERIAL
feature GPS
feature AIRMODE

# beeper

# beacon

# map

# serial
serial 0 64 115200 57600 0 115200
serial 2 8192 115200 57600 0 115200
serial 5 2 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 1 1625 2100 0
aux 1 13 0 1750 2100 0

# adjrange

# rxrange

# vtx

# rxfail

# master
set gyro_lowpass_hz = 140
set acc_calibration = 8,-117,311
set mag_hardware = NONE
set min_check = 1020
set rssi_channel = 14
set rc_interp = MANUAL
set rc_interp_ch = RPYT
set rc_interp_int = 8
set rc_smoothing_type = FILTER
set serialrx_provider = IBUS
set blackbox_p_ratio = 64
set blackbox_record_acc = OFF
set min_throttle = 1159
set max_throttle = 1920
set motor_pwm_protocol = ONESHOT42
set align_board_pitch = 180
set align_board_yaw = 180
set current_meter = NONE
set small_angle = 180
set gps_provider = UBLOX
set gps_sbas_mode = EGNOS
set gps_auto_baud = ON
set deadband = 4
set yaw_deadband = 6
set pid_process_denom = 1
set osd_warn_esc_fail = OFF
set osd_warn_core_temp = OFF
set osd_alt_alarm = 150
set osd_tim1 = 2561
set osd_vbat_pos = 2123
set osd_rssi_pos = 2130
set osd_tim_1_pos = 2402
set osd_gps_speed_pos = 2410
set osd_gps_lon_pos = 2448
set osd_gps_lat_pos = 2434
set osd_gps_sats_pos = 2115
set osd_home_dir_pos = 2382
set osd_home_dist_pos = 2415
set osd_altitude_pos = 2422
set osd_pidrate_profile_pos = 2137
set osd_warnings_pos = 2089
set osd_stat_tim_1 = ON
set osd_stat_tim_2 = OFF
set osd_stat_min_rssi = OFF
set osd_stat_max_curr = OFF
set osd_stat_used_mah = OFF
set osd_stat_max_alt = ON
set debug_mode = GYRO_SCALED
set vtx_band = 0
set vtx_channel = 3
set vtx_power = 3
set vtx_freq = 5790
set vcd_video_system = NTSC

# profile
profile 0

set dterm_lowpass_hz = 160
set dterm_lowpass2_hz = 0
set dterm_notch_cutoff = 0
set iterm_relax = RPY
set p_pitch = 53
set p_roll = 48

# rateprofile
rateprofile 0

set roll_expo = 20
set pitch_expo = 20
set yaw_expo = 20
set tpa_breakpoint = 2000
```

mikeller, 2018-06-14:
How very weird. We changed some of the time conversion / stringification functions, could be that we introduced an off by 1.

tavdog, 2018-06-14:
cli output uses : `if (rtcGetDateTime(&dt)) {
        dateTimeFormatLocal(buf, &dt);
        cliPrintLinef("Current Time: %s", buf);
    }`  

could it possibly be a local timezone issue ?

I checked the math on line 806 of gps.c and it seems correct.
[image]

tavdog, 2018-06-14:
Just to be sure: the time is correct but the day is not ?  Just want to make sure this isn't some stale gps data being used before it's gets a good sat lock.  Since the real time clock seems to only get set once at startup, I want to make sure that first setting is based on new data and not stale data.

atomiclama, 2018-06-15:
Confirmed time correct, day behind.

jeffystockes, 2018-06-15:
Hello and thanks for everything. Sorry for my verry bad English. For information, I have the same problem with a Matek F722 & GPS Ublox neo m8n. The altitude changes a lot : Yesterday i have 1600 m. Today it's 400m, for a real altitude arround 130m (more than 4 satellites at the receiver - Ublox protocol)... Another information (may be it's a different problem) : I have gps data on the osd, but i don't have them on my taranis x9d+ (opentx 2.2.2), as if they did not pass through the smart port of the RX (X8R FrSky).

digitalentity, 2018-06-15:
Since we share the RTC code with INAV - see https://github.com/iNavFlight/inav/pull/3307 for the fix