betaflight/betaflight #6119: Feature Request: ABX test

karmaraver, 2018-06-14:
One problem with tweaking and testing PIDs, filters, rates, etc, on a quad is whether the effects you're seeing and feeling in flight are actually happening or if they're just confirmation bias. ABX testing is a way to figure out statistically whether or not a particular change is significant enough to be perceived.

From [wikipedia](https://en.wikipedia.org/wiki/ABX_test): 
> An ABX test is a method of comparing two choices of sensory stimuli to identify detectable differences between them. A subject is presented with two known samples (sample A, the first reference, and sample B, the second reference) followed by one unknown sample X that is randomly selected from either A or B. The subject is then required to identify X as either A or B. If X cannot be identified reliably with a low p-value in a predetermined number of trials, then the null hypothesis cannot be rejected and it cannot be proven that there is a perceptible difference between A and B.

Is it possible to include such a feature in betaflight? I'm picturing a procedure activated through the OSD where the first arm is profile A, the second arm is profile B, and the third arm is profile X. After the last disarm the OSD asks for your guess, and the process repeats until you have enough data or you're bored. You then exit the procedure to see your results and whether or not the difference between A and B was statistically significant. If you have a three way switch you could also switch between A, B, and X in flight and guess what X is on disarm.

This would help not only individual pilots in determining whether a particular setting is working for them, but would also give developers another tool to determine if a particular feature is worth the effort to include or maintain.

Thoughts?

DieHertz, 2018-06-14:
It's definitely possible and a good idea, I don't see a way of getting untainted data from users at this point though.

ProfessorStoke, 2018-06-14:
Untainted? Do you mean untainted as in poorly executed tests? Falsified reports?

DieHertz, 2018-06-14:
Falsified reports, custom builds that fake the data.
Such tests require an infrastructure which guarantees mutability and authority of data.
With embedded devices it's hard to guarantee the code or the result wasn't modified either intentionally or unintentionally

ProfessorStoke, 2018-06-14:
I take it that falsified reports are an actual, concrete problem for you as a developer for things of that nature to be such a concern. Trust is a difficult, if not impossible, problem to tackle.

DieHertz, 2018-06-14:
It's more like users are unlikely to send you logs where they haven't guessed the outcome correctly, unless the process is automated.

wind0r, 2018-06-14:
This may easily could be added to the confiugrator. Just add 2 different settings and the Configurator will choose for you. This could be greate to find compare and find new rate settings

ProfessorStoke, 2018-06-14:
You're concerned that users would do a number of tests and cherry pick the ones that confirm their bias? Or just that, in the aggregate, you'd see a selection bias in the reports you see?

mikeller, 2018-06-16:
This is a duplicate of #5317. Closing this to keep the discussion in one place.

closed 2018-06-16 by mikeller