betaflight/betaflight #6120: False detects of the QMC5883L mag sensor

MJ666, 2018-06-14:
With actual master including RC1 the QMC5883L mag sensor is sometimes detected if there is no mag present or even if a different mag is installed. I can see this behavior on a couple of different FC boards. The issue i see here is the QMC mag is using 0xFF as the chipid at register 0x0D. You will read also 0xFF on an no connected sensor address in most cases. This will cause the false detections. I.e. all AFNG F7 boards will show a QMC mag sensor now. Some of them have an SPI connected AK8963 mag. If you change the mag_hardware manually it will work but automatic detection fails. Using 0xFF ans an chipid is a very bad choice from the sensor vendor. Not sure how we can effetely work around this? One option is reading another register to confirm there is really a sensor presents. Not very nice and error free better as it is now. Propose the following PR to fix it.

ledvinap, 2018-06-14:
Slave must ACK address, testing this ACK should handle no slave case

MJ666, 2018-06-14:
@ledvinap in this case the sensor is configured for SPI bus and in my knowledge at least the SPI bus always acknowledges the request. There will be also the possibility to block QMC5883L detection if SPI bus is used since this sensor is I2C only.

jflyper, 2018-06-15:
@MJ666 There is no way to auto reject non-existent SPI device that is designed to return 0xff as an indication of existence. Users must explicitly specify the location of the device to be used, or disable the mag if false detection is not desired.

The auto-detection will go away and almost everything will be configured explicitly in 4.0. It's just beginning already now with the mag.

ledvinap, 2018-06-15:
@MJ666 : Checking for I2C first should be quite clean and it will also avoid detection traffic on spi.
Hopefully nobody designs SPI device with 0xff/0x00 signature...

MJ666, 2018-06-16:
@jflyper I think dropping autodetection features for sensors is not a proper step for the universal target. First we expect to have all drivers included in the firmware and second we should stll allow to connect additional sensors to be used on the external bus. Would we expect this to be done always wit manual configuration or even modify the bas FC config. Some configurator support could be helpfull.helpful .

@ledvinap the AFNG F7 board is expecting an SPI sensor there. Some logic in the code is setting this based on definitions in the target.h file. This is not to easy to influence. May be it is still a good plan to disable SPI in  the QMC5883L driver detection completely. Unfortunately I have no boards with an QMC to test. The proposed workaround in #6121 is fixing my problems but need to be tested for the QMC.

MJ666, 2018-07-16:
Issue is resolved.

closed 2018-07-16 by MJ666