betaflight/betaflight #6136: Throttle twitches on F3 with analog ESC protocols

andrejpodzimek, 2018-06-17:
I have random throttle twitches with Betaflight 3.3. When the drone si idle, propellers wobble and sometimes spin a revolution or two, all by themselves. Sometimes they do so synchronously, sometimes it appears to affect just a subset of motors. The twitches occur roughly once in 5 seconds, but appear to be quite random otherwise.

Of course I've calibrated the ESCs and set my throttle values so that the ESCs see them somewhere between 1000 and 2000 microseconds, definitely not outside that range.

When the drone is armed, there's an equivalent of the twitches, so to say, in the form of abrupt short wobbles, as if a few ESCs tried to brake for a very short moment and then kept on working normally.

I've observed this on two completely different builds, a hexacopter with a [BeeRotor F3](http://beerotor.com/product/beerotorf3-aio-flight-controller-with-osd/) and a quadcopter with a [KingKong Micro F3 Plus](https://www.banggood.com/Kingkong-Micro-F3-Plus-6DOF10DOF-Flight-Controller-for-FPV-Racer-p-1093911.html).

On the hexacopter I could only reproduce the problem with MULTISHOT and ONESHOT42; it doesn't seem to occur with ONESHOT125. On the quadcopter it occurs all the time with ONESHOT125 (which is the fastest protocol its ESCs can take).

I can't tell whether this occurs in flight, either it doesn't or the higher power / RPM / noise obscures the problem.

jflyper, 2018-06-18:
Saw similar symptom last week, when I was testing something, probably hexa related.
However, I can't reproduce it with Omnibus F3 AIO (gen 1), flashed 3.3.3 and 3.4-RC2, default board config, tested with motor test slider in OS125, OS42 and MS.

@andrejpodzimek I dag out a BRF3 from my parts bin. You can provide a non-working `diff`.

andrejpodzimek, 2018-07-19:
> @andrejpodzimek: Can you please check if this problem persists with 3.4-RC1?

@mikeller  Sorry for the delay. Tested with 3.4 release now. Yes, it twitches. All motors twitch.

> @andrejpodzimek I dag out a BRF3 from my parts bin. You can provide a non-working diff.

@jflyper  Again, sorry for the delay. Here's a heavily twitching diff:

```shell
# diff all
# Betaflight / SPRACINGF3 (SRF3) 3.4.0 Jul  8 2018 / 04:32:14 (cce8a59be) MSP API: 1.39

board_name SPRACINGF3
manufacturer_id 
mcu_id 001c00234134570a20383837
defaults nosave
name SRD240
resource MOTOR 1 A02
resource MOTOR 2 A03
resource MOTOR 7 NONE
resource MOTOR 8 NONE
mixer HEX6X
mmix 0  1.000 -0.629  0.892  1.000
mmix 1  1.000 -0.629 -0.892  1.000
mmix 2  1.000  0.629  0.892 -1.000
mmix 3  1.000  0.629 -0.892 -1.000
mmix 4  1.000 -1.000  0.000 -1.000
mmix 5  1.000  1.000  0.000  1.000

feature -RX_PPM
feature -RSSI_ADC
feature RX_SERIAL
feature LED_STRIP
feature ESC_SENSOR
serial 1 1024 115200 57600 0 115200
serial 2 64 115200 57600 0 115200
led 0 4,7:SW:ATOIW:0
led 1 5,7:SW:ATOIW:0
led 2 6,7:SW:ATOIW:0
led 3 7,7:S:ATOIW:2
led 4 8,7:S:ATOIW:0
led 5 9,7:ES:ATOIW:0
led 6 10,7:ES:ATOIW:0
led 7 11,7:ES:ATOIW:0
led 8 10,0:NW:CIW:2
led 9 11,0:NW:CIW:2
led 10 12,0:NW:CIW:2
led 11 13,0:NW:CIW:2
led 12 14,0:NW:CIW:2
led 13 15,0:NW:CIW:2
led 14 5,0::CIW:6
led 15 4,0:NE:CIW:6
led 16 3,0:NE:CIW:6
led 17 2,0:NE:CIW:6
led 18 1,0:NE:CIW:6
led 19 0,0:NE:CIW:6
aux 0 0 0 1700 2100 0
aux 1 1 1 900 1625 0
aux 2 2 1 1625 1875 0
aux 3 3 1 900 1375 0
aux 4 13 3 1300 2100 0
aux 5 28 4 1700 2100 0
adjrange 0 0 2 900 2100 12 2 0 0
rxfail 5 s 1000
rxfail 6 s 1000
rxfail 7 s 2000
set mag_hardware = NONE
set baro_cf_vel = 0
set max_check = 1950
set max_aux_channels = 6
set serialrx_provider = FPORT
set serialrx_inverted = ON
set serialrx_halfduplex = ON
set blackbox_p_ratio = 0
set blackbox_device = NONE
set min_throttle = 1060
set max_throttle = 1970
set dshot_idle_value = 400
set dshot_burst = ON
set motor_pwm_protocol = MULTISHOT
set motor_pwm_rate = 400
set current_meter = ESC
set battery_meter = ESC
set vbat_scale = 208
set pid_process_denom = 1
profile 0

set dterm_notch_cutoff = 0
set vbat_pid_gain = ON
set anti_gravity_gain = 2000
set dterm_setpoint_weight = 100
set d_level = 80
profile 1

set dterm_notch_cutoff = 0
set vbat_pid_gain = ON
set anti_gravity_gain = 2000
set dterm_setpoint_weight = 100
set d_level = 80
profile 2

set dterm_notch_cutoff = 0
set vbat_pid_gain = ON
set anti_gravity_gain = 2000
set dterm_setpoint_weight = 100
set yaw_lowpass_hz = 100
set p_pitch = 65
set i_pitch = 30
set d_pitch = 22
set p_roll = 55
set i_roll = 30
set d_roll = 19
set p_yaw = 80
set i_yaw = 30
set p_level = 35
set d_level = 80
profile 2
rateprofile 0

rateprofile 1

set roll_srate = 75
set pitch_srate = 75
set yaw_srate = 75
rateprofile 2

set roll_srate = 80
set pitch_srate = 80
set yaw_srate = 80
rateprofile 3

rateprofile 4

rateprofile 5

rateprofile 0
save
```

It's actually **worse** than with 3.3. If I'm not mistaken, all motors twitch now, all synchronously, but in seemingly random intervals of 1-5 seconds (rough guess). At some point some of the ESCs even rebooted, which may have been caused by the twitches.

andrejpodzimek, 2018-07-19:
Sadly, I also have a severe and dangerous issue also with DShot 600 on 3.4, which affects only motor No. 4. (There were *some* issues with motor No. 4 in 3.3, but not crash-level ones.) I'm mentioning this here because of my vague idea that it might be a digital DShot counterpart to the analog multishot twitches. I'm quite likely wrong, as usual. Please let me know if you want me to open a separate bug for this. The `diff all` would be the same, just `DSHOT600` instead of `MULTISHOT`.

Here's the issue: Sometimes motor No. 4 suddenly loses power for a split-second and sometimes the telemetry reading from that ESC gets bogus for a few seconds. (For example, voltage on that ESC reads 9.something V while all other ESCs show 16.something V, as shown on the Power and Battery tab.) The bogus telemetry data also occurs in-flight &mdash; it sometimes raises the voltage alarm for a while at random. The two events (power loss and bogus telemetry data) mostly do **not** coincide. Both phenomena seem to occur at random.

Things I tried:

1. Swapped [the ESC](https://www.racerstar.com/Racerstar-Tattoo+-42A-BLheli_32-72MHz-GD32F150-2-5S-ESC-Dshot1200-Ready-Current-Sensor-LED-Telemetry-p-201.html) for a completely new one yet again &mdash; the first swap on motor No. 4 was done [after this motor starting problem](https://github.com/betaflight/betaflight/issues/6137#issuecomment-397911247).
1. Double-checked the motor itself with a coil meter (looks OK).
1. Configured automatic motor timing and strong demag compensation in BLHeli_32.

But it's still badly unreliable. Also, whenever the sudden power drops occur in hover,  motor No. 4 tends to be hot after the subsequent "landing".

All these issues either appeared or got way worse in 3.4; with 3.3 this same build was able to fly for a few batteries. I'll try `DSHOT300` to see if it gets any better, but it's quite scary to test this. :fearful:

andrejpodzimek, 2018-07-19:
OK, `DSHOT300` is also bad. Two observations after ~5 minutes of testing:

1. No power drop so far. `DSHOT600` would have crashed a number of times already.
2. Telemetry reading twitches on ESC No. 4; their frequency stabilized on ~3 seconds. Basically the onboard voltage estimate drops suddenly from ~15&ndash;16 V down to ~10&ndash;11V and then recovers.

andrejpodzimek, 2018-07-19:
Back to 3.3.3 and `DSHOT600`. Impressions after a short flight:

1. 1 power drop on No. 4 thus far.
1. 0 voltage glitches, not even a single one; the readings were stable all the time.

Now that^^^ is the exact opposite of the [problem](https://github.com/betaflight/betaflight/issues/6136#issuecomment-406446153) above that 3.4 and `DSHOT300` has. :confused: :fearful:

I have 1 more of those ESCs, but &hellip; so many faulty new ESCs in a series seem just unlikely to me. Not sure if a third ESC swap would solve anything.

andrejpodzimek, 2018-07-20:
Sorry for all this frustrated multi-posting. Perhaps this issue should be only about the multishot twitches that affect ~all 6 motors. The DShot problem specific to motor No. 4 deserves a separate bug. Filed #6403.

andrejpodzimek, 2018-08-20:
This is still an issue.
However, if analog ESC protocols get deprecated and removed one day, then the problem may disappear all by itself.

closed 2018-09-07 by andrejpodzimek