betaflight/betaflight #6137: DShot doesn't work on BeeRotor F3 hexacopters

andrejpodzimek, 2018-06-17:
I'm using Betaflight 3.3. Motor No. 6 won't work with DShot on my [BeeRotor F3](http://beerotor.com/product/beerotorf3-aio-flight-controller-with-osd/) hexacopter. I thought these problems would disappear as #4020 got implemented. Or was it for F4 and F7 only? Here's what `dma` says:

```
DMA:
DMA1 Channel 1: MOTOR 3
DMA1 Channel 2: LED_STRIP
DMA1 Channel 3: MOTOR 1
DMA1 Channel 4: MOTOR 4
DMA1 Channel 5: MOTOR 5
DMA1 Channel 6: FREE
DMA1 Channel 7: MOTOR 2
DMA2 Channel 1: ADC
DMA2 Channel 2: FREE
DMA2 Channel 3: FREE
DMA2 Channel 4: FREE
DMA2 Channel 5: FREE
```

Indeed, motor #6 is not there. (But the motor runs fine under analog ESC protocols and BLHeli_32 can talk to  / flash that ESC.)

For BeeRotor F4 (not F3), [there's a know issue](https://github.com/betaflight/betaflight/wiki/dshot-esc-protocol#beerotor-f4----for-use-with-a-hexacopter-motor-6-needs-to-be-moved-to-the-led-pin-on-the-sc-connector) that requires one to remap motor No. 6 to the LED pin and disable LEDs.

However, I've just refurbished my hexacopter and added lots of fancy LEDs. So even if that^^^ same recipe worked for BeeRotor F3, it wouldn't be an option for me. :disappointed: Are there any other ways to get DShot working? Are the DMAR timers available on an F3?

jflyper, 2018-06-17:
Do you have any unused output signals?

jflyper, 2018-06-17:
I found a pin map sheet.

If you have M7 and M8 unused, you can assign motors 1 and 2 to these two outputs and then do `set dshot_burst = ON`.

andrejpodzimek, 2018-06-17:
Thanks! Yes, it works! I resoldered the two motors etc. and took off on DShot 600 just fine.

[One annoying ESC](https://www.rcgroups.com/forums/showthread.php?2063090-Motor-stuttering-Won-t-run-Read-before-asking/page7#post39811443) is the only thing that spoils the pure joy of this. It would be great to have some diagnostic data in the ESC telemetry, such as *this ESC is [[censored]], replace it*.

BTW, considering #4626, is it safe / recommended to use DShot 1200 on anything other than F7? I'd like to test it with the F3 in flight, but I've recently had [a very bad experience](https://www.rcgroups.com/forums/showthread.php?3066106-DSHOT-1200-issues/page1#post39802166) with another F3 and DShot 1200, which is how I learned about #4626. I've never seen any such issues with DShot 1200 on my F7s.

mikeller, 2018-06-17:
@andrejpodzimek:

> One annoying ESC is the only thing that spoils the pure joy of this. It would be great to have some diagnostic data in the ESC telemetry, such as this ESC is [[censored]], replace it.

The problem here is that in most cases, the ESC itself won't detect that it's faulty. For out-of-spec condidions for RPM / temperature, we have `osd_warn_esc_fail`.

andrejpodzimek, 2018-06-21:
BTW, I switched to Proshot1000, out of curiosity, and something weird happened: Only 2 motors worked, 3 and 4. They idled quite normally. The other motors didn’t start at all. DShot600 works just fine and all motors spin. (DShot1200 also works, but I’m not sure whether it’s safe to use it on an F3.) So it looks like there’s also an issue with Proshot.

DieHertz, 2018-06-21:
One thing to note is TIM_UP (burst) DSHOT wasn't meant to solve all problems on existing boards.
Besides your DMA list shows you had burst disabled

andrejpodzimek, 2018-06-21:
I had `dshot_burst` enabled, as recommended by @jflyper above. I also use outputs 7 and 8 for motors 1 and 2, or else only 5 motors would work with DShot. But if Proshot doesn’t use the fancy new timers, then that kind of explains why it doesn’t work. Not a big deal. I was just curious about that protocol after I read about its “pulse width” data encoding.

mikeller, 2018-06-23:
@DieHertz: As far as I can tell, burst DMA should work with ProShot as well - the implementations for both protocols are generic enough.

andrejpodzimek, 2018-07-23:
I think this can be closed, because DShot 600 currently works on BeeRotor F3.
As for DShot 1200, that's still causing weird issues when connecting from BLHeli_32, suddenly there are bad frames reported by BLHeli_32 and sometimes even firmware flashing fails. Not sure if it's #4626 or not, but DShots faster than 600 probably won't work on BeeRotor F3 with hexacopters. Switching to DShot 600 makes BLHeli_32 ~great~ work again.

closed 2018-07-23 by andrejpodzimek

mikeller, 2018-07-24:
> Switching to DShot 600 makes BLHeli_32 ~great~ work again.

:rofl: