betaflight/betaflight #6138: Improvement - Provide option to disable beeper tones when in "crash flip" mode aka turtle mode

Saxin, 2018-06-17:
Please provide configuration option, possible for beeper cli command to be able to disable the default beeping in crash flip mode.

etracer65, 2018-06-17:
Not saying it won't get implemented, but note that the beeping was added as a safety feature.  The concern is that while in crash flip mode the quad may appear to be disarmed because the motors are not spinning.  We don't want somebody "trying to help" and pick up the crashed quad.

Saxin, 2018-06-17:
@etracer65 thanks, I fully understand.
I'm not proposing disabling or altering default behavior, only provide option to disable upon user decision, one can argue that not all quads are equipped with a buzzer.

Saxin, 2018-06-18:
@mikeller thanks will try my evening, closing for now.

closed 2018-06-18 by Saxin