betaflight/betaflight #6139: 3.4 RC2 - LED strip not working

TheIsotopes, 2018-06-17:
tested now 3.4 rc2 release and must realize that the led strip does not work anymore ... back to 3.4 rc1 and all is ok ... tested on matek f405

imho i think it came from #6131 ... @DieHertz pls can you check this

mikeller, 2018-06-17:
@TheIsotopes: Some context like hardware used / `diff` would be helpful.

TheIsotopes, 2018-06-17:
i'm using the TAIL LIGHT WS2812B from Matek 

diff output:
```
# Betaflight / MATEKF405 (MKF4) 3.4.0 Jun 17 2018 / 08:39:32 (2fe9d7990) MSP API: 1.39

# name
name Mark1

# feature
feature -SOFTSERIAL
feature -TELEMETRY
feature -OSD
feature LED_STRIP
feature AIRMODE

# map
map TAER1234

# led
led 0 9,2::CW:6
led 1 9,3::CW:6
led 2 9,4::CW:6
led 3 9,5::CW:6
led 4 6,5::CW:6
led 5 6,4::CW:6
led 6 6,3::CW:6
led 7 6,2::CW:6

# aux
aux 0 0 0 1600 2000 0
aux 1 13 1 1000 1200 0
aux 2 26 0 1600 1800 0
aux 3 31 0 1200 1400 0
aux 4 36 1 1800 2000 0

# master
set gyro_lowpass_hz = 120
set acc_hardware = NONE
set mag_hardware = NONE
set baro_hardware = NONE
set min_check = 1010
set max_check = 2000
set rc_smoothing_type = FILTER
set max_aux_channels = 2
set serialrx_provider = SUMD
set blackbox_device = SPIFLASH
set dshot_idle_value = 550
set motor_pwm_protocol = DSHOT300
set bat_capacity = 1550
set use_vbat_alerts = OFF
set use_cbat_alerts = ON
set cbat_alert_percent = 28
set pid_process_denom = 1

# profile
set dterm_lowpass2_hz = 250
set setpoint_relax_ratio = 50
set dterm_setpoint_weight = 80
set iterm_relax = RP
set iterm_relax_cutoff = 10
set throttle_boost = 8
set p_pitch = 58
set i_pitch = 65
set d_pitch = 35
set p_roll = 50
set i_roll = 60
set d_roll = 30
set i_yaw = 60

# rateprofile
set roll_rc_rate = 135
set pitch_rc_rate = 135
set yaw_rc_rate = 120
set roll_expo = 15
set pitch_expo = 15
set roll_srate = 75
set pitch_srate = 75
set tpa_rate = 30
set tpa_breakpoint = 1100
```

DieHertz, 2018-06-17:
I'll look into it when I get home.

TheIsotopes, 2018-06-17:
@DieHertz @mikeller when i revert #6131, the led strip works again

jflyper, 2018-06-17:
@TheIsotopes Try this
[betaflight_3.4.0_MATEKF405.hex.zip](https://github.com/betaflight/betaflight/files/2108979/betaflight_3.4.0_MATEKF405.hex.zip)

jflyper, 2018-06-17:
@TheIsotopes Apologies for our oversight; we were too excited last night 🤣

AirBreak69, 2018-06-17:
Can confirm: Introduced somewhere between build 931 and 936.
Most likely with build 936.

jflyper, 2018-06-17:
@AirBreak69 Thanks for testing!

PR #6141 (2018-06-17): F4 DMA buffer should not reside in CCM
F4 DMA buffer should not reside in CCM, as DMA engines do not have path to CCM.
Tested with Omnibus F4 AIO (gen1).

Fixes: #6139
closed 2018-06-17 by mikeller