betaflight/betaflight #6143: arming disabled msp...i cant get this off will someone help please

510Rascloud, 2018-06-17:
# If your issue looks like a hardware fault or a configuration problem please don't raise an issue here.

## Please consider using other user support options such as asking the manufacturer of the hardware you are using, RCGroups: https://rcgroups.com/forums/showthread.php?t=2464844, or Slack: https://slack.betaflight.tech or other user support forums & groups (e.g. facebook).

## If you believe there is an issue with the firmware itself please follow these steps:
1. Describe your problem;
2. Include ways to reproduce the problem;
3. Provide information about your flightcontroller and other components including how they are connected/wired;
4. Add the used configuration and firmware version;
5. Create a `diff` and post it here in a code block. Put ``` (three backticks) at the start and end of the `diff` block (instructions  on how to do a diff: https://oscarliang.com/use-diff-not-dump-betaflight/); and
6. Remove this Text :).

jflyper, 2018-06-17:
You are probably connected to the FC with BFC.

To force arming in this state, you have to enable motor test.

For further assistance, PLEASE READ AND FOLLOW the instruction.

closed 2018-06-17 by jflyper

510Rascloud, 2018-06-18:
Enable motor test is not turning my motors on with my receiver. I can only spin them with my mouse...My receiver is connected to Betaflight...
I’m getting an arming disabled. cli status says cli msp I cant get the msp Ive been reading and watching videos for 2 days...I don’t know what I’m missing...

mikeller, 2018-06-18:
@510Rascloud: So setting 'Enable Motor Test' to on and then arming with your TX does not work?

510Rascloud, 2018-06-18:
No this does not work.

510Rascloud, 2018-06-19:
How do you enable motor test!!!???

mikeller, 2018-06-19:
[image]

510Rascloud, 2018-06-19:
Yeah I’ve done this can you help me bro been working on this for stew weeks last couple days I keep getting the msp I don’t know how to fix this