betaflight/betaflight #6144: OMNINXT7 - uncommanded throttle burst upon launching :-O

RipperDrone, 2018-06-17:
Have the OMNINXTF7 tested with both FW3.3 and 3.4RC2, all settings are default: After arming, the quad sits there with idle (DSHOT600, 8/4kHz). When I increase throttle, the quad slowly speeds up rpm (still not high enough to take off the ground), but then all of a sudden all motors give an uncommanded throttle burst and make the quad shoot up by 3+ meters. I tried several times, quickly disarmed every time to avoid more severe impacts.

It seems throttle is not going to 100%, but to a fast ascend rate still.

Tried gyro BOTH as well, no change... :-/

Creative ideas what to look for?

RipperDrone, 2018-06-17:
Here is the diff: https: //drive.google.com/open?id=1atHNO2nmka_lzx6nZJx23fqbBcUIw52J

Saxin, 2018-06-17:
I think a video can be very helpful and a log also can help a lot

RipperDrone, 2018-06-17:
Here is the rpm runaway event: 
[image]

RipperDrone, 2018-06-17:
Looks like oscillations ramping up, doesn't it? I have DYNAMIC_FILTER = ON, default double PT1 as per BF3.4...

RipperDrone, 2018-06-17:
Gyro resonance seems around 100Hz (I counted 10 wavelengths on the time scale), maybe the Svelte frame is soo delicate that it has a core resonance around 100Hz? Should I try decrease Dterm and Gyro Lowpass to 70 or 80Hz then?

McGiverGim, 2018-06-18:
@RipperDrone if your gyro loop is not the same than pid loop the blackbox has a bug in the spectrum. It is fixed in master. You can use the current master to analyze it.

RipperDrone, 2018-06-18:
OK, here is a blackbox dump for 8k/8k:

[image]

RipperDrone, 2018-06-18:
weird motor spectrum... (lowpass is already at 100Hz...?)
[image]

bforbort, 2018-06-19:
I started to get a lot of Dterm oscillation like that when I updated last time.  I made the following changes to my dterm filter...

> set dterm_lowpass_type = PT1
> set dterm_lowpass_hz = 48
> set dterm_lowpass2_hz = 100
> set dterm_notch_hz = 0
> set dterm_notch_cutoff = 160

...and reduce my PIDs.

> Reduced P by 35%
> Reduced I by 62%
> Reduced D by 75%

You're issue might be different, but this got me back in the air.

RipperDrone, 2018-06-19:
didn't help, seems 3 of my motors are running pretty rough, same motors were ok 4wks ago. MAybe they are rusty :-/ (rumored problem of FinX30), I think it's a HW issue...

closed 2018-06-19 by RipperDrone

RipperDrone, 2018-06-20:
After an odyssee of replacing motors, FC, 4in1ESC and fiddling hours over hours with different BF firmwares and settings, I was able to nail it down: Frame resonance was around 100Hz, after setting DTerm and Gyro LP to 80 Hz, quad flew fine. However, still 1 or 2 first launches it kept on twitching, only from 3rd flight everything was ok. I still suspect that Dynamic_Filter or other settings need some settle in time with clean gyro to determine some parameters which are then set for subsequent flights? Observed this several times in the past as well.

Anyway, this specific issue can be closed now...