betaflight/betaflight #6145: No telemetry on F.Port with HGLRC Zeus (OMNIBUSF4)

andrejpodzimek, 2018-06-17:
I can't get any telemetry on F.Port in Betaflight 3.3 and 3.4-RC2. I'm using a [HGLRC Zeus](https://www.hglrc.com/products/hglrc-f4-zeus-f4-flight-controller-integrated-with-osd-bec-pdb-aio-15a-blheli_s-4-in-1-esc-for-rc-racing-drone) and an [R-XSR](https://www.frsky-rc.com/product/r-xsr/) receiver connected to TX3 using the receiver's [uninverted "P" pad](https://oscarliang.com/uninverted-sbus-smart-port-frsky-receivers/). The receiver has [the latest beta F.Port firmware](https://www.frsky-rc.com/wp-content/uploads/Downloads/Firmware/R-XSR/Beta/R-XSR_FPORT_180615.zip) flashed in it. The transmitter's (Devo 10) signal is interpreted correctly by the FC, but I can't see any telemetry on the transmitter. Of course I followed the [installation steps](https://github.com/betaflight/betaflight/wiki/The-FrSky-FPort-Protocol#installation) for F4. I have 3 other F.Port setups on F3 and F7 that just work, including telemetry. This is my first attempt at using this on an F4 and thus far it's not going too well. :sob:  Here's my `diff all`:

```bash
# diff all

# version
# Betaflight / OMNIBUSF4 (OBF4) 3.4.0 Jun 17 2018 / 08:44:35 (2fe9d7990) MSP API: 1.39

board_name OMNIBUSF4
manufacturer_id AIRB
mcu_id 0020001a3436470137333638
signature 

# reset configuration to default settings
defaults nosave

# name
name Windy

# resources

# mixer

# servo

# servo mix

# feature
feature -RX_PARALLEL_PWM
feature RX_SERIAL
feature TELEMETRY
feature LED_STRIP

# beeper

# beacon

# map

# serial
serial 2 64 115200 57600 0 115200
serial 5 2048 115200 57600 0 115200

# led
led 0 6,0:D:ATOIW:0
led 1 7,0:D:ATOIW:0
led 2 8,0:D:ATOIW:0
led 3 9,0:D:ATOIW:0

# color

# mode_color

# aux
aux 0 0 0 1700 2100 0
aux 1 1 1 900 1300 0
aux 2 2 1 1300 1700 0
aux 3 13 3 1300 2100 0
aux 4 28 4 1700 2100 0

# adjrange
adjrange 0 0 2 900 2100 12 2 0 0

# rxrange

# vtx

# rxfail

# master
set gyro_lowpass_hz = 0
set mag_hardware = NONE
set baro_hardware = NONE
set max_check = 1950
set serialrx_provider = FPORT
set serialrx_halfduplex = ON
set blackbox_p_ratio = 0
set blackbox_device = NONE
set dshot_idle_value = 400
set motor_pwm_protocol = DSHOT600
set pid_process_denom = 1
set osd_vbat_pos = 2466
set osd_throttle_pos = 2485
set osd_crosshairs_pos = 2282
set osd_ah_sbar_pos = 2282
set osd_ah_pos = 2282
set osd_current_pos = 2475
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF

# profile
profile 0

set vbat_pid_gain = ON
set anti_gravity_gain = 3000
set dterm_setpoint_weight = 100
set d_level = 80

# profile
profile 1

set vbat_pid_gain = ON
set anti_gravity_gain = 3000
set dterm_setpoint_weight = 100
set d_level = 80

# profile
profile 2

set vbat_pid_gain = ON
set anti_gravity_gain = 3000
set dterm_setpoint_weight = 100
set d_level = 80

# restore original profile selection
profile 0

# rateprofile
rateprofile 0

# rateprofile
rateprofile 1

set roll_srate = 80
set pitch_srate = 80
set yaw_srate = 80

# rateprofile
rateprofile 2

set roll_srate = 85
set pitch_srate = 85
set yaw_srate = 80

# rateprofile
rateprofile 3

# rateprofile
rateprofile 4

# rateprofile
rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save
```

jflyper, 2018-06-17:
Can't reproduce.
Omnibus F4 AIO (gen1) flashed with 3.4-RC2, loaded with your configuration works with R-XSR in F.Port mode by UIM F/S.Port pad connected to TX3.

jflyper, 2018-06-19:
I have 180228 flashed.

andrejpodzimek, 2018-06-19:
I see. Now I spotted #5498. So I'll try to re-flash it, just to make sure there's no glitch of that kind.

andrejpodzimek, 2018-06-20:
I have telemetry. After downgrading the R-XSR firmware to 180228 it all just works, telemetry + fport.

I don't understand what exactly was wrong with the new beta firmware. My guess is that your setup might still differ from mine &mdash; I'm using Devo 10 with Deviation. If you're using OpenTX or something else, then that may be the difference.

It seems that there's a bug in Deviation, which somehow doesn't understand telemetry from the new beta R-XSR firmware. That's the only plausible theory I have at the moment.

closed 2018-06-20 by andrejpodzimek

mikeller, 2018-06-20:
@andrejpodzimek: I have got no clue how Deviation works, but just to make sure, did you try to rescan for sensors on the TX before you downgraded?

andrejpodzimek, 2018-06-20:
There is no such thing as scanning for sensors on Deviation. But I did try to rebind ~twice. Also tried both FCC and LBT versions of the R-XSR firmware. Both work perfectly fine with Deviation (LBT is called “EU” in the menu), but the new beta just doesn’t get any telemetry from the FC, no matter if I pick FCC or LBT. The RSSI telemetry from the RX works and shows up in the telemetry data (so some telemetry is being talransmitted from the RX to the TX, obviously), but there’s no current, voltage, rpm etc. Flashing the older (~February) R-XSR firmware restores all the telemetry data.

mikeller, 2018-06-20:
@andrejpodzimek: That's odd. I tried this version with a Taranis, and it seems to be working just fine. What TX module are you using?

andrejpodzimek, 2018-06-20:
@mikeller [Banggood classics](https://www.banggood.com/CC2500-NRF24L01-A7105-CYRF6936-4-In-1-RF-Module-For-Walkera-D7E-D10-Transmitter-p-1096707.html). It can talk to all my drones (10+ different protocols).

mikeller, 2018-06-21:
@andrejpodzimek: I have found the multiprotocol module TX implementation to have problems and act differently to FrSky TX hardware in other cases - probably just some internal detail of the (undocumented) TX implementation that they got wrong. Try updating your TX module to the latest version from https://github.com/pascallanger/DIY-Multiprotocol-TX-Module, and if that does not help, open an issue here: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/issues

mikeller, 2018-06-21:
Yeah, sorry, looks like I was a bit too quick on that. But this does not look like it is a FrSky product, so whatever firmware it runs will have been created by reverse engineering, or probably adapting the reverse engineered solution that is used in the multiprotocol module. In both these cases it is unlikely that FrSky will be particularly concerned about releasing RX firmware that does not work with this module...