betaflight/betaflight #6148: Pitch forward after jamming and cutting off throttle (3.4 RC2)

dronejunkie, 2018-06-18:
FC is Matek F405,  I will upload the dump later on.  The setting are pretty much default.  I have only reduced Pterm on RP by 2 points and increased dTerm on RP by 2 points.  The issue is when I jam and cut the throttle the quad pitch forward a lot.  It is reproducible.  I checked the BBL logs,  when the throttle is at peak   PID iterm also peaked but when the throttle was cut PID iterm went to zero and pretty much stayed there for a while.  It did not detect the drift at all. You can see the gyro pitch started to pick up as soon as throttle was cut. Item should had countered this but instead it stayed at around zeo.
 
This is a new quad so flashed to the latest FW.   Here is the BBL log

https://drive.google.com/open?id=1SZy9g98-jLmeq18aJGxylvXKWy0sM6LU

theSkonk, 2018-06-18:
its an ongoing mystery.. https://github.com/betaflight/betaflight/issues/6018
do you have one motor flatlining? thats whats happening to me.. different esc/FC combos and different hardware doing the same thing. only common thing between them is the esc/fc firmware

dronejunkie, 2018-06-18:
I checked the Pterm pitch, it is trying to counter the pitch forward. Motor 2 and 4 spun up as  soon as it detected the uncommanded pitch,   it appears like the correct was not sufficient.

theSkonk, 2018-06-18:
but is the kick been caused by  motor flatlining.. thats the issue with the other people including my self  your seems to be motors 1 and 3 they don't recover until you start lifting the throttle. mine is just motor 3 and it flatlines any time you have a throttle rc command of less than 1100.
the p term is trying to recover the attitude which is th pitch change but its not causing the motors to flatline..

dronejunkie, 2018-06-18:
If you check the BBL and look at the P term on pitch,  P item is the integral of that and as such the iterm should not be zero.  In the two punch outs in the log, item was at zero for 0.7-0.8 sec while the Pterm was -ve all this time. This appears to be a PID calculation error somewhere.   It does not seem to be a hardware issue.

McGiverGim, 2018-06-18:
Yes, something strange with I is happening. A lot of error appears and the I is not fixing it.

I suspect you are using some of the new features (`iterm_relax`, `absolute_control`...) and something is wrong or there is some very bad bug in RC2.

Can you paste a `diff` ?

dronejunkie, 2018-06-18:
diff 
https://drive.google.com/open?id=1S7SBq7qAcDV34VJ5j1xgCpIou19uXiPy

dump
https://drive.google.com/open?id=10-Ik5iohTggfuRxdWYoyp64uTQvoJSBc

McGiverGim, 2018-06-18:
@dronejunkie the only reason to ITerm not acting, is when the motor mixed is saturated. I see that this only happens when your throttle is zero, so this makes me thing that you maybe are not using AIRMODE and with throttle zero and without airmode the I will not act. Is that true? Can you enable AIRMODE and test again (as a feature or a switch)?

faceforfpv, 2018-06-18:
AIRMODE wasn't turned on by default in 3.4 RC1. Caught me by surprise. Could it be made part of the default configuration?

dronejunkie, 2018-06-18:
I thought airmode is on by default.  Let me check again.

McGiverGim, 2018-06-18:
No, it has not been made default in 3.4. And your `dump` shows it disabled as expected:
> feature -AIRMODE

flint723, 2018-06-18:
Airmode has never been on by default has it?  For me it could be on as a feature, but not sure how many people still use it as a mode (on a switch).

dronejunkie, 2018-06-19:
Looking at the code, you are right iterm will only increase if not saturated (as per the snippet below).  I checked the mixer.c   file Saturation occurd if motorMixRange  is bigger than 1.      This variable is not affected by AIR MODE from looking at the code.  It affects the motor output but not the mix.    Mix is calculated base in the mix on RPY.  motorMixRange is the different between the max mix in any of the four motors minus the lowest mix of any of the four.

I will test and see.

const bool outputSaturated = mixerIsOutputSaturated(axis, errorRate);
        if (outputSaturated == false || ABS(ITermNew) < ABS(ITerm)) {
            // Only increase ITerm if output is not saturated
            pidData[axis].I = ITermNew;
        }

dronejunkie, 2018-06-28:
> @flint723 <https://github.com/flint723> In our freestyle + race pilots
> group, most use it as a feature-on-a-switch. We have a dipping arm on
> sudden throttle cut phenomenon since BF 3.1.x so not sure what's going on
> as far as I-term / saturation... we have learnt to fly around it by not
> cutting throttle so hard. Anyway, this bug seems to be there unresolved for
> a very long time...

Air mode on fixed my issue. I checked the bbl and the iterm looks normal
after jamming the throttle. Before the iterm was zero for a while and did
not correct the pitch.

flint723, 2018-06-28:
@RipperDrone I assume you tried higher idle % doesn't help?  I don't know but since I've been trying multishot (32k32k), I noticed default idle for older protocols seems much higher than default for Dshot.

RipperDrone, 2018-06-29:
@flint723 yep, cranked it up to 7% on some quads, didn't help. Running 8k/8k only. Also tried MS, OS protocols...

dronejunkie, 2018-06-29:
You need to check the bbl to see if it is an iterm issue or something else.
Mine was definitely iterm was not accumulating after the throttle was zero
and transition to full.  I still do not understand why having air mode on
changed the motor saturation from true to false.  I looked at the mixer.c
file and could not see a link between air mode and motor mix. Air mode
affects the motor output not mix.   May be someone could please point out
this link in the code. I am really interested.

kenkamm, 2018-08-18:
This is definitely still an open issue. The brief nose bob on pitch and roll upon fast throttle changes is an issue regardless of airmode, antigravity, and PIDs.

mikeller, 2018-08-18:
@dronejunkie: Has this issue been fixed? If so, can you close it please?

closed 2018-08-19 by dronejunkie

RipperDrone, 2018-08-19:
No it hasn't been fixed at all, hasn't it? IMPOV it is linked to the arms drop issue after throttle cut, https://github.com/betaflight/betaflight/issues/6018

Seems ok we track it under #6018 from now...

dronejunkie, 2018-08-19:
My case was different the I term stopped accumulating and it was fixed
after I have airmode on.