betaflight/betaflight #6149: set iterm_relax = RP and other stuffs says ""###ERROR### Invalid name"" on CLI

Arclim, 2018-06-18:
set iterm_relax = RP and other stuffs says ""###ERROR### Invalid name"" on CLI
whats seems to be the problem? or am i missing something?

mikeller, 2018-06-18:
@Arclim: If you provided any of the information that you were asked to provide when opening an issue then there would be a chance that somebody was able to help you...

wind0r, 2018-06-18:
my magic 8 ball says that you try to use a new 3.4 feature with 3.3 firmware. this will not work. your fc doesnt magicily update

Arclim, 2018-06-18:
I just updated my firmware with the latest rc2 candidate

McGiverGim, 2018-06-18:
Start pasting a `diff` of your config, please.

Arclim, 2018-06-18:
CLI # diff
# Betaflight / SPRACINGF3 (SRF3) 3.4.0 Jun 10 2018 / 02:49:42 (a5ba01666)
MSP API: 1.39

board_name SPRACINGF3
manufacturer_id

feature -TELEMETRY
feature -RSSI_ADC
feature AIRMODE
aux 0 0 0 1700 2100 0
aux 1 13 1 1700 2100 0
rxfail 3 h
set gyro_sync_denom = 2
set gyro_lowpass2_hz = 500
set acc_hardware = NONE
set mag_hardware = NONE
set baro_hardware = NONE
set motor_pwm_protocol = ONESHOT42
profile 0

set dterm_lowpass2_hz = 300
set dterm_notch_cutoff = 0
set p_pitch = 64
set i_pitch = 52
set d_pitch = 54
set p_roll = 64
set i_roll = 51
set d_roll = 53
set i_yaw = 49
rateprofile 0

set roll_rc_rate = 136
set pitch_rc_rate = 136
set yaw_rc_rate = 136
set roll_expo = 20
set pitch_expo = 20
set yaw_expo = 10
set roll_srate = 66
set pitch_srate = 66
set yaw_srate = 64

Arclim, 2018-06-18:
> The iterm_relax is nos available for F3 by space problems. If you want it
> you must compile your own version disabling other things that you don't use.

how? can you pls guide me? im only running at 13% cpu load

EGParadox, 2018-06-18:
@Arclim @glenb2 from release notes: 

> unfortunately bugfixes and improvements in the flight controller core functionality have led to an increase of the firmware size, causing it to overflow the available space on a number of F3 based flight controllers. As a result, some features have had to be removed from a number of F3 based flight controllers in order to make the firmware fit into flash. The following targets are affected: BETAFLIGHTF3, COLIBRI_RACE, FRSKYF3, FURYF3OSD, LUX_RACE, MIDELICF3, OMNIBUS, RCEXPLORERF3, RG_SSD_F3, SPRACINGF3EVO, SPRACINGF3NEO;

McGiverGim, 2018-06-18:
First you must be able to build the firmware by yourself. Then you have some documentation about how to add or remove features here: https://github.com/betaflight/betaflight/blob/master/docs/Customized%20Version.md

mikeller, 2018-06-18:
I think @joelucid is maintaining a fork that has the new tuning features enabled for F3.

Arclim, 2018-06-18:
Thats a goodnews! Keep it up. We f3's are willing to wait.

robomaniac, 2018-06-26:
Here is Joelucid Flight Performance Edition
https://github.com/joelucid/betaflight/releases/tag/3.4-RC1

I would love to see the list or description of thing that was done to that edition.

Yamaford, 2018-06-26:
#Betaflight 3.4 RC1 - Flight Performance Edition

This is a flight performance release of betaflight 3.4-RC1. It adds the features AbsoluteControl, ItermRelax and SmoothingFilter to F3 flight controllers. To make flash space available the AcroTrainer feature was removed.

See https://github.com/betaflight/betaflight/wiki/Tuning-Tips-for-Betaflight-3.4 for an explanation of these new features.

Watch https://www.youtube.com/watch?v=U4FeVfSssNk to get a feel of what's possible with AbsoluteControl for LOS flight.

SmoothingFilter is not available on FRSKYF3, MIDELICF3, OMNIBUS and SIRINFPV due to space constraints.

Arclim, 2018-06-26:
How do you enable the item relax and smoothing?

Tuning-Tips-for-Betaflight-3.4 for an explanation of these new features.

Yamaford, 2018-06-26:
It's not a CPU intensity problem it's the size of the flash on the f3 chips. You can ditch all the RX protocol you don't use and free up a ton of space. You can  bring back all these features plus GPS and still use less flash space than 3.4 master.

Arclim, 2018-06-26:
Can you specify on clearing some space? I only started 7months ago and ican
only do tuning and some common jobs.

Arclim, 2018-06-26:
Im currently running cleanflight f3 4k4k and 12% cpu load