betaflight/betaflight #6154: Blackbox erase function in CMS

atomgomba, 2018-06-18:
Putting an option in the OSD stick menu for erasing blackbox data would have the benefit of not having to make any preliminary configuration in order to be able to clear data on the field before flight.

mikeller, 2018-06-18:
It also has the advantage over assigning a switch for BLACKBOX_ERASE that it is less likely to get accidentally activated by bumping the switch.

McGiverGim, 2018-06-18:
The option exists, if FLASHFS is defined:

https://github.com/betaflight/betaflight/blob/02efd46833155f43b77a15f8ebe795484898d488/src/main/cms/cms_menu_blackbox.c#L203-L205

mikeller, 2018-06-18:
Lol. :laughing:

atomgomba, 2018-06-18:
So resolving this issue was easy! :sweat_smile:

McGiverGim, 2018-06-18:
Please check if it is what you need, and if it is "fixed" you can close the issue ;)

timman2er, 2018-08-27:
Issue should be closed.

mikeller, 2018-08-28:
@atomgomba: Would you mind closing this?

atomgomba, 2018-08-28:
Not at all

closed 2018-08-28 by mikeller