betaflight/betaflight #6157: 3.4 RC2 LED_STRIP does not work on F4

jurkust, 2018-06-18:
With Betafligh 3.4 RC2 LED_STRIP does not work on F4, with RC1 works.
I'm testing on FRTFQ Flight Controller using CLRACINGF4 3.4.0-RC2. 
Diff:
```
# version
# Betaflight / CLRACINGF4 (CLR4) 3.4.0 Jun 17 2018 / 08:22:07 (2fe9d7990) MSP API: 1.39

# feature
feature LED_STRIP

# led
led 0 0,0::C:6
led 1 1,0::C:6
led 2 2,0::C:6
led 3 3,0::C:6
led 4 4,0::C:6
led 5 5,0::C:6
led 6 6,0::C:6
led 7 7,0::C:6
led 8 8,0::C:6
led 9 9,0::C:6
```
Same config for with RC1.

DieHertz, 2018-06-18:
Duplicate of #6146
Please use search before posting

closed 2018-06-18 by DieHertz

jurkust, 2018-06-18:
Sorry @DieHertz, searched, but used default filter, which is `is:issue is:open` and all these issues are closed, next time will do it right :)