betaflight/betaflight #6158: Saving dataflash to file - super-slow due to >1,000 packet errors...

RipperDrone, 2018-06-18:
FC OMNINXT7, dataflash transfer under Win x64 is awfully slow, thousands of packet errors. Looks like serial (?) protocol is not working correctly?

DieHertz, 2018-06-18:
Did you try `msc`?

RipperDrone, 2018-06-18:
sry, not familiar yet with msc, let me try to read some Wiki...

(btw, 526kB ended downloading after 6min, >8,000packet errors, satus bar shows D: ~800%, U: ~5% - looks like a buffer / timing prob)

RipperDrone, 2018-06-18:
ok, CLI'ed msc, then found Betaflight Flash Storage as drive, download smoothly and fast. This seems the way to move forward. Other method looks flawed...

DieHertz, 2018-06-18:
CDC is not the way to go for downloading large files :)

mikeller, 2018-06-26:
@RipperDrone: We still have to figure out how to best make use of MSC. Having it the default mode will make it all but impossible to configure your flight controller, since it needs to be in MSP mode for that. Adding a button to reboot into MSC in configurator is one way, but it needs the configurator to be installed just to be able to push the button. Having a button on the flight controller would be an elegant solution, but most flight controllers don't have a freely assignable button...

RipperDrone, 2018-06-26:
I see. Why not boot into MSC + download the flash file once users press the "Save to flash file..." button BFC?

mikeller, 2018-06-26:
That would be an option. Still needs the configurator to be installed, so it won't work on a mobile phone / tablet for example, where a physical button to boot the flight controller into MSC mode would work.

RipperDrone, 2018-06-26:
How 'bout the SpeedyBee Android/iOS app? Love it, it has the full BFC functionality (maybe net latest version add-ons). Can boot into MSC since it has a CLI interface if authors follow the BFC functionality and include the same feature into their app...

RipperDrone, 2018-06-26:
https://www.speedybee.com/download/

mikeller, 2018-06-26:
@RipperDrone: That one's gonna be a bit harder - it obviously does not use USB, and the bluetooth adapter is most likely connected to the flight controller over an UART. As far as I know, there is no standard for bluetooth attached storage, and even if there were, the serial protocol to facilitate this would first have to be added to Betaflight. But then, the flight controller only has 2 MB onboard flash - downloading that in configurator won't be too bad. :wink:

RipperDrone, 2018-06-26:
Many users like me never connect SpeedyBee over BT module, but over a simple USB cable. App supports both :-)

BT might be harder to implement, understood.

RipperDrone, 2018-06-26:
...and I'm using SpeedyBee with all my 10+ different FCs, it's not limited to the SpeedyBee F4 FC whatsoever. This makes this little app so versatile... :-)

mikeller, 2018-06-26:
That's easy then - any app can start MSC mode. My point is that ideally, you should not have to have any app installed to be able to do this.

RipperDrone, 2018-06-26:
U mean just connecting the FC over USB to a smartphone in the field should bring up the Betaflight emulated drive for accessing the BBlogs? Don't know... what do we want to do with the log file on our smartphone then? Needs another app for analysis anyway, I guess. So I wouldn't mind having to use ezGUI or SpeedyBee to be able to access/download log files in the field...

closed 2018-08-02 by stale[bot]