betaflight/betaflight #6159: SPRACINGF7DUAL won't go into DFU via software request

richard-scott, 2018-06-18:
I have just got an SPRACINGF7DUAL FC and it won't go into DFU mode via "bl" or the BFC Flashing tool. The board locks up and dies.

Using the boot button works just fine to get DFU on USB connection.

DieHertz, 2018-06-18:
Yeah, we know about the issue and none of the ideas gave any result :( something is probably binding the bootloader before you get a chance to talk to it

richard-scott, 2018-06-18:
A known issue... is there an issue ticket for it? I could not find one? :-(

DieHertz, 2018-06-18:
There's no ticket. I encountered it long ago when helping Dominic with BF support, we discussed it in our Dev chat and all the ideas I tried didn't work out.

AndersHoglund, 2018-06-18:
On other FCs (F3/F4 based) the bootloader can hook on to UART and ignore USB if there are activity on the UART Rx line. Not sure about F7. Have you tried a clean test with USB only connected and nothing else?

mikeller, 2018-06-18:
Just a thought: Are there any pins on this board that default to UART pins, but are permanently used for something else (SPI?), so that on warm reboot there is an initialised device connected to them, making the MCU bind the UART to it? Do we have to shut down peripherals before the reboot for this board?

MJ666, 2018-06-19:
We had this behavior on one of our AlienFlightNG prototypes while using an pull up resistor at the STM reset pin. After removing this all was normal. The reset pin shuld only use an 100nF cap to ground. May be good to check for this with this board.

DieHertz, 2018-06-19:
I think I tried resetting all GPIO and AF to default state before reboot to bootloader, but that didn't help

flint723, 2018-06-19:
I just tried this, my board goes into DFU fine via cli `bl`.  It is not installed, just the bare board on the bench.  FW 3.4.0 rc2

[diff 32k32kdshot1200.txt](https://github.com/betaflight/betaflight/files/2115802/diff.32k32kdshot1200.txt)

DieHertz, 2018-06-19:
That's interesting, could you take some detailed photos of both sides of your board?
Also try performing it multiple times via flashing firmware like following:
CLI -> bl -> flash firmware -> CLI -> bl -> flash firmware

flint723, 2018-06-19:
Yeah false alarm.  It worked the first time, no idea why.  Now it freezes every time, tried 12 more times or so.

flint723, 2018-06-19:
Another update.  Found a method of 100% repeat-ability.  

The f7 Dual will enter DFU with `bl` command one time after another board is plugged in.

- f7 dual won't enter DFU
- plug in matek f405 (uses different com port)
- unplug matek f405
- f7 dual will enter DFU with `bl` command
- unplug f7 dual
- plug in f7 dual, it won't enter DFU

DieHertz, 2018-06-19:
Weird and unexpected. Maybe need some USB sniffer to find out what happens

bigd942, 2018-06-24:
Any progress on this issue, my board seem's completely locked, I flashed BF (3.3) ok, managed to flash CF (2.3) back onto it again but now it won't connect at all it seems - seems to have messed up the STM drivers on my PC, I tried the ImpulseRC driver fixer tool and it failed to fix/recognise it. Is it best to raise this as a CF issue or have just this one issue open?

DieHertz, 2018-06-24:
This issue doesn't prevent any normal operation, it just required you to press the boot button manually.
Your issue sounds unrelated

mikeller, 2018-07-24:
@DieHertz: I think this is still a problem?

DieHertz, 2018-07-24:
Yes

haslinghuis, 2024-05-12:
Please reopen new issue if unresolved.

closed 2024-05-12 by haslinghuis