betaflight/betaflight #6160: DShot beacon delay before arming should not apply to beeper

nmaggioni, 2018-06-18:
The delay added with #6079 caught me by surprise when I tried to arm my quad: I use the DShot beeper and have a momentary **switch on my TX set to both the prearm and the beeper**. This way nearby people get an audible alert that my props are about to start spinning; this has turned out as an useful thing quite a lot of times when the paddock is packed with people.

At the moment, on **v3.4.0 RC1**,  I have to:

1) Activate the prearm switch.
2) Activate the arm switch.
3) Keep the prearm activated for >2s ([hardcoded delay](https://github.com/betaflight/betaflight/pull/6079/files#diff-db1df01656afde7b672c4ed2dffa57e0R28)).
4) Deactivate the prearm switch.

If I don't keep the prearm switch pulled for at least that amount of time, the quad refuses to arm due to prearm not being active.

My suggestion is to either apply the delay only on the beacon and not on the beeper _(is that feasible?)_, or make its behavior ignore the (pre)arming procedure _(disarmed -> beeper/prearm enabled via RC channel -> armed)_. Making it configurable by the user is my last idea, but that could be cumbersome and undeeded.

etracer65, 2018-06-18:
The delay is only related to the beacon - not sure what the complaint is.  It actually has nothing even to do with the beeper or beacon settings.  It simply looks to see if a beacon command was sent to the ESC's recently before arming (within the guard time).

You apparently also have the beacon set up with `beacon RX_SET`.

Note that the delay has been reduced to 1.2 seconds in 3.4RC2.

nmaggioni, 2018-06-18:
Is there any other way to use the DShot beeper (e.g.: motors beeping) without enabling RX_SET, then? Aren't the beeper and the beacon two different things?

My choice of momentary switches on the TX is limited to a single one, having to move either the beeper or the prearm to another, non-momentary one, would be an unnecessary nuisance in my opinion.

etracer65, 2018-06-18:
Maybe you're confused by the terms beeper vs beacon.  Beeper refers to a physical buzzer (or beeper) connected to a pin on your flight controller.  Beacon is when the motors are used to make a tone.

So if you're using beacon then there must be a delay before arming otherwise the ESC's won't accept the motor direction commands because they ignore all commands while playing the tone through the motor.

nmaggioni, 2018-06-18:
This terminology is rather confusing, especially because BLHeli interprets the "beacon" as the one automatically triggered after a period of inactivity of the ESCs.

Anyhow: has my quad been arming correctly out of pure luck every time up to now, then? I don't wanna be rude, just asking :)

etracer65, 2018-06-18:
Also BLHeli refers to the DSHOT commands to "beep" the motors as "beacon commands" as well.

nmaggioni, 2018-06-18:
I guess that pilots that use my same arming setup will just have to live with keeping the prearm switch toggled for at least the duration of the delay.

closed 2018-06-18 by nmaggioni