betaflight/betaflight #6167: Add option to include temperature in Blackbox logging

ckwkalk, 2018-06-19:
Would it be possible to add the option to log temperature information (e.g. from the Barometer sensor) in Blackbox? This could be useful in a number of ways including looking at effects on other sensors which are affected by temperature changes (such as accelerometer).

timman2er, 2018-09-02:
@ckwkalk If I were to code this are you willing to test?

ckwkalk, 2018-09-11:
Hi, yes I would be happy to test - have a spare board sitting not doing
anything!

timman2er, 2018-09-11:
@ckwkalk in that case, if you can wait another month I'll drop you my email and we can get it working.

ckwkalk, 2018-09-11:
Great thanks - do you think that the specific hardware will be an issue I
am using an SPRacing F3 Deluxe board that has the combined
pressure/temp/humidity sensor?

timman2er, 2018-09-12:
@clwkalk if you can get temperature to show up in OSD on that board there should be 0 issues on your end.

timman2er, 2018-09-20:
Sorry for the clutter @ckwkalk but in my bio you'll find my email, drop me an email and I'll get to work on it somewhere next week.

tavdog, 2022-05-24:
Was this every completed?

closed 2026-09-26 by blckmn