betaflight/betaflight #6169: 3.3.3 & 3.4-RC1,2 OMNIBUSF4* targets do not allow softserial to be configured

jflyper, 2018-06-19:
This is caused by `FEATURE_RX_PARALLEL_PWM` inhibiting `FEATURE_SOFTSERIAL` to persist:
https://github.com/betaflight/betaflight/blob/713e72321b7cb44ee87dc8e0f8b7c70ddcfe176c/src/main/fc/config.c#L214-L226

The `featureClear(FEATURE_SOFTSERIAL)` should not be called only because parallel PWM is specified. Probably a left over from non-resource reconfigurable time.

I would propose to remove the call. Thoughts?

BTW, a workaround exists; configuring receiver other than parallel PWM will  (naturally) let `FEATURE_SOFTSERIAL` to persist.

mikeller, 2018-06-20:
Apart from this fix, should we change the default for RX to be *not* PWM? That's probably the least commonly used protocol these days.

shaiss, 2018-06-20:
@mikeller.  Making it serial would make it easier for new users since they right away see the common serial types. I ibus sbus spektrum etc...

mikeller, 2018-06-20:
@shaiss: The problem with serial is that it requires a serial port to be configured for RX, otherwise it will reset, since the serial driver can not start up without a port.

PR #6173 (2018-06-20): Don't let PWM reset SOFTSERIAL
Fixes: #6169
closed 2018-06-20 by mikeller