betaflight/betaflight #6170: Emax F4 Magnum All In One FPV Stack Tower + BF 3.4.0 RC2  - LEDs and Baro don`t work

Dieselfan, 2018-06-19:
With firmware OBF4 - BF 3.4.0 RC2 the external LEDs and a external Baro-sensor at the Emax F4-FC do not work.
With OBF4 - BF 3.3 it was fine.

The same at a Matek F405 AiO-FC with the BF 3.4.0 RC2 MATEKF405 - no LEDs and Baro.

coolername, 2018-06-19:
Hallo together!
Today i tried the new 3.4 rc2 on my DYS F4 V1 and on the DYS F4 V2
On both FC‘s the LED strips don‘t work anymore.

But:
 Fantastic flight-Feeling! Thank You!

bout100, 2018-06-19:
I've just loaded 3.4.0-RC2 on a CLracing F4S and the LED strip functionality also doesn't work. Verified by re-flashing 3.3.3 with default settings, and the LEDs work.

DieHertz, 2018-06-19:
Duplicate of #2965

closed 2018-06-19 by DieHertz

Dieselfan, 2018-06-19:
@DieHertz 
Hello, why do you closed this issue. It isn`t solved, or?
Kindly regards

DieHertz, 2018-06-19:
I closed it because it's a duplicate of an issue solved 2 days ago and reported some 5 times already.

Dieselfan, 2018-06-19:
Thanks, but it didn`t work at my FCs with RC2

DieHertz, 2018-06-19:
And it shouldn't.

bout100, 2018-06-19:
Latest build works for me, cheers.

Dieselfan, 2018-06-19:
What? I didn`t understand you, sorry. My english is not so good.

bout100, 2018-06-19:
It’s fixed in master. To test you have to manually download the firmware from the releases page, or wait for the next release.

bout100, 2018-06-19:
Sorry, beta page, not releases:
https://ci.betaflight.tech/job/Betaflight/lastBuild/artifact/obj/ <https://ci.betaflight.tech/job/Betaflight/lastBuild/artifact/obj/>

Dieselfan, 2018-06-19:
Ok. Thank you. Regards.

Dieselfan, 2018-06-19:
You are great.

Dieselfan, 2018-06-20:
@bout100 
Thank you, now the led-strip do work, but the Baro-sensor didn`t. The sensor is activated in the GUI, but in the sensor-tab I didn´t see parameters - pemanently on 0-line.
Please you can help me?
Kindly regards

[get-baro_cli_FROG_F4 Magnum_20180620_175915.txt](https://github.com/betaflight/betaflight/files/2120149/get-baro_cli_FROG_F4.Magnum_20180620_175915.txt)