betaflight/betaflight #6171: SPI-RX frsky_x twitches on 3.4 RC4

brycedjohnson, 2018-06-19:
On newest 3.4_RC2 the frsky_x twitches are still there. Maybe there are less of them? They were fixed in the dead pull request #5741.  I wasn't going to look at this issue in code anymore so if anyone wants to fix it.

The red trace dropping down to the minimum is an example of what is causing the issue:
[image]

# diff all
# Betaflight / CRAZYBEEF3FR (CBFR) 3.4.0 Jun 17 2018 / 08:24:01 (2fe9d7990) MSP API: 1.39

board_name CRAZYBEEF3FR
manufacturer_id 
mcu_id 001f002d5434571320383734
defaults nosave
name Bryce[\]^_

feature -MOTOR_STOP
feature AIRMODE
beacon RX_LOST
beacon RX_SET
aux 0 0 0 1200 2100 0
aux 1 1 2 1200 2100 0
aux 2 13 1 1700 2100 0
aux 3 35 1 1300 1700 0
aux 4 36 3 1700 2100 0
set gyro_sync_denom = 2
set gyro_lowpass_hz = 120
set acc_calibration = -21,-67,-183
set min_check = 1010
set max_check = 1990
set rc_interp_ch = RPYT
set rx_min_usec = 800
set dshot_idle_value = 900
set motor_pwm_protocol = DSHOT300
set failsafe_delay = 10
set bat_capacity = 260
set vbat_max_cell_voltage = 45
set vbat_min_cell_voltage = 28
set vbat_warning_cell_voltage = 29
set vbat_scale = 113
set beeper_dshot_beacon_tone = 4
set yaw_motors_reversed = ON
set crashflip_motor_percent = 50
set 3d_switched_mode = ON
set small_angle = 180
set pid_process_denom = 1
set runaway_takeoff_prevention = OFF
set osd_units = IMPERIAL
set osd_warn_esc_fail = OFF
set osd_vbat_pos = 385
set osd_rssi_pos = 38
set osd_tim_1_pos = 54
set osd_tim_2_pos = 2455
set osd_flymode_pos = 333
set osd_throttle_pos = 224
set osd_vtx_channel_pos = 352
set osd_current_pos = 2400
set osd_mah_drawn_pos = 352
set osd_craft_name_pos = 2441
set osd_gps_speed_pos = 218
set osd_gps_lon_pos = 82
set osd_gps_lat_pos = 65
set osd_gps_sats_pos = 51
set osd_home_dir_pos = 302
set osd_home_dist_pos = 303
set osd_compass_bar_pos = 266
set osd_altitude_pos = 247
set osd_pid_roll_pos = 423
set osd_pid_pitch_pos = 455
set osd_pid_yaw_pos = 487
set osd_debug_pos = 96
set osd_power_pos = 320
set osd_pidrate_profile_pos = 345
set osd_warnings_pos = 2055
set osd_avg_cell_voltage_pos = 2433
set osd_pit_ang_pos = 257
set osd_rol_ang_pos = 289
set osd_battery_usage_pos = 465
set osd_disarmed_pos = 138
set osd_nheading_pos = 311
set osd_nvario_pos = 279
set osd_esc_tmp_pos = 82
set osd_esc_rpm_pos = 472
set osd_rtc_date_time_pos = 2080
set osd_stat_rtc_date_time = ON
set osd_stat_tim_1 = ON
set osd_stat_max_dist = ON
set osd_stat_min_batt = OFF
set osd_stat_battery = ON
set osd_stat_bb_no = OFF
set debug_mode = RX_FRSKY_SPI
set vcd_video_system = NTSC
set frsky_spi_tx_id = 72,197
set frsky_spi_offset = -7
set frsky_spi_bind_hop_data = 6,219,199,179,159,139,119,99,79,59,39,19,234,214,194,174,154,136,114,94,74,54,34,14,229,209,189,169,149,129,109,89,69,49,29,9,224,204,184,164,144,124,104,84,64,44,24,0,0,0
profile 0

set anti_gravity_gain = 3000
set dterm_setpoint_weight = 50
set crash_recovery = ON
set iterm_rotation = OFF
set smart_feedforward = ON
set iterm_windup = 50
set throttle_boost = 30
set p_pitch = 80
set d_pitch = 35
set p_roll = 80
set i_roll = 50
set d_roll = 35
set p_yaw = 110
set p_level = 120
set level_limit = 60
profile 1

set dterm_lowpass_type = BIQUAD
set dterm_lowpass2_hz = 0
set dterm_notch_hz = 260
set anti_gravity_gain = 1000
set setpoint_relax_ratio = 100
set dterm_setpoint_weight = 0
set iterm_rotation = OFF
set iterm_windup = 50
set throttle_boost = 0
set p_pitch = 58
set d_pitch = 35
set p_roll = 40
set i_roll = 40
set d_roll = 30
set p_yaw = 70
profile 2

set dterm_lowpass_type = BIQUAD
set dterm_lowpass2_hz = 0
set dterm_notch_hz = 260
set anti_gravity_gain = 1000
set setpoint_relax_ratio = 100
set dterm_setpoint_weight = 0
set iterm_rotation = OFF
set iterm_windup = 50
set throttle_boost = 0
set p_pitch = 58
set d_pitch = 35
set p_roll = 40
set i_roll = 40
set d_roll = 30
set p_yaw = 70
profile 0
rateprofile 0

set roll_expo = 20
set pitch_expo = 20
set yaw_expo = 20
set roll_srate = 80
set pitch_srate = 80
set yaw_srate = 80
rateprofile 1

rateprofile 2

rateprofile 3

rateprofile 4

rateprofile 5

rateprofile 0
save
#

mikeller, 2018-06-20:
So what you are claiming is that when your TX is sending a value of 1000 then this should be ignored? Because that is what the dangerous hack that you tried to add in #5741 effectively did, on analysis of the code.

This is most likely a hardware problem.

closed 2018-06-20 by mikeller

brycedjohnson, 2018-06-20:
I'm not saying that is the best way to fix it, I just was saying that I did before made it go away.  If you are saying it is a taranis issue, I don't have this issue on any x4r or xsr. The overflow fix didn't appear to fix it for 100% for me.

Looks like there are other people with this problem still. https://github.com/betaflight/betaflight/pull/6076#issuecomment-398507194

brycedjohnson, 2018-06-20:
I captured the issue a couple more times on another taranis.  Also this is a different crazybee f3 board as well.

[image]

[image]

What more info did you need to show it is an issue that should remain open?

mikeller, 2018-06-20:
@brycedjohnson: This is not about feelings. But working with people who do not listen to what they are told is a waste of time, and I am not going to do more of that.

brycedjohnson, 2018-06-20:
@mikeller I don't know why you edited your above after the fact instead of asking me to check that.

> The diff you provided shows that you have telemetry enabled. You were told a number of times that this is still broken, so why do you expect that this would not cause problems?

Still get the twitches with telemetry off. 

[image]

```
# diff dump
# Betaflight / CRAZYBEEF3FR (CBFR) 3.4.0 Jun 17 2018 / 08:24:01 (2fe9d7990) MSP API: 1.39

board_name CRAZYBEEF3FR
manufacturer_id 
name Bryce[\]^_

feature -MOTOR_STOP
feature -TELEMETRY
feature AIRMODE
beacon RX_LOST
beacon RX_SET
```

Of course it is about feelings. You likely pissed I wouldn't do the fix exactly how you liked and am wasting your time.  I'm pissed that you don't listen to my reasons or my compromise solution and that I spent far too much time looking for the bug and now it seems you want to believe it doesn't exist and close it outright.  Working with people is a 2 way street (especially when you are herding a bunch of volunteers) it isn't telling people to do what they are told.

I do have a lot of respect for you, I certainly would not want run this project while contributing a ton of code and deal with all the developers and users.  I just wanted to fix an issue on the quad I am flying. If you don't want me to do ever do another pull request or comment in a issue I won't, but maybe someone else can come along and fix it at some point.  Or at least let people know there is a issue so they don't try it.

mikeller, 2018-06-20:
Sorry for the editing - I hadn't looked at the config diff in enough detail before, and had assumed that you tested with telemetry off since you'd been told enough times before that this is still broken.

And yes, I am getting pissed off because this is wasting way too much time. The integer overrun that you spotted (thanks for that) has been fixed in the correct way, and the fix seems to be working - otherwise the drop outs that you are seeing should occur for everybody.

If there still is an issue with incorrect values occurring in some configurations, we need to look deeper - just ignoring valid values is certainly not a solution.

While you are here, are you going to fix the bug that causes ESCs to lock up in BLHeli_S 16.67:
https://github.com/betaflight/betaflight/issues/6088#issuecomment-397825355

ryamoo, 2018-06-24:
@mikeller I'm seeing the same thing that @brycedjohnson is seeing. I noticed it in flight on RC2, and thought I had a motor or ESC problem. After digging, I spotted the channel twitches on the receiver tab on betaflight. 
The two yellow circles highlight where this happens. 
[image]

mikeller, 2018-06-26:
Interesting. They twitches go up and down, not just to the minimum of the range. This means that simple filtering out the minimum values isn't going to fix this.

I wonder how well RC smoothing will work to work around this problem?

Long term we'll have to work on figuring out what causes the twitches, and fix this underlying problem.

ryamoo, 2018-06-26:
@mikeller the channels that twitch are going to default/idle.
So the one that went up, was a slider input on my taranis, I had slid to its lowest point, and it jumped up to idle middle (around 1500, along with the other idle/unused channels).
The blue line that dropped down was throttle. I had the stick pushed up to high, about 1700, and it dropped down to idle bottom (about 988).

mikeller, 2018-06-26:
Ok, that is weird, looks almost like failsafe is kicking in for these channels.

ryamoo, 2018-06-26:
@mikeller yes, that's the impression I got.
I'd love to dive in and see if I can find the problem, but have no experience whatsoever and am definitely not a developer

githubDLG, 2018-07-26:
@brycedjohnson 
@mikeller 
@ryamoo 
Hey, thanks for you guys working and testing on Frsky SPI RX !!!!!
Hope we can finally fix this twitch issue~~

ryamoo, 2018-07-26:
Unfortunately I haven't had much chance to test it yet.
I did get round to resetting my Taranis to default settings, I've re-updated my multi-protocol module (my Taranis is missing it's internal module), and I've flashed a fresh copy of betaflight 3.4 (final version, not an RC). 

The "twitches" still appear on Frsky_X / D16 on betaflight configurator, but I haven't been able to fly that quad yet (poorly arm...). Using Frsky_D / D8 the twitches didn't appear to happen, but I may have missed them; not entirely sure there, didn't have enough time to keep watching for them. 
I found a way of using the 8 channels for what I needed, but it would be nice to have the 16 channels.

I'll hopefully be getting a new frame soon, and can take it for a flight, however when I last tried arming the quad before these test, it would failsafe as soon as I armed (throttle "twitched" down to failsafe level, causing failsafe).

ryamoo, 2018-10-25:
@mikeller Do you know if there was there ever any progress on this? Is this now being looked at as part of #6172?
As of 3.5.2 the issue remains as per my last comment - channel twitches when using frsky_x / D16, but not on frsky_d / D8.

brycedjohnson, 2018-10-25:
@ryamoo No one is looking at this issue.

I had a couple taranis running opentx that had this issue and I had a code workaround that fixed it for me.  I switched to ersky9x on my taranis and a deviationtx and I haven't seen the blips anymore.  Not sure what changed but I didn't want to investigate it anymore since the issue was closed anyways.

ryamoo, 2018-10-25:
@brycedjohnson frustrating.
I came to look if there was any progress and saw the 6172 link from another and wondered if that was being linked as the same cause. 
I might have a look at your #5471 fix and see if I can use it myself (never tried before, so that'll be interesting...)

brycedjohnson, 2018-10-25:
If I were you (and didn't have a bunch of models or was using lua scripts),  I'd flash ersky9x on your taranis and see if it goes away.  The latency is way better than opentx anyways!  From my testing: opentx is 16-35ms and it goes down to 11-22ms on ersky9x

[image]

mikeller, 2018-10-25:
@brycedjohnson: So from what you are saying, the twitches went away when you replaced the OpenTx firmware with er9x, but were still using the built in XJT, with the same firmware?
This makes it look like the twitches were actually sent by the TX firmware, and not an artefact of the transmission, or caused by a problem in the SPI RX. What options do you have on in your OpenTx? Since I did not get any twitches after #6076 was merged, I suspect that having / not having one of the compile time options in OpenTx is the culprit causing the twitches to be sent.

brycedjohnson, 2018-10-25:
@mikeller I really don't want to deal with this issue again, you closed this and didn't reopen it, when I asked and showed that I was still having the bug.  I was just letting @ryamoo know no one was looking at it.

Anyways, briefly... 3.4_RC2 from the top of this issue still had the twitches and that included #6076.  I don't know exactly when/why the twitches went away but I still was getting them on 3.4_rc2 and later.  https://github.com/betaflight/betaflight/pull/6076#issuecomment-415677850  Had the twitches on 3.5

After that, I didn't notice since I was using my mod to filter out the bad samples until recently, when I was already running only ersky9x and deviationtx.

mikeller, 2018-10-28:
@brycedjohnson:

> I was just letting @ryamoo know no one was looking at it.

No, you are not:

> I had a couple taranis running opentx that...

> I switched to ersky9x on...

> If I were you...

If you don't want to help fixing this, but instead just want to spread rumours and then refuse to supply data, and cling on to absurd and dangerous ideas, then you are not helping anybody, you're just confusing users looking for information. In this case please stay away from here.

ryamoo, 2018-10-28:
@mikeller I know that my taranis is flashed with multimodule on and heli off because I use a multimodule to connect to everything (issue with the internal antenna) and most setup guides say to turn heli off.
I can't remember what other options I used but I'll check.
I may have some time this evening to run through different options and try your build too. 
I'll also try ersky9x too if I can figure that out. 

These FCs (I have 3) on frsky_x are the only FCs that i have the issue with. I'll check to see if my other RX are frsky_x / D16 because I can't remember (I'm sure my whoop is). Also use this module with a couple of flysky which are fine. 

I'll also try a flying buddy's TX with one of my matek F411-one too.

mikeller, 2018-10-28:
@ryamoo: Ok. I can't make a statement on if this works correctly with the Multimodule (plus the Multimodule comes with an entire set of settings of its own), I have only ever tested this with the internal XJT module, I am using the Multimodule for non-FrSky modes.

mikeller, 2018-10-29:
@brycedjohnson:

> I had noheli lua luac ppmus multi on opentx.

Could be ppmus then - this seems to be likely since this is a TX output issue (and I was using a build with lua and luac a while back and didn't have the issue either. Or it could be a different firmware version or even hardware version of the XJT.

> You can't confirm it because you've never had the issue.

So the fact that I tested with a number of different boards, with a lot of different configurations, and in fact tried and was not able to replicate the problem with every single configuration that you reported as 'faulty' after #6076 does not count for anything in your opinion?

You really need to let go of the notion that 'there must be something wrong with Betaflight that needs fixing', this is getting beyond absurd.

Even you seem to have learned by now that the 'Betaflight issue' that you want to investigate is going away when switching TX, but you are still accusing me of not being able to confirm that this is a Betaflight issue because it does not happen with my TX.

> I'm not clinging to any dangourous ideas, 

You did, for way longer than was reasonable:

> I'm not saying that is the best way to fix it, I just was saying that I did before made it go away. 

> I've always tried to be respectful to you, please try and do that same for me even if we disagree.

I am trying to be respectful. It is just that my patience is wearing thin if I am being confronted with the same absurd and disproved notion over and over again. My apologies for that.

This is now definitely done for me.