betaflight/betaflight #6172: SPI-RX frsky_x lockup

brycedjohnson, 2018-06-19:
Example of the issue and my cat grooming.  At about 1:02 you can see the on time stop incrementing at 1:33.  If you were flying it would fall out of the air.  Seen this issue with Matek411RX and crazybeeFR F3.

https://www.youtube.com/watch?v=GtK06Mvrtwc

The debug info is the spi_frsky_x debugging.  I was walking around out of the room and holding the antenna so it might have missed some packets.  Not sure if that causes the issue to happen more?

In this issue I was looking at the root cause and it seemed like it was related to smartport telemetry:
https://github.com/betaflight/betaflight/issues/5576#issuecomment-385857925

Haven't tried disabling the telemetry slidered on the configurator.  That may stop this from happening, but you loose telemetry. 

You can also switch to frsky_d and it doesn't have this issue.

mikeller, 2018-06-20:
FrSky D still has a problem with occasional lockups when telemetry is on as well - it seems to be tied to the pre-existing CPU load, i.e. it happens more often if faster loop times are used.

Recommendation: Do not use telemetry when using FrSky SPI RX.

brycedjohnson, 2018-06-20:
I haven't had any lockups on frsky D and telemetry on pre-3.4RCs, but I was only running at 4k/4k.

mikeller, 2018-06-21:
@brycedjohnson: Interesting. My suspicion is that the lockups are caused by the generation of telemetry frames taking up too much time, and causing the SPI RX to get out of synch. If this is the case, then whether or not it's happening on a given setup will be dependent on the MCU type, and also on the number of present / activated features.

JamilFayad, 2018-06-21:
Hey I am facing the same problems it is just locking up and I'm losing controls mid-air. That's not good at all. It is not falling out of the sky tho as someone noted here before. https://youtu.be/RA1teE-yYfc
Provided is a link to the DVR footage recorded as such issues were faced. Important moments to note out in the video is the 26th second, 41st, and 2 minutes and 15 seconds into the video. Any recommendations so I could have this flying the way it should? Thanks :)

pkruger, 2018-06-23:
I have the same problem.  Didn't see this ticket and got a replacement FC from Banggood, still have the issue.  Tried different option PID loop and Gyro speeds, also tried disabling telemetry, but it made no difference.

JamilFayad, 2018-06-23:
I personally asked for a refund when banggood offered a replacement and I got my money back. Though my problem with this FC wasn't just that and that's probably why. I ended up just switching my entire FC with an FrSky XSRF4O which is an fc with an integrated receiver however its PDB-less so I had to use a Matek PDB that was hanging around. Haven't tested it yet nevertheless as far as reviews can go, I was told that this FC is a lot better and doesn't seem to encounter such problems so I am expecting good results on a fine day. Will update you tho for sure :)

Ask1me, 2018-06-24:
I'm sorry if this is Out Of Topic, I stressed Out becouse of this problems :

My board FC CrazybeeF3 BNF Frysky,
my latest firmware : betaflight_3.3.3_crazybeeF3FR (Stable)
SPI_Rx Support Receiver Mode

I accidently choosing wrong SPI Reciever Provider to A7105_Flysky_2A and save it

now my FC can not connect to USB anymore, everytime connect it is says " Unkown USB device (Device descriptor request Failed) for windows 10"

is there any suggestions to fix it? please help me anyone..

pkruger, 2018-07-28:
This is still an issue, it can't be closed because it has not been fixed - yet.

mikerc2017, 2018-08-08:
midelic who create version FC integrade FRSKY . i think he can fix it .  Matel fc using his code .

H4xus, 2018-08-13:
I found this when searching and wanted to add that I'm still experiencing this with Betaflight 3.5 RC2.

Flash target: CRAZYBEEF3FR
Board model: Racerstar Crazybee F3 Flight Controller
ESC protocol: D600
FrSky protocol: D16 (8 channel with telemetry)
4K/4K loop timings.

D8 works fine, even with telemetry. I've not had any dropouts. However with D16 8 channel and telemetry my quadcopter falls out of the sky seemingly randomly, and I can only re-establish a link when I power cycle the flight controller. 

I will disable telemetry and test tonight.

H4xus, 2018-08-13:
Update:

D16 working now with no failsafes/rx failure after 6 packs when disabling telemetry both on the Taranis when binding, and within Betaflight. I still get RSSI, which is all I really want. 

About to fly another 6. Will report back if things change, but this workaround appears to be ok so far.

mikeller, 2018-08-14:
@H4xus: You should also get battery voltage in A1.

pd5dj, 2018-09-02:
Hi all, I just stumbled on this page looking for the "drop out of the sky" issue..

What is the best thing to do lately?
I have the CrazyBee F3 FR FC, (snapper7)
BF 3.3.0 installed right now.
Bind in D16 mode
I have tried D8 mode, but when arming, it directly disarms.

mikeller, 2018-09-02:
@pd5dj: Update to 3.5.0. FrSky D works with telemetry, FrSky X works with telemetry off.

pd5dj, 2018-09-03:
Hi, upgraded to 3.5
Binded in D8 mode, But I have no telemetry.
I did not look further because I had no time anymore. do I have to do a
rescan again to obtain all the sensors?

Bjorn

mikeller, 2018-09-03:
@pd5dj: You will have to make sure the feature 'TELEMETRY' is enabled. And you might have to rescan the sensors.

pd5dj, 2018-09-03:
Ok, cleared all the sensors, and did a rescan.. all the sensors came up
fine..
Although only 1 time,, only active sensors now are A1 and A2, the rest is
not showing up.
Even after re-power of TX and FC.. only A1 and A2.

No VFAS and Current sensors,, no ACC's too.
Puzzled here :-)

mikeller, 2018-09-04:
Hmm, I've seen this before, and I think there was a bug causing this at some stage. But that got fixed in 3.5.

H4xus, 2018-09-06:
I was still seeing it in Betaflight 3.5 RC2, and just tested again with 3.5. Still get lockups with telemetry on, fine without.

mikeller, 2018-09-07:
@H4xus: Yes, this has not yet been identified / fixed.

19Gerhard85, 2018-09-24:
Yesterday I flew 10+ packs on the Crazybee F3. After 5 packs I had a "failsafe", the rebooted and the status message in OSD display "BADRX".

- Board: CRAZYBEEF3FR
- Betaflight Version: 3.5.1
- Rx Mode: D8
- Telemetry: On
- Enabled Features: OSD, TELEMETRY
- PID Loop: 4k/4k

mikeller, 2018-09-24:
@19Gerhard85: For now, try disabling TELEMETRY - the voltage will still be received on the TX as A2.

jdedonato1, 2018-09-30:
I have the latest firmware 3.5.1. 2 quads using the target MATEKF411RX. Tried D8 and D16. D8 with telemetry still doesn't work. D16 telemetry works. One quad ( this one is a brushed) will keep at full throttle randomly and would cause a runaway so can go outdoors ( mostly when crashing) and one falls out of sky ( brushless on dshot600). Please tell me this will be fixed and I will be able to use telemetry

jflyper, 2018-09-30:
@jdedonato1 We at least your CLI `diff` output to say anything meaningful.

jflyper, 2018-09-30:
@jdedonato1 
At least, local bench test (no flight test) with a bare MATEKF411RX flashed with 3.5.1 basically works with
D8 : OK
D16 Ch1-8 Telem On : OK

Although issue involves in-flight failure, your issue with D8 suggest there is something wrong with your setup.

jdedonato1, 2018-09-30:
I will check it out soon and provide better info to help with issue. Thanks
for response.

closed 2018-11-06 by stale[bot]

enhemh, 2019-01-04:
> I'm sorry if this is Out Of Topic, I stressed Out becouse of this problems :
>
> My board FC CrazybeeF3 BNF Frysky,
> my latest firmware : betaflight_3.3.3_crazybeeF3FR (Stable)
> SPI_Rx Support Receiver [...]

same here... any fix ?

eitama, 2019-01-23:
To bad it's closed, d16 is still being used by default on mobula7 quads, which made me mess up my frame.

ijarboe, 2019-01-27:
> same here... any fix ?

I did the same thing to a different board. DId anyone find a fix to this?

mikeller, 2019-01-27:
@enhemh, @ijarboe: Connect the two tiny 'boot' pads on the board, then connect USB - the board will be recognised in DFU mode - flash the correct firmware, and you'll be good to go.

mikeller, 2019-01-29:
@ijarboe: Ok. Have you tried flashing with the 'Full Chip Erase' option enabled?

mikeller, 2019-01-29:
@ijarboe: In that case it looks like the board was damaged by flashing an incorrect firmware - with the SPI RX this is a possibility.

Protonus, 2019-02-28:
I am having this issue with a brand new Eachine Trashcan with the FrSky SPI FC, the CRAZYBEEF4 PRO FR (target CrazybeeF4FR).  This quad ships on OpenTX 4.0.0 out of the box.  I have an FrSky X-Lite, running OpenTX 2.2.3.  If I bind as D16 with Telemetry enabled in Betaflight, this issue occurs.

The FC locks up when this happens, as the LED scanner rear lightbar on the Trashcan also stops moving, and it will not reconnect to r/c until it's power cycled. On my very first battery of real flight time, I had 3 distinct events of this where it would fall out of the sky and telemetry would be lost as soon as it happened, while it was only about 60 feet away.  I've even had this happen while the quad was plugged in to USB with no batteries in!  

Turning OFF Telemetry in Betaflight, seems to stop the the lockups of the FC; however, I still get "stutters" where the quad will lose power for a brief moment or I'll lose control of it for a moment.  

If however I change it to FRSKY_D (D8) instead of the stock FRSKY_X (D16) under Receiver configuration on the Configuration page of Betaflight, and then pair my radio in D8 mode to it instead, all problems stop, and it flies as expected, no stutters, no lockups.

Though the issue may be related to OpenTX, this seems like a problem in Betaflight to me with these SPI FC's. 

There is a thread on this here: https://www.rcgroups.com/forums/showthread.php?3244881-Dropouts-In-D16-Mode-With-Telemetry-On-In-SPI-Receiver-Flight-Controllers

Asizon, 2019-02-28:
the problem of temeletry at the moment has no solution, even so Frsky_X with D16 works perfectly for me with the latest builds.

Protonus, 2019-03-01:
> the problem of temeletry at the moment has no solution, even so Frsky_X with D16 works perfectly for me with the latest builds.

What version of OpenTX do you run?  In researching this and reading reports, it's looking like it's an issue on OpenTX 2.2.2 and newer, and not being reported on 2.2.1 and older.

Asizon, 2019-03-01:
2.2.3,and 2.2.2 OK

Asizon, 2019-03-01:
Also,now is LBT available for Frsky_X

Asizon, 2019-03-01:
Some fixes:
https://github.com/betaflight/betaflight/pull/7345

MacGWik, 2019-03-06:
Hello, I have a question about this too.
Is the locks up problem only happen on SPI RX ?
Because I want to use my xm+ on the crazybee f4 on trashcan, can I use the D16 with RSSI on ?
Or just go with the D8 with RSSI on ?

SebastiaanRutgers, 2019-04-02:
Flew 3 packs on my Snapper7 yesterday (CrazybeeF3) with betaflight 4.0RC4 and bound with LBT Frsky_X. Worked like a charm but had 1 hard lockup in 3 packs. Had to do a power cycle to get the quad to do anything again. This was with telemetry feature enabled.

fOmey, 2019-04-02:
CrazybeeF4 plagued with the same lockups.. 4.0 RC4, Frsky_X.. first pack in lock up.

deepdigger, 2019-04-07:
@fOmey I have the same problem

chupocro, 2019-10-11:
I have the same problem but with SBUS receiver (XSR-M), Betaflight 3.5.0 and SP F3. There is RX Signal Received = 0 event in the flight log whenever that happens.

drewzh, 2019-10-13:
This is still occurring for me and apparently others using the latest betaflight 4.0.6. I'm using FRSKY_X_LBT on a Tinyhawk S and I get lockups routinely (can't get through a single battery without a lockup).

Asizon, 2019-10-13:
Please guys, flash 4.1.0,it fix all of frsky_X protocol problems, thanks to @mikeller

drewzh, 2019-10-13:
> Please guys, flash 4.1.0,it fix all of frsky_X protocol problems, thanks to @mikeller

Sounds good! Do all the rates/profiles work as per the factory defaults? (I'm a noob at tuning and can't do this myself yet).

chupocro, 2019-10-13:
> 
> 
> Please guys, flash 4.1.0,it fix all of frsky_X protocol problems, thanks to @mikeller

"As announced before the release of Betaflight 4.0, support for F3 based flight controllers has been removed from Betaflight 4.1."

If there was a bug in FrSky_X protocol which was resolved in Betaflight 4.1.0 it would be nice if someone would fix the bug in the latest Betaflight version that works with F3 flight controllers too.

BTW, according to various forum and github posts it seems the "FrSky_X bug" was introduced somewhere between Betaflight 3.1 and 3.5.

drewzh, 2019-10-13:
I just flashed 4.1 on my Tinyhawk S and I'm still getting random fail-safes just as before :-/

mikeller, 2019-10-13:
@drewzh: What version exactly did you flash?

Asizon, 2019-10-14:
This seems like a hardware or configuration issue.

Asizon, 2019-10-14:
@chupocro,come to slack plataform, a few guys can make 4.1.0 for f3 with the fix, like this

Asizon, 2019-10-14:
[betaflight_4.1.0_CRAZYBEEF3FR.zip](https://github.com/betaflight/betaflight/files/3723070/betaflight_4.1.0_CRAZYBEEF3FR.zip)

Asizon, 2019-10-14:
by the way @chupocro  this problem has been in Frsky_X since birth, nobody introduced any problem, it is a very closed code that developers have had to open

Asizon, 2019-10-14:
Good test @drewzh, if you still have problems like this, it will be a hardware problem, check the antenna

chupocro, 2019-10-14:
> 
> 
> @chupocro,come to slack plataform, a few guys can make 4.1.0 for f3 with the fix, like this

Thank you very much for the info!! I would really like to see the diff between the code before and after the change and/or discussion about debugging that problem.

BTW, the target I am using is **SRF3** which I am assuming is quite common F3 FC.

Regards

chupocro, 2019-10-14:
> 
> 
> by the way @chupocro this problem has been in Frsky_X since birth, nobody introduced any problem, it is a very closed code that developers have had to open

Does it mean FrSky designed the receivers that can be used by only those who reverse engineer how they work? If someone wants to use SBUS he has to first reverse engineer the protocol? If that is the case I don't understand **to whom** they planned to sell their receivers? It would be a situation similar to producing the game console using some ASIC and FPGA circuits and starting to sell it without releasing the datasheets. If someone would like to make a game for that console he had to first reverse engineer the device :-/

mikeller, 2019-10-15:
@drewzh: The fix (#9010) isn't in 4.1.0-RC6. You'll have to use a nightly build or wait for the next release to test it.

@chupocro: I think you are misunderstanding what the FrSky SPI RX code does - it eliminates the need for a dedicated FrSky RX by having the flight controller drive the RF chipset directly. There is no SBus involved, but the flight controller has to 'speak' the OTA protocol used by FrSky, which isn't publicly documented.

chupocro, 2019-10-15:
@mikeller That is another thing then. I was [here](https://github.com/betaflight/betaflight/issues/6172#issuecomment-541123952) reffering to the apparent problem with FrSky SBUS protocol. I am aware this thread is about SPI but I did say: "I have the same problem **but with SBUS receiver**...". There are reports where people said the same receiver was working with some FC firmwares other than Betaflight and there was a problem with Betaflight and it seems it started somewhere around Betaflight 3.1. [This](https://github.com/betaflight/betaflight/issues/7084#issue-380221537) was one of such reports.

I work with µCs for 20+ years and I've been working with UART, 1-wire, SPI, I2C, ... protocols since bit banging days and am aware SPI communication bypasses the µC in the receiver. However, what I misunderstood was - I thought the new version of Betaflight firmware solves **all** problems with FrSky X series of the receivers (including the problem with random **RX Signal Received = 0** events which, when lasting more than failsafe phase 2 delay, cause motors to stop in the air).

BTW, since SBUS problem is present only when using D16 **and telemetry**, that sounds similar to the problem that was solved for SPI receivers (eliminated **potential deadlock in processing received telemetry data** in the case of sync loss). The problem can be avoided by using D8 mode (+ telemetry) or D16 mode with telemetry OFF but EU (LBT) receiver firmwares don't have D8 mode.

mikeller, 2019-10-15:
@chupocro: This issue is specifically a discussion of issues with the FrSky SPI RX implementation, so I do not know why you are posting here about a problem with different hardware.

Also, note that #7084 that you reference was identified as a hardware issue. So if your problem is the same as this then I think opening a support case with FrSky and asking them for a fix is the right course of action.

chupocro, 2019-10-15:
@mikeller Well, I just wanted to mention the similar problem but with SBUS hoping someone might take a look the SBUS code too and then I misunderstood the new firmware resolves **all** problems with FrSky receivers. It may be true the #7804 was marked as a hardware problem but by reading through all the messages (and other reports) it could be noticed:

- Some people reported while the same receiver worked well with Cleanflight, it didn't work well with Betaflight
- Many reported the problem is only when using the telemetry
- Many reported the problem is only when using D16 mode
- There are reports everything worked well before Betaflight 3.x.x

My conclusion is the problem is not as visible because:

- Many don't use the telemetry
- Many don't use D16 mode (which is not possible with EU firmware)
- Many switched to some other receivers
- Some think this is a hardware problem
- Some think this is a problem with the receiver firmware

I think much more people is using non-EU firmware where they have the D8 mode which works well and that is the major reason why this problem is not exposed more.

[29 more replies]