betaflight/betaflight #6174: Ledstrip not working holybro kakute f4v2

glynnyfpv, 2018-06-19:
Since updating from bf3.3.3 the ledstrip does not work. i have even swapped out the leds twice with new ones just in case it was a hardware problem, still to no resolve. im unsure how github fully works so i hope this is right. cheers

```# diff

# version
# Betaflight / KAKUTEF4V2 (KTV2) 3.4.0 Jun 17 2018 / 08:34:44 (2fe9d7990) MSP API: 1.39

board_name KAKUTEF4V2
manufacturer_id 

# name
name furious220

# resources

# mixer

# servo

# servo mix

# feature
feature RSSI_ADC
feature LED_STRIP
feature AIRMODE

# beeper
beeper -RX_LOST
beeper -RX_LOST_LANDING
beeper -DISARM_REPEAT

# beacon

# map

# serial
serial 0 0 115200 57600 0 115200
serial 2 4160 115200 57600 0 115200
serial 4 1024 115200 57600 0 115200
serial 5 8192 115200 57600 0 115200

# led
led 0 0,0::A:2
led 1 1,0::A:2
led 2 2,0::A:2
led 3 3,0::A:2
led 4 4,0::A:2
led 5 5,0::A:2
led 6 6,0::A:2
led 7 7,0::A:2

# color

# mode_color
mode_color 6 0 8

# aux
aux 0 0 0 1700 2100 0
aux 1 1 1 1300 1700 0
aux 2 2 1 1700 2100 0
aux 3 27 3 1700 2100 0
aux 4 13 2 1550 2100 0
aux 5 35 2 1700 2100 0

# adjrange

# rxrange

# vtx

# rxfail
rxfail 7 s 1900

# master
set gyro_sync_denom = 2
set gyro_use_32khz = ON
set mag_hardware = NONE
set baro_hardware = NONE
set fpv_mix_degrees = 30
set serialrx_provider = IBUS
set blackbox_p_ratio = 16
set blackbox_device = SPIFLASH
set motor_pwm_protocol = DSHOT1200
set failsafe_delay = 10
set bat_capacity = 1500
set current_meter = ESC
set pid_process_denom = 2
set osd_warn_batt_not_full = OFF
set osd_warn_core_temp = OFF
set osd_rssi_alarm = 50
set osd_tim1 = 2561
set osd_vbat_pos = 2476
set osd_rssi_pos = 2445
set osd_tim_1_pos = 2100
set osd_flymode_pos = 2093
set osd_vtx_channel_pos = 2081
set osd_current_pos = 2507
set osd_mah_drawn_pos = 2455
set osd_warnings_pos = 2250
set osd_esc_tmp_pos = 2488
set osd_stat_tim_1 = ON
set osd_stat_tim_2 = OFF
set osd_stat_max_dist = ON
set osd_stat_endbatt = ON
set osd_stat_battery = ON
set osd_stat_max_alt = ON
set vtx_band = 1
set vtx_channel = 4
set vtx_power = 3
set vtx_freq = 5805

# profile
profile 0

set dterm_notch_cutoff = 0
set vbat_pid_gain = ON

# rateprofile
rateprofile 0

set roll_rc_rate = 140
set pitch_rc_rate = 140
set yaw_rc_rate = 140```

I can do a full dump if that helps cheers

AirBreak69, 2018-06-19:
Duplicate of #6168.
Fixed with #6141.
Please close if you tested OK using the next release after bf3.4rc2.
You can try a nightly build from here: https://ci.betaflight.tech/job/Betaflight/ -> artifact

glynnyfpv, 2018-06-19:
@AirBreak69 sorry I don't quite understand, I had a look at the fix #6141 I'm not sure how to implement it

AirBreak69, 2018-06-20:
Your issue has already been noticed by other users of the same and other FCs.
The issue was fixed with PR #6141. If you don't want to wait for the next BF release: follow the given link to the prerelease builds, download the ZIP from "artificats". The ZIP should contain a .hex file for your FC, too.

glynnyfpv, 2018-06-20:
Thanks @AirBreak69 I understand now. Cheers. Kudos to you and other developers

glynnyfpv, 2018-06-20:
Fix #6141 works thanks for everyone's help

closed 2018-06-20 by glynnyfpv

SgtDimes, 2018-06-20:
Also confirmed working on the Revolt. Color output are incorrect though. @glynnyfpv Do you have the correct colors represented by your leds?

DieHertz, 2018-06-20:
Thanks @AirBreak69

mikeller, 2018-06-20:
@SgtDimes: Check if changing the setting for `ledstrip_grb_rgb` fixes the colours for you.

SgtDimes, 2018-06-20:
@mikeller Yes, that fixed it. Well done.