betaflight/betaflight #6175: change stock TPA threshold to 1050

theSkonk, 2018-06-20:
as recommended by  Flint  i changed my TPA threshold to 1050. this has had a dramatic effect on my PID settings allowing  me to use the stock settings with a small reduction in D on my high power to weight racing rig with out any mid to high oscillations what so ever while using plenty of P low in the throttle to give good control and response. Can we change the stock threshold for the next RC version as out of the box its going to help a lot of people get better flight characteristics out of the stock tune and have basically no negative effect on those who's machines don't need it. setting it lower also makes the effect more gentle over the whole throttle range.

# diff

# version
# Betaflight / OMNIBUSF4FW (OBFW) 3.4.0 Jun 14 2018 / 08:55:46 (1066ff9) MSP API: 1.39

board_name OMNIBUSF4FW
manufacturer_id 

# name
name raptor alpha

# resources
resource MOTOR 1 B05
resource MOTOR 4 B00

# mixer
mixer CUSTOM
mmix 0  1.000 -0.815  1.000 -1.000
mmix 1  1.000 -0.815 -1.000  1.000
mmix 2  1.000  0.815  1.000  1.000
mmix 3  1.000  0.815 -1.000 -1.000

# servo

# servo mix

# feature
feature AIRMODE
feature ESC_SENSOR

# beeper
beeper -GYRO_CALIBRATED
beeper -RX_LOST_LANDING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -BAT_CRIT_LOW
beeper -BAT_LOW
beeper -RX_SET
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -ARMED
beeper -ON_USB

# beacon
beacon RX_LOST
beacon RX_SET

# map
map TAER1234

# serial
serial 2 2048 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1700 2100 0
aux 1 27 2 1700 2100 0
aux 2 13 2 1700 2100 0
aux 3 26 1 1300 1700 0
aux 4 35 1 1700 2100 0

# adjrange

# rxrange

# vtx

# rxfail

# master
set align_gyro = CW270
set gyro_to_use = SECOND
set acc_hardware = NONE
set mag_hardware = NONE
set rc_smoothing_type = FILTER
set rc_smoothing_input_hz = 40
set rc_smoothing_derivative_hz = 40
set fpv_mix_degrees = 40
set serialrx_provider = SBUS
set blackbox_p_ratio = 64
set dshot_idle_value = 650
set motor_pwm_protocol = DSHOT1200
set vbat_min_cell_voltage = 32
set vbat_warning_cell_voltage = 33
set current_meter = ESC
set beeper_dshot_beacon_tone = 4
set yaw_motors_reversed = ON
set deadband = 1
set yaw_deadband = 1
set pid_process_denom = 1
set runaway_takeoff_prevention = OFF
set osd_warn_esc_fail = OFF
set osd_warn_core_temp = OFF
set osd_alt_alarm = 135
set osd_esc_temp_alarm = 105
set osd_throttle_pos = 2049
set osd_current_pos = 2102
set osd_altitude_pos = 2295
set osd_avg_cell_voltage_pos = 2071
set osd_esc_tmp_pos = 2273
set osd_esc_rpm_pos = 2081
set osd_stat_tim_1 = ON
set osd_stat_tim_2 = OFF
set osd_stat_max_spd = OFF
set osd_stat_endbatt = ON
set osd_stat_min_rssi = OFF
set osd_stat_max_alt = ON
set osd_stat_bb_no = OFF
set debug_mode = GYRO_SCALED
set vtx_band = 5
set vtx_channel = 2
set vtx_power = 4
set vtx_freq = 5695
set vcd_video_system = PAL

# profile
profile 2

set dterm_lowpass_hz = 90
set dterm_lowpass2_hz = 180
set dterm_notch_cutoff = 0
set anti_gravity_threshold = 250
set anti_gravity_gain = 6000
set dterm_setpoint_weight = 80
set d_pitch = 22
set d_roll = 22

# rateprofile
rateprofile 0

set thr_expo = 65
set roll_rc_rate = 75
set pitch_rc_rate = 75
set yaw_rc_rate = 75
set roll_expo = 50
set pitch_expo = 50
set yaw_expo = 50
set roll_srate = 85
set pitch_srate = 85
set yaw_srate = 85
set tpa_rate = 12
set tpa_breakpoint = 1050

https://l.facebook.com/l.php?u=https%3A%2F%2Fwww.dropbox.com%2Fs%2Fgqfqo9c52nricr3%2FBTFL_BLACKBOX_LOG_raptor_alpha%2520stock%2520pids%2520lowerd%2520d%25205149%2520props%2520.BBL.zip%3Fdl%3D0&h=AT2vvlPhoYqPdakGzfO0q_711z6yiwOa982y81EgAxoCb0Mp3aiMMUZMjVsdkULVNzH2AloKXI6q6F-rk2B4vE2-rqfML80A8ZvbBTdP4LNPVTt4r-Z3_k27d99g-Pq066pO5k-k

flint723, 2018-06-20:
If anyone who tested the "proposed defaults for 3.4" noticed a change with 1050 breakpoint, could you post here?  I noticed a big improvement on some of my quads, some it did not seem to affect them much (but did not notice any bad behavior).  @theSkonk already described some advantages.  In addition, it allows for higher P below 5% throttle for crisper flips and rolls, but the P and D start dropping immediately, so does not introduce oscillation at 25-60% throttle.

As more powerful motors get more and more common for 5" (even 6") we see a problem emerge. The motors have much more authority over the prop at lower throttle.  P and D settings that create crisp rolls and flips at idle (0% throttle) are too high almost immediately when the stick is raised.  I think this is why the setting is needed.  Where in the past (2204, 2205 motors for mid weight 5") higher P and D was ok up above half throttle.

Also if anyone else is interested in testing this setting (doesn't have to be in 3.4), I'm looking for feedback.  I'm already sure it's a very good change for light-mid weight powerful quads.  What I'm mainly looking for is how the setting affects heavier freestyle quads with gopro (I don't own any to test).  The heaviest quad I tested was a Helix (no go pro) which is 600g AUW.

TIA, Flint

theSkonk, 2018-06-20:
my heaviest is is 410g AUW lol

DusKing1, 2018-06-24:
I'm a bit of confused. I know that setting tpa berakpoint to 1050 will improve the quad's performance, but if I have a little bit of propwash at 1750 for example, what should I do to my tpa value ? After all, betaflight only have one single liner tpa line.

theSkonk, 2018-06-24:
@DusKing1 TPA wont effect prop wash adding  more D or lifting your lower filters will ( reducing filter delay ). i'd actually like to see a 2 point TPA so you can have a mid and a high tpa letting you tune it into a curve.

ctzsnooze, 2018-06-29:
TPA is not for prop wash.  Typically prop wash is better controlled with high P and D.

TPA is for wobble at high throttle that you know is less bad because you tested what happened when you cut PIDs. 

Setting the TPA start point there means zero effect at 1750.  It works by linearly reducing P and D *above* the setpoint.  If an effect is needed at 1750, a threshold of 1500 and amount of 20% will result in a 10% reduction at 1750 and a 20% reduction at 2000.

closed 2018-08-05 by stale[bot]