betaflight/betaflight #6178: BF3.4.0 RC2: No GPS data through smart port Omnibus F4 V3

Ant73, 2018-06-20:
1. No GPS data through smart port Omnibus F4 V3 (pin-out attached as ID'ing these boards is difficult), works with BF3.3.3. I have confirmed by flashing BF3.3.3, got all smart port sensors working, then re-flashed BF3.4.0-RC2 and confirmed GPS data is not being received by Tx after following the same setup procedure. (see 4 below) BF3.3.3 diff also attached for comparison if needed.

2. Smart port connected from R-XSR main plug to PWM 6. All other smart port sensors are working (VFAS, current, Alt, etc) 

3. FC: Omnibus F4 V3 clone (non-pro) - target OMNIBUSF4SD-RC2
Rx: R-XSR -smart port connected from the plug.
GPS: Ublox M8N with HMC5883 mag.
Tx: Taranis X9D+

4. BF3.4.0-RC2. 
Resources set for compass to I2C_SCL/SDA B10/B11 and smart port SERIAL_TX 11 A08 (PWM 6). GPS on UART 1, confirmed sats in BF Configurator 10.3.0 and position accurate. 
Setup configuration after flash;
a. Ports tab>UART6>serial rx 
b. Ports tab>UART1>GPS auto
c. Config tab>Set Rx to serial based receiver>serial provider SBUS
d. Config tab>softserial on + telemetry on
e. Config tab>GPS on, UBLOX
f.In CLI;
resource serial_tx 3 none
resource i2c_scl 2 b10
resource serial_rx 3 none
resource i2c_sda 2 b11

set align_mag =CW270FLIP

resource motor 6 none
resource sonar_echo none
resource serial_tx 11 a08
save

g. Ports tab>softserial to smartport/auto

5. Copy of diff.

```
# diff

# version
# Betaflight / OMNIBUSF4SD (OBSD) 3.4.0 Jun 17 2018 / 08:45:19 (2fe9d7990) MSP API: 1.39

board_name OMNIBUSF4SD
manufacturer_id 

# name

# resources
resource MOTOR 6 NONE
resource SONAR_ECHO 1 NONE
resource SERIAL_TX 3 NONE
resource SERIAL_TX 11 A08
resource SERIAL_RX 3 NONE
resource I2C_SCL 2 B10
resource I2C_SDA 2 B11

# mixer

# servo

# servo mix

# feature
feature -RX_PARALLEL_PWM
feature RX_SERIAL
feature SOFTSERIAL
feature GPS
feature TELEMETRY

# beeper

# beacon

# map

# serial
serial 0 2 115200 115200 0 115200
serial 5 64 115200 57600 0 115200
serial 30 32 115200 57600 0 115200

# led

# color

# mode_color

# aux

# adjrange

# rxrange

# vtx

# rxfail

# master
set align_mag = CW270FLIP
set serialrx_provider = SBUS
set gps_provider = UBLOX

# profile
profile 0

# rateprofile
rateprofile 0
```
Additionally, LED strip doesn't work. I don't have that in the diff above but, using the same setup, it works in BF3.3.3 and not in BF3.4.0-RC2. (LED =WS2812B)

First time reporting a potential bug. I hope I have provided the information correctly.
[image]
[BF3.3.3_diff.txt](https://github.com/betaflight/betaflight/files/2118444/BF3.3.3_diff.txt)

jeffystockes, 2018-06-20:
Good morning, i have the same problem with a MatekF722, gps ublox Neo-M8N on UART3. GPS OK on OSD but not on my FrSky X9D (RX X8R). The sensor discovering procedure don't find gps. I try any builts of BF 3.4 without results. Protocol Ublox, on UART4. A BIG thanks :-) for all the job about BF.
For informations  : 
# diff

# version
# Betaflight / MATEKF722 (MKF7) 3.4.0 Jun 20 2018 / 09:32:24 (6acbed6d6) MSP API: 1.39

board_name MATEKF722
manufacturer_id 

# name

# resources

# mixer

# servo

# servo mix

# feature
feature GPS
feature RSSI_ADC

# beeper
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL

# beacon

# map

# serial
serial 2 32 115200 57600 0 115200
serial 3 2 115200 57600 0 115200
serial 4 8192 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 1 1900 2100 0
aux 1 1 2 1300 1700 0
aux 2 3 3 900 1175 0
aux 3 11 4 900 1175 0
aux 4 27 1 900 1175 0
aux 5 46 4 1875 2100 0
aux 6 13 1 900 1100 0
aux 7 28 2 1800 2100 0

# adjrange

# rxrange

# vtx

# rxfail

# master
set gyro_use_32khz = ON
set acc_calibration = 15,-13,7
set mag_hardware = QMC5883
set mag_calibration = -9064,13759,10193
set baro_hardware = BMP280
set min_check = 1000
set max_check = 2000
set fpv_mix_degrees = 15
set blackbox_p_ratio = 16
set dshot_idle_value = 250
set motor_pwm_protocol = DSHOT300
set failsafe_switch_mode = KILL
set gps_provider = UBLOX
set gps_auto_baud = ON
set deadband = 3
set yaw_deadband = 3
set osd_warn_esc_fail = OFF
set osd_warn_core_temp = OFF
set osd_rssi_alarm = 49
set osd_cap_alarm = 1550
set osd_alt_alarm = 150
set osd_vbat_pos = 2145
set osd_rssi_pos = 2113
set osd_tim_2_pos = 2337
set osd_remaining_time_estimate_pos = 2389
set osd_flymode_pos = 2081
set osd_throttle_pos = 2358
set osd_crosshairs_pos = 2253
set osd_ah_sbar_pos = 2254
set osd_ah_pos = 2126
set osd_current_pos = 2242
set osd_mah_drawn_pos = 2371
set osd_gps_speed_pos = 2344
set osd_gps_lon_pos = 2065
set osd_gps_lat_pos = 2053
set osd_gps_sats_pos = 2049
set osd_home_dir_pos = 2093
set osd_home_dist_pos = 2095
set osd_compass_bar_pos = 2122
set osd_altitude_pos = 2262
set osd_power_pos = 2274
set osd_warnings_pos = 2409
set osd_avg_cell_voltage_pos = 2177
set osd_battery_usage_pos = 2376
set osd_disarmed_pos = 2442
set osd_nheading_pos = 2132
set osd_nvario_pos = 2294
set osd_core_temp_pos = 2421
set osd_stat_rtc_date_time = ON
set osd_stat_tim_1 = ON
set osd_stat_max_dist = ON
set osd_stat_battery = ON

# profile
profile 0

set p_pitch = 58
set d_pitch = 35
set p_roll = 40
set i_roll = 40
set d_roll = 30
set p_yaw = 70

# rateprofile
rateprofile 0

set thr_mid = 30
set thr_expo = 50
set tpa_rate = 25

AirBreak69, 2018-06-20:
@codecae Sounds like breaking change in the smartport protocol. Probably by PR #6142.
Issue #6152 may have the same origin. Is this yours?

jflyper, 2018-06-21:
#6142 is about the broken SmartAudio, so it's not related.

#6152 is about CRSF telemetry, so it's not related either.

mikeller, 2018-06-24:
@Ant73, @jeffystockes: I think I've got a fix. Can you please try these hex files (from #6202):

[betaflight_3.4.0_OMNIBUSF4SD.hex.zip](https://github.com/betaflight/betaflight/files/2130354/betaflight_3.4.0_OMNIBUSF4SD.hex.zip)
[betaflight_3.4.0_MATEKF722.hex.zip](https://github.com/betaflight/betaflight/files/2130355/betaflight_3.4.0_MATEKF722.hex.zip)

jeffystockes, 2018-06-24:
I test this hex file for my matekF722 and it's OK, i find gps both on my taranis and osd. Thank's verry mutch. You're the best ! ;-)