betaflight/betaflight #6179: Windows 7 unable to connect when HID is enabled

strangecalling, 2018-06-20:
Betaflight 3.4 rc2 usb HID option

After enabling HID, betaflight configurator does not see the Flight control (in my case Matek F405-STD), I have seen other with this issue.

Windows 7 tries to look for drivers by does not find anything.

Work around; boot Ubuntu which does see the comms port and disable HID

conkerkh, 2018-06-20:
Try to right click inf from this zip file to install driver. I don't really have a way to test it since no W7 on any of my machines:/ 

https://www60.zippyshare.com/v/2YhFKOJT/file.html

strangecalling, 2018-06-20:
I will try later

strangecalling, 2018-06-20:
How do I use the zippyshare link, takes me to a site but does not download
anything?

strangecalling, 2018-06-20:
@conkerkh I have installed the driver and it works on Windows 7

HypurrFPV, 2018-06-22:
I had the same issue with a FPVModel XRacer F4.  Enabled USB_HID_CDC and FC worked as game controller but was no longer a comm port so could not reconnect to disable.  Used bootloader to re-flash firmware to connect to FC.  I will try this solution tonight to see if it works for me, too, and update with the result.

mikeller, 2018-06-23:
Thanks @conkerkh for the driver file. Added it to the wiki:

https://github.com/betaflight/betaflight/wiki/HID-Joystick-Support

@strangecalling: Is this working for you? If so, can you please close this issue?

closed 2018-06-23 by strangecalling

arulrajesh, 2018-08-23:
I have the same issue with windows 8.1 is there a separate driver for this? Couldn't use the windows 7 driver as it gives the message install failed.