betaflight/betaflight #6181: LED strip no longer working on 3.4.0 RC2; working on 3.3.3

jamesdunn, 2018-06-20:
1. Describe problem:
Upon updating to 3.4.0 RC2, LED strip no longer works. The LEDS remain turned off (no color or activity) and cannot be activated. This was functional on 3.3.3.
2. Reproduce:
Flashing 3.4.0 RC2, and setting up LED per configuration in attached dump.
3. Hardware info:
Flight controller: HGLRC F4 FLAME. LEDs: Standard WS2812 RGB LED strip (6 elements on strip).
4. Versions:
Configurator: 10.3.0. BF: 3.4.0 RC2
5. Diffs:
Diff of my previous 3.3.3 configuration (LEDs confirmed working): [diff_3_3.txt](https://github.com/betaflight/betaflight/files/2120951/diff_3_3.txt)
Diff of my new 3.4.0 RC2 configuration (LEDs not working): [diff_3_4.txt](https://github.com/betaflight/betaflight/files/2120902/diff_3_4.txt)

Thank you.

DieHertz, 2018-06-20:
Closed as duplicate of #6146, fixed in master.

closed 2018-06-20 by DieHertz

jamesdunn, 2018-06-20:
Thanks -- sorry I didn't catch this in the earlier issues.