betaflight/betaflight #6183: YUPIF7 - Orientation issue of the Gyro Accelero with 3.4 RC2

Faduf, 2018-06-20:
I have just got time to make more tests with the 3.4 RC2 on the YupiF7 target and it appears that the orientation of the gyro/accelero is no more correct. 

I have found no changes on this variables in the history of the target.h file but the gyro accelero is turned by 90° on the yaw axis, when I checked the inclinations on the Configurator.

#define ACC_ICM20689_ALIGN      CW90_DEG 
#define GYRO_ICM20689_ALIGN     CW90_DEG

Faduf, 2018-06-22:
I did some debug and it seems that the function alignSensors doesn't work properly.
```
alignSensors(acc.accADC, acc.dev.accAlign);
```

Faduf, 2018-06-22:
the function alignSensors doesn't get correctly the variable acc.dev.accAlign.
Bugfix with PR #6192

etracer65, 2018-06-22:
If that function didn't work then every target would have incorrectly aligned sensors.

Faduf, 2018-06-22:
This function is working correctly. I think that this problem is a memory usage issue or a code optimisation issue. thats why it may not affect all the other targets.

Faduf, 2018-06-22:
This function works correctly. I think this is a memory usage or a compilation optimisation issue. That's why it doesn't affect all other targets.

Faduf, 2018-06-27:
No changes with the 3.4.0 RC3, the board orientation is still not Ok with `FAST CODE` on the YupiF7 target.
Removing `FAST CODE` before this function `alignSensors` is solving the problem.

See #6192

closed 2018-07-02 by Faduf