betaflight/betaflight #6185: Emax F4 Magnum All In One FPV Stack Tower + BF 3.4.0 RC2 - external I2C-Baro don`t work

Dieselfan, 2018-06-21:
With firmware OBF4 - BF 3.4.0 RC2, the I2C-baro-sensor is activated in the GUI, but in the sensor-tab I don´t see parameters - pemanently on zero-line.
Also a BMP280 don`t work.
In BF 3.3. it was fine.

[get-baro_cli_FROG_F4.Magnum_20180620_175915.txt](https://github.com/betaflight/betaflight/files/2122748/get-baro_cli_FROG_F4.Magnum_20180620_175915.txt)

jflyper, 2018-06-22:
@Dieselfan 
Please provide the entire `diff` output, as well as `resource` and `resource list` output.

Dieselfan, 2018-06-22:
@jflyper 
Hello,
here are the files. Regards
[DIFF_BTFL_cli_FROG_F4 Magnum_20180622_110544.txt](https://github.com/betaflight/betaflight/files/2126997/DIFF_BTFL_cli_FROG_F4.Magnum_20180622_110544.txt)

[RESOURCE_BTFL_cli_FROG_F4 Magnum_20180622_110642.txt](https://github.com/betaflight/betaflight/files/2126999/RESOURCE_BTFL_cli_FROG_F4.Magnum_20180622_110642.txt)

[RESOURCE-LIST_BTFL_cli_FROG_F4 Magnum_20180622_110742.txt](https://github.com/betaflight/betaflight/files/2127000/RESOURCE-LIST_BTFL_cli_FROG_F4.Magnum_20180622_110742.txt)

jflyper, 2018-06-24:
@Dieselfan 
I can't reproduce the issue.
- Omnibus F4 AIO (gen1)
- MS5611 connected to I2C2 (PB10=TX3 and PB11=RX3)
- Flashed with 3.4-RC2 and loaded with your config diff

Dieselfan, 2018-07-03:
@jflyper 
Sorry for the long time.
Problem solved. It was a cable break.

closed 2018-07-03 by Dieselfan