betaflight/betaflight #6190: Caddx SDR2 Camera control not working

sniperxnl, 2018-06-22:
Betaflight tested: 3.3 and 3.4rc1
Caddx SDR1 is working, SDR2 i can't enter the osd menu.

Any fix for it?

DieHertz, 2018-06-22:
Caddx did not provide their new cameras to Betaflight developers, only you can fix it now

jumatic, 2018-07-13:
Got it working with omninus F4 nano V5.1 and MAtek F405 AIO.
With 220 Ohm or below down press doesnt work.
Take a 320 Ohm resistor and it works like a charm on the Matek Board on LED PIN.
For the Omnibus F4 nano V5.1 same settings as below, also LED Pin, but SOFTWARE_PWM. All directions work, but not as good as on the matek F4. Needs some tuning ,maybe in key delay.

These are the settings works perfectly for me.

# get camera
camera_control_mode = HARDWARE_PWM
Allowed values: HARDWARE_PWM, SOFTWARE_PWM, DAC

camera_control_ref_voltage = 317
Allowed range: 200 - 400
Default value: 330

camera_control_key_delay = 120
Allowed range: 100 - 500
Default value: 180

camera_control_internal_resistance = 264
Allowed range: 10 - 1000
Default value: 470

camera_control_inverted = OFF
Allowed values: OFF, ON

measured values from the control board:
Enter 2,01V 45,8K
Left 1,63V 27,9K
Up 1,20V 15,9K
Right 0,72V 7,7K
Down 0,11V 0,99K

closed 2018-08-19 by stale[bot]