betaflight/betaflight #6191: JetiEX serial not working

divenet, 2018-06-22:
Betaflight Configurator 10.2.0
Betaflight F3
BFF3 3.4.0 RC2
In the Configuration tab the Receiver Mode serial receiver, will not stay on Serial Receiver Provider "JETIEXBUS" as the other version did. i have also changed it in CLI and no success.

jflyper, 2018-06-23:
@divenet 
Looks like JETIEXBUS` was dropped from F3 targets in 3.3 four month ago.

divenet, 2018-06-24:
Why, i worked so hard to get it working and learn everything about it so i could coach other Jeti Duplex owners, and now its out, i guess i am going to ask to get it back even for F3. All my quads are set up with F3 and i can't take advantage of the goodies in 3.4. please update me as to why and can we get it back?

divenet, 2018-07-18:
please find a way to reinstate Jeti exbus for F3 platform as all my quads have them and i am sure i am not alone in that predicament, i know we Jeti owner have a very small market share but we would love to enjoy the work all dev have put into 3.4 as we are stuck in 3.25. I am sure one could find a legacy code that could be dropped and make place for the memory needed for Jeti support in F3

mikeller, 2018-07-19:
@divenet: Unfortunately pretty much everything that isn't absolutely required to make it fly and give it good flight performance has already been stripped out of F3.

The option that you have is to go the route of doing custom builds with other RX disabled and Jeti enabled instead: https://github.com/betaflight/betaflight/blob/master/docs/Customized%20Version.md

closed 2018-08-25 by stale[bot]