betaflight/betaflight #6198: Betaflight 3.4 RC2 no GPS no arming problem

SchonRichtig, 2018-06-23:
Hi,

I have a problem with a new build.
I put a GPS on it, to have a backup on long range flights.
With bf3.4rc1 and RC2 I can't arm the quad without enough sat's.
I wanted to try it inside, but no way.
I know rth won't work if I don't have good GPS on start, but prevent arming is a mit too much for me.
A warning in osd "no GPS,  no rth" and maybe another arming beeping should be enough.
Or a switch to disable all GPS would be a solution.

So if I have to disarm the quad and it lands upside down, I can't arm it, even in crash recovery mode, or you have to arm it (again) inside a building.

King regards 

Steffen

mikeller, 2018-06-23:
@SchonRichtig: This is by design - allowing users who have set up GPS rescue to take off without having any chance of GPS rescue working does not make much sense - if you don't want GPS rescue to work then don't configure it.

However, I can see the point that you are making about bench testing / indoor testing. One option would be to assign a double function to the GPS rescue switch:
- when disarmed / arming, 'on' means GPS rescue will be disabled after arming;
- when armed, 'on' means the craft goes into GPS rescue mode.

What do you guys think, @s0up, @dbasch?

SchonRichtig, 2018-06-23:
Yes, maybe a mode option GPS on or GPS off is absolutely enough.
So you can use the quad inside or outside.
Or if you land Upside down you are able to arm it in crash recovery mode.

I only need the GPS backup if I go for long range, but I want also use the quad inside buildings. 

I tried to disable GPS in the configurator, deleted the rescue mode, but the arming problem stayed.

mikeller, 2018-06-23:
@SchonRichtig:

> I tried to disable GPS in the configurator, deleted the rescue mode, but the arming problem stayed.

The dependency on a GPS signal for arming is tied to the 'GPS RESCUE' switch. So if you remove that switch in configurator the dependency will go away.

SchonRichtig, 2018-06-23:
Ah, Ok,

I have the BN-180 GPS in a case to mount in on a battery strap only when I need it.

So if it is possible to make the quad arm while the GPS rescue is aktive by a switch and not arm if GPS rescue is disabled, your double Funktion solution.
If bf then change the failsafe mode, when started with GPS, failsafe is rth, without GPS is drop, then I am satisfied :)

Great job!!!!

s0up, 2018-06-23:
You can simply set gps_rescue_min_sats to zero if you want to arm or test
indoors 😀

SchonRichtig, 2018-06-23:
Thanks,

if I set it to zero and want to use the rth Funktion I simply wait until I have at least 8 sat's before I arm the quad?
And indors I can arm without any.
I will try it later, thank you very much!

But there should be the possibility to use the crash recovery mode in the standard configuration.

Kind regards 

Steffen

s0up, 2018-06-23:
I would hang tight on testing rescue until RC3 this weekend.  There are
known issues on rc2!  And yeah if you plan on maybe needing rescue mode
just wait for a good lock before you take off.

Smeat, 2018-06-23:
I would love to be able to disable GPS rescue with a switch on a per flight basis.
Sometimes I fly at locations, where I either don't have a good signal or where GPS rescue would be worse than a normal failsafe. Additionally I could fly my first pack without having a GPS fix yet :)
Since I don't want to change the settings every time I ended up disabling it.

Another concern would be if you crash in a tree (or something similar) and lose GPS signal (e.g. due to orientation or damage) you won't be able to arm and (potentially) free your quad.

mikeller, 2018-06-24:
Ok, this is addressed in #6204. Arming is now disabled only the first time after plugging in a battery, if there is no GPS home position recorded (we absolutely need this for GPS rescue to be able to work). After that, arming is never disabled by the GPS rescue code.

As a workaround for testing / indoor flying, set `gps_rescue_min_sats` to 0. This will put GPS rescue into a 'best effort' mode, where it may or may not work, depending on if there was a home fix when taking off.

SwiftFlying, 2018-06-24:
hi mikeller
Can you tell me your mailbox address? I want to discuss some problems with you

mikeller, 2018-06-24:
Hi @SwiftFlying. You can join the Betaflight Slack (registration at https://slack.betaflight.com/), and we can discuss there.

closed 2018-06-24 by mikeller

matoy, 2019-03-24:
Hi guys,

since BF4.0 (tested with rc2 & 3), it's not possible to set gps_rescue_min_sats to 0 anymore.
CLI anwser is:
###ERROR: INVALID VALUE###
Allowed range: 5 - 50

any workaround be able to arm even if there is no gps signal?
i too would like to be to fly indoor; gps_rescue_min_sats to 0 was fine to me before BF4.0

thanks.

matoy, 2019-03-24:
Ok; i answer to myself, answer is here: #7320 ;)

PR #6204 (2018-06-24): Enabled rearming with 'GPS rescue' enabled and no GPS fix.
Fixes #6198.