betaflight/betaflight #6208: Accidently choosing wrong SPI Reciever Provider for SPI_RX support Receiver Mode

Ask1me, 2018-06-24:
My board FC CrazybeeF3 BNF Frysky,

I accidently choosing wrong SPI Reciever Provider to A7105_Flysky_2A and save it

now my FC can not connect to USB anymore, everytime connect it is says " Unkown USB device (Device descriptor request Failed) for windows 10"

I can not connect betaflight configuration to Re-Flash

is there any suggestions to fix it? please help me anyone..

Ask1me, 2018-06-25:
@mikeller i'm very sory if im rude, i was new in betaflight and github forum.. this is my hope to get right my FC board

so how step to step procedure sir? 

I really dont know how i must do.. hopefully that my mistakes is not wrecked or damage my crazybeeFC board

Thank You very much for your help

Ask1me, 2018-06-25:
@etracer65 thank you for your advice sir, but how can i do it? if my PC cannot read my FC board it keep saying " Unkown USB device (Device descriptor request Failed) for windows 10" this is comes when everytime i try to connect my FC board to PC

etracer65, 2018-06-25:
@Ask1me You need to jumper the boot pads on the flight controller - see the manual for that board.  Then follow the instructions on the Firmware flashing page in the configurator.

closed 2018-08-06 by stale[bot]