betaflight/betaflight #6209: Target feature request: SPracing f7 dual:  Ability to choose which gyro to use on each axis.

flint723, 2018-06-24:
Based on testing I've just done, which gyro is used does affect noise seen on each axis.  For `first` gyro roll is much cleaner than pitch.  For `second` gyro pitch is much cleaner.  So the ability to choose which gyro to use for each axis will give a better result than averaging the two gyros.

In this testing all used 32k/32k/multishot.  No other changes other than `gyro to use`

`gyro to use = first`:
[image]

`gyro to use = second`:
[image]

Logs:
[spracing f7 dual gyro first vs second.zip](https://github.com/betaflight/betaflight/files/2131018/spracing.f7.dual.gyro.first.vs.second.zip)
I only pictured the "lower pids" logs but all of the logs exhibit the same effect.  Also yaw axis has pretty much zero difference.

timman2er, 2018-07-28:
Working on it 👍

DieHertz, 2018-07-28:
It can't record the average as there no average until all filtering is done.

closed 2026-09-26 by blckmn