betaflight/betaflight #6212: Throttle Correction Feature broken in BF3.4

K1nslayer, 2018-06-25:
Hey it looks like the "Throttle Correction" feature is broken in the latest BF3.4 RC candidates.

This can be reproduced by maxing out the throttle adjustment and testing with BF3.4RC1-3 firmware, vs BF3.3 (where it worked great).

REPRO:
If you commit the following CLI lines, the quad should dramatically gain altitude when pitching or rolling.  With BF3.4 this has stopped working as described in the CLI docs.
set thr_corr_value = 150
set thr_corr_angle = 400

I'm running a Tiny Whoop with a BetaFPV board.  Target = BetaflightF3.  I know Its possible this feature was turned off intentionally to save space in the FW because of the F3 processor, but I see no evidence of that.  More likely some change related to the new Throttle Boost feature may be interfering, at least that is my uninformed guess.
Let me know if I can provide more information if the repro proves difficult.  Thanks for all your awesome work!

```

# Betaflight / BETAFLIGHTF3 (BFF3) 3.4.0 Jun 10 2018 / 02:20:23 (a5ba01666) MSP API: 1.39

board_name BETAFLIGHTF3
manufacturer_id 
name SUPER WHOOP

feature -TELEMETRY
feature -ANTI_GRAVITY
feature MOTOR_STOP
map TAER1234
aux 0 0 0 1300 2100 0
aux 1 1 1 900 1100 0
aux 2 2 1 1150 1350 0
aux 3 28 1 1650 1850 0
aux 4 35 1 1900 2100 0
adjrange 0 0 2 900 2100 12 2 0 0
rxfail 3 h
set gyro_lowpass_hz = 120
set acc_trim_pitch = 34
set acc_trim_roll = 2
set max_check = 1950
set rssi_channel = 8
set blackbox_device = NONE
set motor_pwm_protocol = DSHOT600
set thr_corr_value = 150
set thr_corr_angle = 400
set failsafe_off_delay = 15
set failsafe_throttle = 1400
set failsafe_procedure = AUTO-LAND
set bat_capacity = 550
set vbat_max_cell_voltage = 44
set vbat_min_cell_voltage = 27
set vbat_warning_cell_voltage = 29
set small_angle = 180
set pid_process_denom = 2
set osd_vbat_pos = 2401
set osd_rssi_pos = 2426
set osd_craft_name_pos = 10
set osd_warnings_pos = 2410
set osd_pit_ang_pos = 289
set osd_rol_ang_pos = 321
set vcd_video_system = NTSC
profile 0

set dterm_lowpass_hz = 80
set dterm_lowpass2_hz = 160
set dterm_notch_cutoff = 0
set anti_gravity_gain = 2000
set setpoint_relax_ratio = 50
set dterm_setpoint_weight = 0
set level_limit = 39
rateprofile 0

set roll_rc_rate = 90
set pitch_rc_rate = 90
set yaw_rc_rate = 201
set roll_srate = 0
set pitch_srate = 0
set yaw_srate = 0

# 
```

mikeller, 2018-06-25:
I can't quite put my finger onto why this doesn't work any more. Can you please quickly check if throttle correction starts working again when you disable Iterm rotation (`set iterm_rotation = off`)?

K1nslayer, 2018-06-26:
Sadly no change from "set iterm_rotation = off".  Throttle Correction still fails to produce a noticeable result.  The thr_corr_value is still set at 150 on my FC, so any result should be obvious.  Any other suggestions to try on my end?

etracer65, 2018-06-28:
@K1nslayer I have a fix for this problem: #6235 

Can you please give it a try?  

[betaflight_3.4.0_BETAFLIGHTF3.hex.zip](https://github.com/betaflight/betaflight/files/2147148/betaflight_3.4.0_BETAFLIGHTF3.hex.zip)

K1nslayer, 2018-06-29:
Hey etracer, thanks for trying to fix this.  I tested out the private build you linked, but still no significant change in behavior.  With thr_corr_value turned up to max it should basically be hitting the ceiling after the first pitch/roll adjustment, so its easy to binary test.  At least that is what it does in BF3.3 with these settings.  Any other suggestions, or information I can provide?  FYI - I have turned iterm_rotation back on if it makes any difference.

K1nslayer, 2018-06-29:
Something else that might be of use.  I was previously using a development build with a fix for
[#5345](https://github.com/betaflight/betaflight/issues/5325) and throttle correction worked great in that build.  So the offending commit happened after Mar15 or Jenkins build ~650.

K1nslayer, 2018-06-29:
OK!  I went back through the Jenkins builds and did a manual binary search.  **Build** [#918](https://ci.betaflight.tech/job/Betaflight/918/changes) **is the first build where the throttle correction feature stops working** (on my F3 Tiny Whoop anyway)  Hopefully this narrows the search radius enough that it will be easier to fix.

etracer65, 2018-06-29:
@K1nslayer The problem is fixed. I went all the way back to the build applicable to the #5325 issue you referenced - build 670. The behavior for that build is identical to the fixed behavior now.

Possibly you were confused by the bugs in the 3.4RC3 version.  In addition to the feature not working when throttle was smoothed, the feature was generally broken even when throttle wasn't smoothed.  In particular it produced a sawtooth pattern of large magnitude on the throttle that ended up multiplying the boost by about a factor of 10!  So it was producing way too much boost.  So possibly you saw that as "normal" and thus the fixed version (which matches previous behavior) seemed like it wasn't working since you'd need to use a higher `thr_corr_value` then with the broken version in 3.4RC3.

It now behaves identically to BF3.3 versions regardless of whether throttle is smoothed.

closed 2018-06-30 by mikeller

K1nslayer, 2018-07-01:
Can you please provide more info on how you are testing this fix?  In my physical repro, the bug persists.  Perhaps you only fixed part of the bug?  It clearly manifests for me in every build > #918  I'm testing on a Brushless Tiny Whoop with throttle correction turned all the way up (thr_cor_value = 150).  Current behavior doesn't match the behavior from BF3.3.0, I just went back and double confirmed it.  Should I open another issue, or re-activate this one?  I'm happy to provide more info, or do additional tests.  Thanks!

etracer65, 2018-07-01:
I tested on 3.3.3, the old pre-3.3 build 670 and with the fix.  For all I used:
```
set thr_corr_value = 150
set thr_corr_angle = 400
```
I tested by hovering, taking my hand off throttle and pitching forward and observed the quad climbing as moving forward.  The behavior was the same on all versions.  Additionally while testing I built debugging versions of both 3.3.3 and 3.4 that logged the actual throttle boosting amount based on angle change and confirmed the values were the same.

What is it you're expecting it to do that it isn't?  The actual amount of throttle boost isn't (and never was) that large.  So maybe you have too high of expectations for this.  I don't think it was every really designed with brushed tiny whoops in mind.

Also as I said previously, prior to the fix 3.4 had a bug that when throttle *wasn't* smoothed the throttle boosting was way over applied - by more than a factor of 10.  So if you're expecting that behavior then that was a bug as well.

K1nslayer, 2018-07-02:
Thanks for responding, hopefully we can figure out why my observed behavior is so different! 
I was hoping you might be debugging purely in code, and therefor not seeing the huge delta in physical behavior I'm seeing between the builds, but so much for that theory.  I'll layout my expectations and observations. 
From my point of view the "throttle correction" feature is intended to auto compensate for pitch and roll inputs, such that the quad generally maintains altitude.  This was done by defining two parameters (thr_corr_value and thr_corr_angle).  The thr_corr_value is the maximum throttle amount that will be added to the mix, and the thr_corr_angle is the angle that determines what percentage of the thr_corr_value is applied per degree.  So generally you would tune this by first setting the thr_corr_angle to some reasonable forward attack angle, then slowly adjust the thr_corr_value up until the quad maintains altitude under pitch & roll.  This is what I originally did in BF3.3.0 and it worked just like I expected it to.  When it seemed like it had stopped working (maintaining altitude) I went back to my configuration and started turning up the thr_corr_angle.  Eventually I was maxed out at thr_corr_angle = 150 with no noticeable effect, so I went back a few builds and my quad immediately hit the ceiling with minimal roll after hovering.  So I logged this bug.  Some observations:

I'm currently flying this quad in "angle" mode, with a lowish angle limit of 39 degrees.  Perhaps that or some other setting specific to my configuration is contributing somehow?

I'm generally testing by rolling side to side, not pitching forward.  Perhaps the roll axis behavior is different?  I will investigate this.

I do expect the throttle contribution from this feature to be significant, because that is what I have observed from previous builds, going all the way back to 3.3.0.  You comment about it being 10X what it should be in previous builds might explain what I've seen, except it doesn't line up with the build numbers.  It would have had to be broken going all the way back to 3.3.0.  Was that the case?

Possibly related to this.  When it was working (<#918) thr_corr_value It was fairly sensitive, only needing a value of 7 or 8 at an angle of 40 to maintain altitude, and higher values would make the quad climb quickly.  But as I said this is how it has always behaved for me, and seemed like logical behavior.

Why don't you expect a maxed out thr_corr_value to have a significant effect?  Is there something about the code implementation that prevents it?  I would expect the tuning knob to go all the way up, where it is possible to add full throttle to the mix.

I question why thr_corr_value seems to have little to no effect on my quad in current builds.  Perhaps the maximum possible value changed?  I believe he CLI output still lists the max as 150 right?  That is what I'm assuming is the max, and testing with.  BTW what units is this value in?

I can flash FW and apply the same DIFF file, and i was literally able to walk through the builds and see exactly when the behavior changed.  However I don't see significant changes in that CL and manually matching the configuration changes has little effect.  I'm kinda stumped.

etracer65, 2018-07-02:
The reason it "broke" on build 918 was because this is when the default for enabling interpolation for throttle got changed - previously throttle was not smoothed unless you manually configured it (and few did).  Since the defaults also included a new feature called throttle boost, that feature requires the throttle to be smoothed hence the default change to interpolation.  The problem was that the bug made throttle angle correction not work when the throttle channel was smoothed - so that's why it appeared to stop working in build 918.  It was broken before that but you didn't know it.

Also broken before that build and after was the other problem I mentioned about the effect being much too large.  So when interpolation was off then you saw the "too large" bug.  This is why you saw a large effect at only 7 or 8 on the `thr_corr_value`.

When working correctly the overall effect is somewhat subtle.  The `thr_corr_value` sets the maximum amount of throttle increase in microseconds and that's scaled based on cosine of the angle.  So in best case it can boost at most ~15%.  In reality it's about half that.  In debugging the most boost I saw was about 90us on the throttle channel (which ranges from 1000-2000us).

As I said, even when working correctly it may not be enough boost for micro brushed quads since it was put in long before they were a thing.  So it's reasonable that there may be a need to increase the maximum throttle "boosting" range.  But the feature is currently working as it did before the bugs in 3.4.

K1nslayer, 2018-07-03:
Ok, that explains everything I'm seeing.  Thanks for taking the time to explain it.  Since currently the maximum effect is so subtle, I agree it would be beneficial to increase the limit.  That way people can quickly confirm its working, and easily tune it to just below gaining altitude.  I suggest pushing the limit up to the full 1000us (clamped obviously) since there shouldn't be any harm in that.  I believe this should be a fairly benign change.  What do you say?  If you were willing to build a BetaflightF3 target, I would be happy to confirm the larger throttle adjustment.  This feature is pretty important to Whoop pilots since they tend to fly in Angle mode.  Right now pilots are trying to solve the throttle mix at the transmitter level (Project Mockingbird).  Having the FC do it automatically would make this a lot easier.  Let me know what you think.

K1nslayer, 2018-07-07:
So do you agree that increasing the range is something that should be done?  Should I log it as a feature request, or is the lack of response an indication that this wouldn't get added?  Please let me know how to proceed.

etracer65, 2018-07-07:
Sorry, just been busy getting ready for the 3.4 release.  I think it's fine to consider increasing the range of `thr_corr_value` although there's no way we would go up to 1000us - people could get into a situation where they couldn't even land.  Please open a separate feature request issue and I'll look into it when I have a chance.

PR #6235 (2018-06-30): Fix throttle angle correction when smoothing throttle; reduce processing overhead
Fixes #6212 

Change the logic to not modify rcCommand directly and instead apply the additional throttle directly in the mixer.

Also move the logic to the attitude task instead of having it calculate in the PID loop. The logic relies on an angle that's only updated in the attitude task so there was no point in running the calculation every PID loop.

Tested with both rc smoothing and interpolation methods, and with throttle smoothed and not smoothed.